danielkucera/esp-ebus #3: ebus 700 not available anymore

Baggerfahrer, 2022-10-16:
Hey there! 

Without any changes from my side the ebus Adapter bought from you some Weeks ago stopped broadcasting data on ebus 700 Topics via mqtt.

Before that i received All the data Points, but somehow that changed.... any clues or help Would Highly be appreciated. 

PS the adapter still sends data on the topics you see in the pic, meaning IT should work Properly

danielkucera, 2022-10-17:
Hi, have you upgraded ebusd or its configs? If you still see some messages (aka adapter works) it sounds more like an issue with ebusd.

Baggerfahrer, 2022-10-18:
Hey 

thanks for the answer,
My log is full with stuff like this, but this has ony to do with readings from ebus bai, no messages with ebus 700.....

Really strange as i did not update anything .....
I will ask in ebus github forum and keep you updated, many thanks

`2022-10-18 19:47:24.017 [mqtt error] decode bai WPPostrunTime: ERR: invalid position
2022-10-18 19:48:08.180 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
2022-10-18 19:48:14.190 [update error] unable to parse poll-read bai ExtFlowTempDesiredMin from 3108b509030d6e04 / 00: ERR: invalid position
2022-10-18 19:48:14.339 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position
2022-10-18 19:48:19.345 [mqtt error] decode bai ExtFlowTempDesiredMin: ERR: invalid position
2022-10-18 19:48:56.154 [update error] unable to parse poll-read bai HcPumpMode from 3108b509030d4904 / 00: ERR: invalid position
2022-10-18 19:48:59.598 [mqtt error] decode bai HcPumpMode: ERR: invalid position
2022-10-18 19:49:20.183 [update error] unable to parse poll-read bai HoursTillService from 3108b509030d2004 / 00: ERR: invalid position
2022-10-18 19:49:24.735 [mqtt error] decode bai HoursTillService: ERR: invalid position
2022-10-18 19:49:32.185 [update error] unable to parse poll-read bai HwcPostrunTime from 3108b509030d1104 / 00: ERR: invalid position
2022-10-18 19:49:34.790 [mqtt error] decode bai HwcPostrunTime: ERR: invalid position
2022-10-18 19:49:50.196 [update error] unable to parse poll-read bai HwcTempDesired from 3108b509030dea03 / 00: ERR: invalid position
2022-10-18 19:49:54.907 [mqtt error] decode bai HwcTempDesired: ERR: invalid position
2022-10-18 19:49:56.168 [update error] unable to parse poll-read bai HwcTempMax from 3108b509030d4304 / 00: ERR: invalid position
2022-10-18 19:49:59.912 [mqtt error] decode bai HwcTempMax: ERR: invalid position
2022-10-18 19:52:20.204 [update error] unable to parse poll-read bai SecondPumpMode from 3108b509030d0b04 / 00: ERR: invalid position
2022-10-18 19:52:24.829 [mqtt error] decode bai SecondPumpMode: ERR: invalid position
2022-10-18 19:52:38.181 [update error] unable to parse poll-read bai Status16 from 3108b5040116 / 00: ERR: invalid position
2022-10-18 19:52:39.892 [mqtt error] decode bai Status16: ERR: invalid position
2022-10-18 19:52:44.183 [update error] unable to parse poll-read bai Status from 3108b5110103 / 00: ERR: invalid position
2022-10-18 19:52:49.999 [mqtt error] decode bai Status: ERR: invalid position
2022-10-18 19:53:56.197 [update error] unable to parse poll-read bai ValveMode from 3108b509030d2a04 / 00: ERR: invalid position
2022-10-18 19:53:59.390 [mqtt error] decode bai ValveMode: ERR: invalid position
2022-10-18 19:54:08.192 [update error] unable to parse poll-read bai VortexFlowSensor from 3108b509030dd500 / 00: ERR: invalid position
2022-10-18 19:54:09.453 [mqtt error] decode bai VortexFlowSensor: ERR: invalid position
2022-10-18 19:54:20.166 [update error] unable to parse poll-read bai WPPostrunTime from 3108b509030df703 / 00: ERR: invalid position
2022-10-18 19:54:24.523 [mqtt error] decode bai WPPostrunTime: ERR: invalid position
2022-10-18 19:55:08.190 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
2022-10-18 19:55:09.778 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position
2022-10-18 19:55:14.182 [update error] unable to parse poll-read bai ExtFlowTempDesiredMin from 3108b509030d6e04 / 00: ERR: invalid position
2022-10-18 19:55:19.837 [mqtt error] decode bai ExtFlowTempDesiredMin: ERR: invalid position
2022-10-18 19:55:56.165 [update error] unable to parse poll-read bai HcPumpMode from 3108b509030d4904 / 00: ERR: invalid position
2022-10-18 19:55:59.079 [mqtt error] decode bai HcPumpMode: ERR: invalid position
2022-10-18 19:56:20.172 [update error] unable to parse poll-read bai HoursTillService from 3108b509030d2004 / 00: ERR: invalid position
2022-10-18 19:56:24.168 [mqtt error] decode bai HoursTillService: ERR: invalid position
2022-10-18 19:56:32.157 [update error] unable to parse poll-read bai HwcPostrunTime from 3108b509030d1104 / 00: ERR: invalid position
2022-10-18 19:56:34.227 [mqtt error] decode bai HwcPostrunTime: ERR: invalid position
2022-10-18 19:56:50.157 [update error] unable to parse poll-read bai HwcTempDesired from 3108b509030dea03 / 00: ERR: invalid position
2022-10-18 19:56:54.392 [mqtt error] decode bai HwcTempDesired: ERR: invalid position
2022-10-18 19:56:56.179 [update error] unable to parse poll-read bai HwcTempMax from 3108b509030d4304 / 00: ERR: invalid position
2022-10-18 19:56:59.398 [mqtt error] decode bai HwcTempMax: ERR: invalid position
2022-10-18 19:59:20.164 [update error] unable to parse poll-read bai SecondPumpMode from 3108b509030d0b04 / 00: ERR: invalid position
2022-10-18 19:59:24.363 [mqtt error] decode bai SecondPumpMode: ERR: invalid position
2022-10-18 19:59:38.162 [update error] unable to parse poll-read bai Status16 from 3108b5040116 / 00: ERR: invalid position
2022-10-18 19:59:39.424 [mqtt error] decode bai Status16: ERR: invalid position
2022-10-18 19:59:44.141 [update error] unable to parse poll-read bai Status from 3108b5110103 / 00: ERR: invalid position
2022-10-18 19:59:49.531 [mqtt error] decode bai Status: ERR: invalid position
[... cut ...]
2022-10-18 20:00:59.960 [mqtt error] decode bai ValveMode: ERR: invalid position
2022-10-18 20:01:08.188 [update error] unable to parse poll-read bai VortexFlowSensor from 3108b509030dd500 / 00: ERR: invalid position
2022-10-18 20:01:09.979 [mqtt error] decode bai VortexFlowSensor: ERR: invalid position
2022-10-18 20:01:20.153 [update error] unable to parse poll-read bai WPPostrunTime from 3108b509030df703 / 00: ERR: invalid position
2022-10-18 20:01:24.046 [mqtt error] decode bai WPPostrunTime: ERR: invalid position`

closed 2022-10-18 by Baggerfahrer