danielkucera/esp-ebus #4: Platform io upgrade

HiQual, 2022-10-31:
Hello.  I see there is a SW change to convert adapter function to Read Only.
I used the Platform io option to do the update.
Fortunately, it failed at the upload because I had forgotten to disable my Windows firewall.
I noted some compile error messages during the compile per the attached img file.

[image]

Compiling .pio/build/esp12e-ota/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:68:53: warning: 'void reset_config()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   68 |   attachInterrupt(digitalPinToInterrupt(RESET_PIN), reset_config, CHANGE);
      |                                                     ^~~~~~~~~~~~
src/main.cpp:46:22: note: declared here
   46 | ICACHE_RAM_ATTR void reset_config() {
      |                      ^~~~~~~~~~~~
src/main.cpp:68:73: warning: 'void reset_config()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   68 |   attachInterrupt(digitalPinToInterrupt(RESET_PIN), reset_config, CHANGE);
      |                                                                         ^
src/main.cpp:46:22: note: declared here
   46 | ICACHE_RAM_ATTR void reset_config() {
      |                      ^~~~~~~~~~~~
Compiling .pio/build/esp12e-ota/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o

Before I try the firmware upload again, could you let me know if the compile error messages can be ignored?
Many thanks ... Ian

danielkucera, 2022-11-05:
Hi Ian,
sorry for late reply, this is only warning, it is safe to continue but I have updated the code so it doesn't complain anymore.
Feel free to download the latest version.

HiQual, 2022-11-05:
Thank you, Daniel. I disbaled my firewall and re-ran the update but got some errors trying both update methods as below ...

CURRENT: upload_protocol = espota
Uploading .pio/build/esp12e-ota/firmware.bin
10:11:29 [DEBUG]: Options: {'esp_ip': 'esp-ebus.local', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 41097, 'auth': '', 'image': '.pio/build/esp12e-ota/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
10:11:29 [INFO]: Starting on 0.0.0.0:41097
10:11:29 [INFO]: Upload size: 383552
10:11:29 [INFO]: Sending invitation to: esp-ebus.local
Traceback (most recent call last):
  File "/root/.platformio/packages/framework-arduinoespressif8266/tools/espota.py", line 342, in <module>
    sys.exit(main(sys.argv))
  File "/root/.platformio/packages/framework-arduinoespressif8266/tools/espota.py", line 337, in main
    return serve(options.esp_ip, options.host_ip, options.esp_port, options.host_port, options.auth, options.image, command)
  File "/root/.platformio/packages/framework-arduinoespressif8266/tools/espota.py", line 103, in serve
    sock2.sendto(message.encode(), remote_address)
socket.gaierror: [Errno -2] Name or service not known
*** [upload] Error 1
============================================= [FAILED] Took 16.68 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
esp12e-ota     FAILED    00:00:16.678
======================================== 1 failed, 0 succeeded in 00:00:16.678 ========================================

C:\Users\ian\Desktop>espota.exe -i esp-ebus.local -f firmware.bin -r -d
10:20:17 [DEBUG]: Options: {'timeout': 10, 'esp_ip': 'esp-ebus.local', 'host_port': 27881, 'image': 'firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
10:20:17 [INFO]: Starting on 0.0.0.0:27881
10:20:17 [INFO]: Upload size: 383552
Sending invitation to esp-ebus.local
10:20:17 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

10:20:29 [INFO]: Waiting for result...
10:20:29 [INFO]: Result: 536
10:20:29 [INFO]: Result: 1072
10:20:29 [INFO]: Result: 536
10:20:29 [INFO]: Result: 1608
10:20:29 [INFO]: Result: 536
10:20:29 [ERROR]: Error response from device

C:\Users\ian\Desktop>

.....Is this still my Windows firewall blocking - even though Windows says it is turned off ?
Many thanks ... Ian

danielkucera, 2022-11-05:
Try to find the IP address of the adapter and replace `esp-ebus.local` in `platformio.ini` with that IP

HiQual, 2022-11-05:
I had forgotten to start Python for Windows b4 running the update. That fixed it.
Sorry to have bothered you.
The adapter has been really useful and helped me to fine tune my heating system for maximum efficiency.
Many thanks again.

closed 2022-11-05 by HiQual