danielkucera/esp-ebus #5: [bus error] device status: unexpected available enhanced byte 2

Wassman2020, 2022-11-09:
Hi.
We use a ebus adapter received last week but we can't see any information in ebusd in home assistant.
We receive only this error.

2022-11-09 19:53:37.054 [bus error] device status: unexpected available enhanced byte 2
2022-11-09 19:53:37.096 [bus error] device status: unexpected available enhanced byte 2
2022-11-09 19:53:37.140 [bus error] device status: unexpected available enhanced byte 2
2022-11-09 19:53:37.183 [bus error] device status: unexpected available enhanced byte 2
2022-11-09 19:53:37.188 [bus notice] signal acquired
2022-11-09 19:53:37.195 [bus error] device status: unexpected available enhanced byte 2
2022-11-09 19:53:37.225 [bus error] device status: unexpected available enhanced following byte 1
2022-11-09 19:53:37.253 [bus error] device status: unexpected available enhanced byte 2

danielkucera, 2022-11-14:
What is your ebusd config/commandline?
What boiler/heater do you use?

Wassman2020, 2022-11-14:
Hi.

I use this configuration in ebusd in home assistant.

scanconfig: true
loglevel_all: debug
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: enh:192.168.2.32:9999
latency: 10
loglevel_network: debug
loglevel_bus: debug

I have connected to ebus these devices:
Valliant ecotec pro
Valliant Vr70
2 valliant  multimatic 700
Valliant Vr920

Br

danielkucera, 2022-11-14:
this line is wrong:
```
network_device: enh:192.168.2.32:9999
```
it doesn't support enhanced protocol, try to use it without `enh:` , although the port seems wrong too, this firmware uses 3333

Wassman2020, 2022-11-21:
I remove ESP12 with Hotair and connected Wemos D1 mini with firmware ebusd-esp of [john30] and i can see information in the bus. I return back to ESP8266 but I think that I have broken the sp12 because now don't power on :-(. led of ebus is O.K but esp12 doesn't switch off.
I am looking for the esp8266-arduino-ebus firmware but I can't find it on the page to program a new esp12 and try the configuration you have shown me.
where can i find the esp12 firmware?

danielkucera, 2022-11-22:
In this repository.

Wassman2020, 2022-11-22:
i I am looking for some *.bin file but I don't see any file in the repository. what directory is it in?

21494, 2023-01-13:
Hi 
I also get a similar error in Home Assistant to https://github.com/danielkucera/esp8266-arduino-ebus/issues/5#issue-1442593261
However I have configured the port to 3333. I tried with 3334, 5555 or 9999 and I have no connection.

I'm using boiler Thema Condens of Saunier Duval.

**ebus config:**
```
scanconfig: true
loglevel_all: debug
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: enh:192.168.1.19:3333
latency: 10
loglevel_network: debug
loglevel_bus: debug

```

**ebus addon HA Logs:**
```
2023-01-13 23:29:11.048 [main notice] ebusd 23.1.p20230106 started with auto scan on enhanced device 192.168.1.19:3333
2023-01-13 23:29:12.794 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-01-13 23:29:13.709 [main info] reading templates /
2023-01-13 23:29:14.758 [main info] read templates in /
2023-01-13 23:29:14.758 [main info] reading file memory.csv
2023-01-13 23:29:15.799 [main info] successfully read file memory.csv
2023-01-13 23:29:15.799 [main info] reading file broadcast.csv
2023-01-13 23:29:16.646 [main info] successfully read file broadcast.csv
2023-01-13 23:29:16.646 [main info] read config files, got 11 messages
2023-01-13 23:29:16.682 [bus notice] device status: resetting
2023-01-13 23:29:16.683 [bus notice] bus started with own address 31/36
2023-01-13 23:29:16.684 [main info] registering data handlers
2023-01-13 23:29:16.684 [mqtt info] mosquitto version 2.0.15 (compiled with 2.0.15)
2023-01-13 23:29:16.692 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.736 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.778 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.793 [mqtt debug] connection requested
2023-01-13 23:29:16.793 [main info] registered data handlers
2023-01-13 23:29:16.822 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.864 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.906 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.949 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:16.993 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.037 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.077 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.121 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.164 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.207 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.251 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.293 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.336 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.379 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.423 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.465 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.508 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.551 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.595 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.640 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.683 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.722 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.767 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.809 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.852 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.895 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.938 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:17.984 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:18.023 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:18.031 [bus notice] signal acquired
2023-01-13 23:29:18.042 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:18.056 [bus error] device status: unexpected enhanced byte 2
2023-01-13 23:29:18.097 [bus error] device status: missing enhanced byte 2
2023-01-13 23:29:18.112 [bus error] device status: missing enhanced byte 2
2023-01-13 23:29:18.123 [bus error] device status: unexpected enhanced byte 2
```

Many thanks for your help.

danielkucera, 2023-01-14:
Remove `enh:` from the config

Veit00, 2023-02-14:
Hallo,

das ist Einstellung ebusd config
EBUSD_OPTS="-d, --device=esp-ebus.local:3333 --scanconfig --latency=200000 c, --configpath=https://cfg.ebusd.eu/"

openhabian@openhabian:~ $ ping esp-eBus.local
PING esp-eBus.local (192.168.100.18) 56(84) bytes of data.
64 bytes from 192.168.100.18 (192.168.100.18): icmp_seq=1 ttl=255 time=6.91 ms
64 bytes from 192.168.100.18 (192.168.100.18): icmp_seq=2 ttl=255 time=98.8 ms
64 bytes from 192.168.100.18 (192.168.100.18): icmp_seq=3 ttl=255 time=97.3 ms
64 bytes from 192.168.100.18 (192.168.100.18): icmp_seq=4 ttl=255 time=99.9 ms
^C
--- esp-eBus.local ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3003ms
rtt min/avg/max/mdev = 6.911/75.714/99.856/39.733 ms

und

openhabian@openhabian:~ $ sudo ebusd --scanconfig --lograwdata=data --log=all:debug -f
2023-02-14 12:59:21.340 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device /dev/ttyUSB0
2023-02-14 12:59:21.340 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-14 12:59:21.576 [main info] reading templates /
2023-02-14 12:59:21.820 [main info] read templates in /
2023-02-14 12:59:21.820 [main info] reading file memory.csv
2023-02-14 12:59:22.065 [main info] successfully read file memory.csv
2023-02-14 12:59:22.065 [main info] reading file broadcast.csv
2023-02-14 12:59:22.310 [main info] successfully read file broadcast.csv
2023-02-14 12:59:22.310 [main info] read config files, got 11 messages
2023-02-14 12:59:22.310 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 12:59:22.310 [bus notice] bus started with own address 31/36
2023-02-14 12:59:22.310 [bus notice] device invalid
2023-02-14 12:59:22.311 [main info] registering data handlers
2023-02-14 12:59:22.311 [main info] registered data handlers
2023-02-14 12:59:27.311 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 12:59:27.311 [bus notice] device invalid
2023-02-14 12:59:32.311 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 12:59:32.311 [bus notice] device invalid
2023-02-14 12:59:32.311 [main debug] performing regular tasks
2023-02-14 12:59:37.311 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 12:59:37.311 [bus notice] device invalid
2023-02-14 12:59:42.311 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 12:59:42.311 [bus notice] device invalid

bitte was mache ich falsch

danke

danielkucera, 2023-02-14:
EBUSD_OPTS don't apply if you run the service directly like `sudo ebusd ...` and it also looks malformed.
Try to put all the options after `sudo ebusd`

danielkucera, 2023-02-14:
check the spelling `--device=esp-eBusd.locel:3334` vs `--device=esp-eBusd.locsl:3334`

Veit00, 2023-02-14:
sorry my English is not good

I checked again, sorry.
but the output remains the same

danielkucera, 2023-02-14:
post the log again, feel free to write in german

Veit00, 2023-02-14:
openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 -d --device=esp-eBusd.local:3334 --loglevel=debug --address=ff -f
2023-02-14 13:41:14.264 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device --device=esp-eBusd.local:3334
2023-02-14 13:41:14.264 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-14 13:41:14.504 [main info] reading templates /
2023-02-14 13:41:14.748 [main info] read templates in /
2023-02-14 13:41:14.748 [main info] reading file memory.csv
2023-02-14 13:41:14.992 [main info] successfully read file memory.csv
2023-02-14 13:41:14.992 [main info] reading file broadcast.csv
2023-02-14 13:41:15.238 [main info] successfully read file broadcast.csv
2023-02-14 13:41:15.238 [main info] read config files, got 11 messages
2023-02-14 13:41:20.248 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 13:41:20.248 [bus notice] bus started with own address ff/04
2023-02-14 13:41:20.248 [bus notice] device invalid
2023-02-14 13:41:20.248 [main info] registering data handlers
2023-02-14 13:41:20.249 [main info] registered data handlers
2023-02-14 13:41:30.249 [main debug] performing regular tasks
2023-02-14 13:41:30.260 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 13:41:30.260 [bus notice] device invalid
2023-02-14 13:41:40.249 [main debug] performing regular tasks
2023-02-14 13:41:40.271 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 13:41:40.271 [bus notice] device invalid

^C2023-02-14 13:41:44.138 [main notice] SIGINT received
2023-02-14 13:41:50.282 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 13:41:50.283 [main notice] ebusd stopped
openhabian@openhabian:~ $

danielkucera, 2023-02-14:
I see... replace `-d --device=esp-eBusd.local:3334` with `--device=esp-eBusd.local:3334`

Veit00, 2023-02-14:
Hallo
ich habe das an einer Vaillant auroMatic 560
die JP1 getrennt und extra Netzteil.

danielkucera, 2023-02-14:
Ok, is that a question of some sort?

Veit00, 2023-02-14:
nein, ich habe es nur erwähnt.

hier noch mal mit Änderung.

openhabian@openhabian:~ $ sudo ebusd -d, --device=esp-eBusd.local:3333 --scanconfig --latency=200000  --configpath=https://cfg.ebusd.eu/
openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 -d --device=esp-eBusd.local:3334 --loglevel=debug --address=ff -f
2023-02-14 14:00:09.165 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device --device=esp-eBusd.local:3334
2023-02-14 14:00:09.166 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-14 14:00:09.406 [main info] reading templates /
2023-02-14 14:00:09.649 [main info] read templates in /
2023-02-14 14:00:09.649 [main info] reading file memory.csv
2023-02-14 14:00:09.894 [main info] successfully read file memory.csv
2023-02-14 14:00:09.894 [main info] reading file broadcast.csv
2023-02-14 14:00:10.132 [main info] successfully read file broadcast.csv
2023-02-14 14:00:10.132 [main info] read config files, got 11 messages
2023-02-14 14:00:15.141 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 14:00:15.142 [bus notice] bus started with own address ff/04
2023-02-14 14:00:15.142 [bus notice] device invalid
2023-02-14 14:00:15.142 [main info] registering data handlers
2023-02-14 14:00:15.142 [main info] registered data handlers
2023-02-14 14:00:25.143 [main debug] performing regular tasks
2023-02-14 14:00:25.160 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 14:00:25.160 [bus notice] device invalid
2023-02-14 14:00:35.143 [main debug] performing regular tasks
2023-02-14 14:00:35.175 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 14:00:35.175 [bus notice] device invalid
2023-02-14 14:00:39.765 [main notice] SIGINT received
^C2023-02-14 14:00:45.186 [bus error] unable to open --device=esp-eBusd.local:3334: ERR: generic I/O error
2023-02-14 14:00:45.186 [main notice] ebusd stopped
openhabian@openhabian:~ $

danielkucera, 2023-02-14:
read carefully again please:  `-d --device=esp-eBusd.local:3334` -> `--device=esp-eBusd.local:3334` - in other words: remove `-d`

Veit00, 2023-02-14:
so in etwa?

openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 --loglevel=debug --address=ff -f                        2023-02-14 14:08:58.734 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device /dev/ttyUSB0
2023-02-14 14:08:58.734 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-14 14:08:59.397 [main info] reading templates /
2023-02-14 14:09:00.442 [main info] read templates in /
2023-02-14 14:09:00.442 [main info] reading file memory.csv
2023-02-14 14:09:00.684 [main info] successfully read file memory.csv
2023-02-14 14:09:00.684 [main info] reading file broadcast.csv
2023-02-14 14:09:00.922 [main info] successfully read file broadcast.csv
2023-02-14 14:09:00.922 [main info] read config files, got 11 messages
2023-02-14 14:09:00.922 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:00.923 [main info] registering data handlers
2023-02-14 14:09:00.923 [main info] registered data handlers
2023-02-14 14:09:00.923 [bus notice] bus started with own address ff/04
2023-02-14 14:09:00.923 [bus notice] device invalid
2023-02-14 14:09:05.923 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:05.923 [bus notice] device invalid
2023-02-14 14:09:10.923 [main debug] performing regular tasks
2023-02-14 14:09:10.924 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:10.924 [bus notice] device invalid
2023-02-14 14:09:15.924 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:15.924 [bus notice] device invalid
2023-02-14 14:09:20.924 [main debug] performing regular tasks
2023-02-14 14:09:20.924 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:20.924 [bus notice] device invalid
2023-02-14 14:09:25.924 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:25.924 [bus notice] device invalid
^C2023-02-14 14:09:27.046 [main notice] SIGINT received
2023-02-14 14:09:30.925 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-02-14 14:09:30.925 [bus notice] device invalid
2023-02-14 14:09:31.025 [main notice] ebusd stopped
openhabian@openhabian:~ $

danielkucera, 2023-02-14:
No, from:
`sudo ebusd --scanconfig --accesslevel=* --latency=200000 -d --device=esp-eBusd.local:3334 --loglevel=debug --address=ff -f`
to:
`sudo ebusd --scanconfig --accesslevel=* --latency=200000 --device=esp-eBusd.local:3334 --loglevel=debug --address=ff -f`

Veit00, 2023-02-14:
***@***.***:~ $ sudo ebusd --scanconfig --accesslevel=*
--latency=200000 --device=esp-eBusd.local:3334 --loglevel=debug --address=ff
-f
2023-02-14 14:16:56.947 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started
with auto scan on device esp-eBusd.local:3334
2023-02-14 14:16:56.948 [main info] loading configuration files from
<https://cfg.ebusd.eu/> https://cfg.ebusd.eu/
2023-02-14 14:16:57.188 [main info] reading templates /
2023-02-14 14:16:57.432 [main info] read templates in /
2023-02-14 14:16:57.432 [main info] reading file memory.csv
2023-02-14 14:16:57.677 [main info] successfully read file memory.csv
2023-02-14 14:16:57.677 [main info] reading file broadcast.csv
2023-02-14 14:16:57.923 [main info] successfully read file broadcast.csv
2023-02-14 14:16:57.923 [main info] read config files, got 11 messages
2023-02-14 14:17:02.934 [bus error] unable to open esp-eBusd.local:3334:
ERR: generic I/O error
2023-02-14 14:17:02.935 [bus notice] bus started with own address ff/04
2023-02-14 14:17:02.935 [bus notice] device invalid
2023-02-14 14:17:02.935 [main info] registering data handlers
2023-02-14 14:17:02.935 [main info] registered data handlers
2023-02-14 14:17:12.936 [main debug] performing regular tasks
2023-02-14 14:17:12.945 [bus error] unable to open esp-eBusd.local:3334:
ERR: generic I/O error
2023-02-14 14:17:12.945 [bus notice] device invalid
2023-02-14 14:17:22.936 [main debug] performing regular tasks
2023-02-14 14:17:22.959 [bus error] unable to open esp-eBusd.local:3334:
ERR: generic I/O error
2023-02-14 14:17:22.959 [bus notice] device invalid
2023-02-14 14:17:32.936 [main debug] performing regular tasks
2023-02-14 14:17:32.969 [bus error] unable to open esp-eBusd.local:3334:
ERR: generic I/O error
2023-02-14 14:17:32.969 [bus notice] device invalid
^C2023-02-14 14:17:35.544 [main notice] SIGINT received
2023-02-14 14:17:42.981 [bus error] unable to open esp-eBusd.local:3334:
ERR: generic I/O error
2023-02-14 14:17:42.981 [main notice] ebusd stopped
***@***.***:~ $

unexpected available enhanced byte 2 (Issue #5)

 

<https://github.com/danielkucera/esp8266-arduino-ebus/issues/5#issuecomment-
OAMVANCNFSM6AAAAAAR3YAHQM> .

Veit00, 2023-02-14:
Hallo hier noch ein Bild

 

Und noch zwei vom Scann der Port's

Veit00, 2023-02-14:
Hallo Daniel Kucera,

 

ich habe noch mal die Verkabelung neu gemacht, alles neu Firmware usw.

 

dann mit PuTTy  Telnet 192.168.100.18 Port 3334 gelesen, siehe Bild.

 

Etwas mit dem Poti gespielt aber entweder gar kein Signal keine Ausgabe oder
misch masch.

 

Meine Frage wie muss das korrekt ausschauen.

 

 

Freundliche Grüße

Veit Tandler

 

 

Veit Tandler

Principala 83

RO-627283 Varnita / Racoasa Jud. Vrancea

 

Mobil: +49 160 95542478

Mail:  ***@***.***> ***@***.***

Veit00, 2023-02-14:
Hallo Daniel Kucera,

 

ich habe noch ein Bild vom IP-Scanner.

Da schaut der Name auch komisch aus.

 

Freundliche Grüße

Veit Tandler

 

 

Veit Tandler

Principala 83

RO-627283 Varnita / Racoasa Jud. Vrancea

 

Mobil: +49 160 95542478

Mail:  ***@***.***> ***@***.***

danielkucera, 2023-02-14:
I see now: `esp-eBusd.local:3334` -> `esp-eBus.local:3334`
Btw the images (if you are sending some) are not attached.

Veit00, 2023-02-15:
Sorry, hier nochmal die Bilder

[image]

[image]

[image]

[image]

Veit00, 2023-02-15:
eins noch
[image]

danielkucera, 2023-02-15:
did you do this change?
esp-eBus**d**.local:3334 -> `esp-eBus.local:3334`

Veit00, 2023-02-15:
guten Morgen,

ja habe ich

danielkucera, 2023-02-15:
please post the full log

Veit00, 2023-02-15:
openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 --device esp-eBus.locel:3334 --loglevel=debug --address=ff -f                    [sudo] password for openhabian:
2023-02-15 09:48:10.070 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device esp-eBus.locel:3334
2023-02-15 09:48:10.070 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 09:48:10.308 [main info] reading templates /
2023-02-15 09:48:10.550 [main info] read templates in /
2023-02-15 09:48:10.550 [main info] reading file memory.csv
2023-02-15 09:48:10.792 [main info] successfully read file memory.csv
2023-02-15 09:48:10.792 [main info] reading file broadcast.csv
2023-02-15 09:48:11.037 [main info] successfully read file broadcast.csv
2023-02-15 09:48:11.037 [main info] read config files, got 11 messages
2023-02-15 09:48:11.048 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:11.049 [bus notice] bus started with own address ff/04
2023-02-15 09:48:11.049 [bus notice] device invalid
2023-02-15 09:48:11.049 [main info] registering data handlers
2023-02-15 09:48:11.049 [main info] registered data handlers
2023-02-15 09:48:16.051 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:16.051 [bus notice] device invalid
2023-02-15 09:48:21.050 [main debug] performing regular tasks
2023-02-15 09:48:21.053 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:21.053 [bus notice] device invalid
2023-02-15 09:48:26.056 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:26.056 [bus notice] device invalid
2023-02-15 09:48:31.050 [main debug] performing regular tasks
2023-02-15 09:48:31.058 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:31.058 [bus notice] device invalid
^C2023-02-15 09:48:33.112 [main notice] SIGINT received
2023-02-15 09:48:36.061 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:36.061 [bus notice] device invalid
2023-02-15 09:48:36.151 [main notice] ebusd stopped

Veit00, 2023-02-15:
openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 --device esp-eBus.locel:3334 --loglevel=debug --address=ff -f                    [sudo] password for openhabian:
2023-02-15 09:48:10.070 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device esp-eBus.locel:3334
2023-02-15 09:48:10.070 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 09:48:10.308 [main info] reading templates /
2023-02-15 09:48:10.550 [main info] read templates in /
2023-02-15 09:48:10.550 [main info] reading file memory.csv
2023-02-15 09:48:10.792 [main info] successfully read file memory.csv
2023-02-15 09:48:10.792 [main info] reading file broadcast.csv
2023-02-15 09:48:11.037 [main info] successfully read file broadcast.csv
2023-02-15 09:48:11.037 [main info] read config files, got 11 messages
2023-02-15 09:48:11.048 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:11.049 [bus notice] bus started with own address ff/04
2023-02-15 09:48:11.049 [bus notice] device invalid
2023-02-15 09:48:11.049 [main info] registering data handlers
2023-02-15 09:48:11.049 [main info] registered data handlers
2023-02-15 09:48:16.051 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:16.051 [bus notice] device invalid
2023-02-15 09:48:21.050 [main debug] performing regular tasks
2023-02-15 09:48:21.053 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:21.053 [bus notice] device invalid
2023-02-15 09:48:26.056 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:26.056 [bus notice] device invalid
2023-02-15 09:48:31.050 [main debug] performing regular tasks
2023-02-15 09:48:31.058 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:31.058 [bus notice] device invalid
^C2023-02-15 09:48:33.112 [main notice] SIGINT received
2023-02-15 09:48:36.061 [bus error] unable to open esp-eBus.locel:3334: ERR: generic I/O error
2023-02-15 09:48:36.061 [bus notice] device invalid

danielkucera, 2023-02-15:
I am sorry but can't you copy one string? It is wrong again ?? `esp-eBus.locel:3334` ??

Veit00, 2023-02-15:
sorry,
openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 --device esp-eBus.local:3334 --loglevel=debug --address=ff -f
2023-02-15 09:56:06.893 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device esp-eBus.local:3334
2023-02-15 09:56:06.894 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 09:56:07.133 [main info] reading templates /
2023-02-15 09:56:07.371 [main info] read templates in /
2023-02-15 09:56:07.371 [main info] reading file memory.csv
2023-02-15 09:56:07.609 [main info] successfully read file memory.csv
2023-02-15 09:56:07.609 [main info] reading file broadcast.csv
2023-02-15 09:56:07.857 [main info] successfully read file broadcast.csv
2023-02-15 09:56:07.857 [main info] read config files, got 11 messages
2023-02-15 09:56:08.168 [bus notice] bus started with own address ff/04
2023-02-15 09:56:08.168 [main info] registering data handlers
2023-02-15 09:56:08.168 [main info] registered data handlers
2023-02-15 09:56:08.213 [bus debug] ERR: SYN received during no signal, switching to ready
2023-02-15 09:56:08.213 [bus notice] signal acquired
2023-02-15 09:56:08.438 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.529 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.573 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.663 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.754 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.843 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:08.934 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.024 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.114 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.204 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.384 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.429 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.474 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.564 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.745 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.835 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:09.925 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.016 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.105 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.195 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.285 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.378 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.420 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.465 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.556 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.646 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.737 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.826 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:10.916 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.006 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.096 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.186 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.276 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.411 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.907 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:11.997 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.403 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.673 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.763 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.854 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.898 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:12.945 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:13.034 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:13.214 [bus debug] ERR: SYN received during receive command, switching to ready
[... cut ...]
2023-02-15 09:56:22.945 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:23.035 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:23.125 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:23.215 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:23.305 [bus debug] ERR: SYN received during receive command, switching to ready
2023-02-15 09:56:23.306 [main notice] ebusd stopped
openhabian@openhabian:~ $

danielkucera, 2023-02-15:
good, now adjust the potentiometer according to:
https://github.com/danielkucera/esp8266-arduino-ebus#quickstart

and change the port to 3333

Veit00, 2023-02-15:
so in etwa

[image]

Veit00, 2023-02-15:
ebus scann schaut jetzt so aus und die Blaue LED blinkt nur zweimal kurz beim einschalten des Adapters sonst immer aus

openhabian@openhabian:~ $ sudo ebusd --scanconfig --accesslevel=* --latency=200000 --device esp-eBus.local:3333 --loglevel=debug --address=ff -f
[sudo] password for openhabian:
Sorry, try again.
[sudo] password for openhabian:
2023-02-15 10:57:47.640 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with auto scan on device esp-eBus.local:3333
2023-02-15 10:57:47.640 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 10:57:47.874 [main info] reading templates /
2023-02-15 10:57:48.111 [main info] read templates in /
2023-02-15 10:57:48.111 [main info] reading file memory.csv
2023-02-15 10:57:48.347 [main info] successfully read file memory.csv
2023-02-15 10:57:48.347 [main info] reading file broadcast.csv
2023-02-15 10:57:48.586 [main info] successfully read file broadcast.csv
2023-02-15 10:57:48.586 [main info] read config files, got 11 messages
2023-02-15 10:57:48.641 [bus notice] bus started with own address ff/04
2023-02-15 10:57:48.641 [bus debug] ERR: SYN received during no signal, switching to ready
2023-02-15 10:57:48.641 [bus notice] signal acquired
2023-02-15 10:57:48.641 [main info] registering data handlers
2023-02-15 10:57:48.641 [main info] registered data handlers
2023-02-15 10:57:58.641 [main debug] performing regular tasks
2023-02-15 10:58:08.642 [main debug] performing regular tasks
2023-02-15 10:58:18.642 [main debug] performing regular tasks
2023-02-15 10:58:28.642 [main debug] performing regular tasks
2023-02-15 10:58:38.643 [main debug] performing regular tasks
2023-02-15 10:58:48.643 [main debug] performing regular tasks
2023-02-15 10:58:58.643 [main debug] performing regular tasks
2023-02-15 10:59:08.643 [main debug] performing regular tasks
2023-02-15 10:59:18.644 [main debug] performing regular tasks
2023-02-15 10:59:28.644 [main debug] performing regular tasks
2023-02-15 10:59:38.644 [main debug] performing regular tasks
2023-02-15 10:59:48.645 [main debug] performing regular tasks
2023-02-15 10:59:58.645 [main debug] performing regular tasks
2023-02-15 10:59:58.926 [main notice] update check: revision 23.1 available
2023-02-15 11:00:08.927 [main debug] performing regular tasks
2023-02-15 11:00:18.927 [main debug] performing regular tasks
2023-02-15 11:00:28.927 [main debug] performing regular tasks
2023-02-15 11:00:38.928 [main debug] performing regular tasks
2023-02-15 11:00:48.928 [main debug] performing regular tasks
2023-02-15 11:00:58.928 [main debug] performing regular tasks
2023-02-15 11:01:08.929 [main debug] performing regular tasks
2023-02-15 11:01:18.929 [main debug] performing regular tasks
2023-02-15 11:01:28.929 [main debug] performing regular tasks
2023-02-15 11:01:38.930 [main debug] performing regular tasks
2023-02-15 11:01:48.930 [main debug] performing regular tasks
2023-02-15 11:01:58.930 [main debug] performing regular tasks
2023-02-15 11:02:08.930 [main debug] performing regular tasks
2023-02-15 11:02:18.931 [main debug] performing regular tasks
2023-02-15 11:02:28.931 [main debug] performing regular tasks
2023-02-15 11:02:38.931 [main debug] performing regular tasks
2023-02-15 11:02:48.932 [main debug] performing regular tasks
2023-02-15 11:02:58.932 [main debug] performing regular tasks
2023-02-15 11:03:08.932 [main debug] performing regular tasks
2023-02-15 11:03:18.933 [main debug] performing regular tasks
2023-02-15 11:03:28.933 [main debug] performing regular tasks
2023-02-15 11:03:38.933 [main debug] performing regular tasks
2023-02-15 11:03:48.934 [main debug] performing regular tasks
2023-02-15 11:03:58.934 [main debug] performing regular tasks
2023-02-15 11:04:08.934 [main debug] performing regular tasks
2023-02-15 11:04:18.935 [main debug] performing regular tasks
2023-02-15 11:04:28.935 [main debug] performing regular tasks
2023-02-15 11:04:38.935 [main debug] performing regular tasks
2023-02-15 11:04:48.936 [main debug] performing regular tasks
2023-02-15 11:04:58.936 [main debug] performing regular tasks
2023-02-15 11:05:08.936 [main debug] performing regular tasks
2023-02-15 11:05:18.936 [main debug] performing regular tasks
2023-02-15 11:05:28.937 [main debug] performing regular tasks
2023-02-15 11:05:38.937 [main debug] performing regular tasks
2023-02-15 11:05:48.937 [main debug] performing regular tasks
2023-02-15 11:05:58.938 [main debug] performing regular tasks
2023-02-15 11:06:08.938 [main debug] performing regular tasks
2023-02-15 11:06:18.938 [main debug] performing regular tasks
^C2023-02-15 11:06:22.046 [main notice] SIGINT received
2023-02-15 11:06:24.055 [main notice] ebusd stopped
openhabian@openhabian:~ $ ^C

danielkucera, 2023-02-15:
try setting `--scanconfig=full ` and remove `--address=ff`

Veit00, 2023-02-15:
Hallo,
`openhabian@openhabian:~ $ sudo ebusd --scanconfig=full --accesslevel=* --latency=200000 --device esp-eBus.local:3334 --loglevel=debug -f
2023-02-15 15:59:55.096 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with full scan on device esp-eBus.local:3334
2023-02-15 15:59:55.096 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 15:59:55.335 [main info] reading templates /
2023-02-15 15:59:55.579 [main info] read templates in /
2023-02-15 15:59:55.579 [main info] reading file memory.csv
2023-02-15 15:59:55.828 [main info] successfully read file memory.csv
2023-02-15 15:59:55.829 [main info] reading file broadcast.csv
2023-02-15 15:59:56.071 [main info] successfully read file broadcast.csv
2023-02-15 15:59:56.071 [main info] read config files, got 11 messages
2023-02-15 15:59:56.215 [bus notice] bus started with own address 31/36
2023-02-15 15:59:56.215 [main info] registering data handlers
2023-02-15 15:59:56.215 [main info] registered data handlers
2023-02-15 15:59:56.216 [bus debug] ERR: SYN received during no signal, switching to ready
2023-02-15 15:59:56.216 [bus notice] signal acquired
2023-02-15 16:00:06.216 [main debug] performing regular tasks
2023-02-15 16:00:06.216 [main notice] starting initial full scan
2023-02-15 16:00:06.216 [bus info] scan 02 cmd: 3102070400
2023-02-15 16:00:06.261 [bus debug] start request 31
2023-02-15 16:00:06.261 [bus debug] arbitration start with 31
2023-02-15 16:00:06.351 [bus debug] arbitration lost
2023-02-15 16:00:06.351 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.351 [bus debug] start request 31
2023-02-15 16:00:06.351 [bus debug] arbitration start with 31
2023-02-15 16:00:06.441 [bus debug] arbitration lost
2023-02-15 16:00:06.441 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.441 [bus debug] start request 31
2023-02-15 16:00:06.441 [bus debug] arbitration start with 31
2023-02-15 16:00:06.533 [bus debug] arbitration lost
2023-02-15 16:00:06.533 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.533 [bus debug] start request 31
2023-02-15 16:00:06.533 [bus debug] arbitration start with 31
2023-02-15 16:00:06.621 [bus debug] arbitration lost
2023-02-15 16:00:06.621 [bus debug] notify request: ERR: arbitration lost
2023-02-15 16:00:06.621 [bus info] scan 04 cmd: 3104070400
2023-02-15 16:00:06.621 [bus debug] start request 31
2023-02-15 16:00:06.621 [bus debug] arbitration start with 31
2023-02-15 16:00:06.711 [bus debug] arbitration lost
2023-02-15 16:00:06.711 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.711 [bus debug] start request 31
2023-02-15 16:00:06.711 [bus debug] arbitration start with 31
2023-02-15 16:00:06.801 [bus debug] arbitration lost
2023-02-15 16:00:06.801 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.801 [bus debug] start request 31
2023-02-15 16:00:06.801 [bus debug] arbitration start with 31
2023-02-15 16:00:06.892 [bus debug] arbitration lost
2023-02-15 16:00:06.892 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:06.892 [bus debug] start request 31
2023-02-15 16:00:06.892 [bus debug] arbitration start with 31
2023-02-15 16:00:06.981 [bus debug] arbitration lost
2023-02-15 16:00:06.981 [bus debug] notify request: ERR: arbitration lost
2023-02-15 16:00:06.981 [bus info] scan 05 cmd: 3105070400
2023-02-15 16:00:06.981 [bus debug] start request 31
2023-02-15 16:00:06.981 [bus debug] arbitration start with 31
2023-02-15 16:00:07.072 [bus debug] arbitration lost
2023-02-15 16:00:07.072 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:07.072 [bus debug] start request 31
2023-02-15 16:00:07.072 [bus debug] arbitration start with 31
2023-02-15 16:00:07.162 [bus debug] arbitration lost
2023-02-15 16:00:07.162 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:07.162 [bus debug] start request 31
2023-02-15 16:00:07.162 [bus debug] arbitration start with 31
2023-02-15 16:00:07.252 [bus debug] arbitration lost
2023-02-15 16:00:07.252 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:07.252 [bus debug] start request 31
2023-02-15 16:00:07.252 [bus debug] arbitration start with 31
2023-02-15 16:00:07.342 [bus debug] arbitration lost
2023-02-15 16:00:07.342 [bus debug] notify request: ERR: arbitration lost
2023-02-15 16:00:07.342 [bus info] scan 06 cmd: 3106070400
[... 168 lines cut ...]
2023-02-15 16:00:10.948 [bus debug] arbitration lost
2023-02-15 16:00:10.949 [bus debug] notify request: ERR: arbitration lost
2023-02-15 16:00:10.949 [bus info] scan 15 cmd: 3115070400
2023-02-15 16:00:10.949 [bus debug] start request 31
2023-02-15 16:00:10.949 [bus debug] arbitration start with 31
2023-02-15 16:00:11.036 [bus debug] arbitration lost
2023-02-15 16:00:11.036 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:11.036 [bus debug] start request 31
2023-02-15 16:00:11.036 [bus debug] arbitration start with 31
2023-02-15 16:00:11.127 [bus debug] arbitration lost
2023-02-15 16:00:11.127 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:11.127 [bus debug] start request 31
2023-02-15 16:00:11.127 [bus debug] arbitration start with 31
2023-02-15 16:00:11.217 [bus debug] arbitration lost
2023-02-15 16:00:11.217 [bus debug] ERR: arbitration lost during ready, retry
2023-02-15 16:00:11.217 [bus debug] start request 31
2023-02-15 16:00:11.217 [bus debug] arbitration start with 31
2023-02-15 16:00:11.307 [bus debug] arbitration lost
2023-02-15 16:00:11.307 [bus debug] notify request: ERR: arbitration lost
2023-02-15 16:00:11.307 [bus info] scan 16 cmd: 3116070400
2023-02-15 16:00:11.307 [bus debug] start request 31
2023-02-15 16:00:11.307 [bus debug] arbitration start with 31
2023-02-15 16:00:11.355 [main notice] ebusd stopped
openhabian@openhabian:~ $
`

Veit00, 2023-02-15:
Sorry, und nochmal mit Port 3333
`openhabian@openhabian:~ $ sudo ebusd --scanconfig=full --accesslevel=* --latency=200000 --device esp-eBus.local:3333 --loglevel=debug -f
2023-02-15 16:04:58.892 [main notice] ebusd 23.1.23.1-8-g9c0af7b9 started with full scan on device esp-eBus.local:3333
2023-02-15 16:04:58.892 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-02-15 16:04:59.129 [main info] reading templates /
2023-02-15 16:04:59.368 [main info] read templates in /
2023-02-15 16:04:59.368 [main info] reading file memory.csv
2023-02-15 16:04:59.611 [main info] successfully read file memory.csv
2023-02-15 16:04:59.611 [main info] reading file broadcast.csv
2023-02-15 16:04:59.857 [main info] successfully read file broadcast.csv
2023-02-15 16:04:59.857 [main info] read config files, got 11 messages
2023-02-15 16:05:00.005 [bus notice] bus started with own address 31/36
2023-02-15 16:05:00.005 [main info] registering data handlers
2023-02-15 16:05:00.005 [main info] registered data handlers
2023-02-15 16:05:00.008 [bus debug] ERR: SYN received during no signal, switching to ready
2023-02-15 16:05:00.009 [bus notice] signal acquired
2023-02-15 16:05:10.006 [main debug] performing regular tasks
2023-02-15 16:05:10.006 [main notice] starting initial full scan
2023-02-15 16:05:10.006 [bus info] scan 02 cmd: 3102070400
2023-02-15 16:05:10.007 [bus debug] start request 31
2023-02-15 16:05:10.007 [bus debug] arbitration start with 31
2023-02-15 16:05:10.062 [bus debug] arbitration won
2023-02-15 16:05:10.062 [bus debug] arbitration delay 10 micros
2023-02-15 16:05:10.062 [bus info] arbitration delay 10 - 10 micros
2023-02-15 16:05:10.062 [bus debug] switching from ready to send command
2023-02-15 16:05:10.070 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.070 [bus info] send/receive symbol latency 7 - 7 ms
2023-02-15 16:05:10.078 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.085 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.093 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.093 [bus debug] switching from send command to send command CRC
2023-02-15 16:05:10.103 [bus debug] send/receive symbol latency 10 ms
2023-02-15 16:05:10.103 [bus info] send/receive symbol latency 7 - 10 ms
2023-02-15 16:05:10.103 [bus debug] switching from send command CRC to receive command ACK
2023-02-15 16:05:10.369 [bus debug] notify request: ERR: read timeout
2023-02-15 16:05:10.369 [bus info] scan 04 cmd: 3104070400
2023-02-15 16:05:10.369 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-02-15 16:05:10.369 [bus debug] start request 31
2023-02-15 16:05:10.369 [bus debug] arbitration start with 31
2023-02-15 16:05:10.627 [bus debug] arbitration won
2023-02-15 16:05:10.627 [bus debug] arbitration delay 12 micros
2023-02-15 16:05:10.627 [bus info] arbitration delay 10 - 12 micros
2023-02-15 16:05:10.627 [bus debug] switching from ready to send command
2023-02-15 16:05:10.635 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.643 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.651 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.658 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.658 [bus debug] switching from send command to send command CRC
2023-02-15 16:05:10.666 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:10.666 [bus debug] switching from send command CRC to receive command ACK
2023-02-15 16:05:10.931 [bus debug] notify request: ERR: read timeout
2023-02-15 16:05:10.931 [bus info] scan 05 cmd: 3105070400
2023-02-15 16:05:10.932 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-02-15 16:05:10.932 [bus debug] start request 31
2023-02-15 16:05:10.932 [bus debug] arbitration start with 31
2023-02-15 16:05:11.190 [bus debug] arbitration won
2023-02-15 16:05:11.190 [bus debug] arbitration delay 10 micros
2023-02-15 16:05:11.190 [bus debug] switching from ready to send command
2023-02-15 16:05:11.200 [bus debug] send/receive symbol latency 9 ms
2023-02-15 16:05:11.209 [bus debug] send/receive symbol latency 8 ms
2023-02-15 16:05:11.216 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:11.224 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:11.224 [bus debug] switching from send command to send command CRC
2023-02-15 16:05:11.231 [bus debug] send/receive symbol latency 7 ms
2023-02-15 16:05:11.231 [bus debug] switching from send command CRC to receive command ACK
2023-02-15 16:05:11.497 [bus debug] notify request: ERR: read timeout
2023-02-15 16:05:11.497 [bus info] scan 06 cmd: 3106070400
2023-02-15 16:05:11.497 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-02-15 16:05:11.497 [bus debug] start request 31
2023-02-15 16:05:11.497 [bus debug] arbitration start with 31
[... 237 lines cut ...]
2023-02-15 16:05:19.245 [bus debug] send/receive symbol latency 6 ms
2023-02-15 16:05:19.245 [bus debug] switching from send command CRC to receive command ACK
2023-02-15 16:05:19.510 [bus debug] notify request: ERR: read timeout
2023-02-15 16:05:19.510 [bus info] scan 1b cmd: 311b070400
2023-02-15 16:05:19.511 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-02-15 16:05:19.511 [bus debug] start request 31
2023-02-15 16:05:19.511 [bus debug] arbitration start with 31
2023-02-15 16:05:19.769 [bus debug] arbitration won
[... cut ...]
2023-02-15 16:05:19.804 [bus debug] switching from send command CRC to receive command ACK
2023-02-15 16:05:20.069 [bus debug] notify request: ERR: read timeout
2023-02-15 16:05:20.070 [bus info] scan 1c cmd: 311c070400
2023-02-15 16:05:20.070 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-02-15 16:05:20.070 [bus debug] start request 31
2023-02-15 16:05:20.070 [bus debug] arbitration start with 31
2023-02-15 16:05:20.321 [main notice] ebusd stopped
openhabian@openhabian:~ $
`

danielkucera, 2023-02-15:
This is correct now, ebusd has found your device:
2023-02-15 16:05:17.472 [update notice] sent scan-read scan.15  QQ=31: Vaillant;SDR_P;0312;6801

Ignore the other read timeouts, it only scan all addresses.

Veit00, 2023-02-15:
danke schön,
und sorry dafür das ich mich manchmal blöd anstelle.

wissen sie zufällig was ich jetzt in openhab einstellen muss in der Bridge

Freundliche Grüsse
Veit