danielkucera/esp-ebus #10: [main error] scan config 08: ERR: arbitration lost

linoul, 2022-12-26:
I make v4.1 ebus-adapter to connect to the vaillant turboTEC pro, but the ebusd can't runing. where is the problem?
```
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=Yu7pheeKutei8ahv0aen3OmaiQuoePie7sooYeeShae3zooghoKah3Ash3och5Ei --scanconfig --mqttjson --latency=10000 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.2.56:3333 --log=all:notice
2022-12-26 16:11:28.718 [main notice] ebusd 22.4.v3.16.2-200-ga8ca21f541 started with auto scan on device 192.168.2.56:3333
2022-12-26 16:11:37.325 [bus notice] bus started with own address 31/36
2022-12-26 16:11:37.325 [bus notice] signal acquired
2022-12-26 16:11:37.328 [mqtt notice] connection established
2022-12-26 16:13:48.860 [main notice] update check: revision v22.4 available
2022-12-26 16:14:41.321 [bus notice] new master 03, master count 2
2022-12-26 16:14:53.440 [main error] scan config 08: ERR: arbitration lost
2022-12-26 16:20:11.447 [main error] scan config 08: ERR: arbitration lost
2022-12-26 16:25:00.592 [main error] scan config 08: ERR: arbitration lost
```

linoul, 2023-01-03:
It looks like the adapter can't write to the bus
```
 ebusd --lograwdata --scanconfig=full -f -d 192.168.2.42:3333 --latency=200000

2023-01-03 09:06:37.274 [main notice] ebusd 22.4.v22.4 started with full scan on device 192.168.2.42:3333
2023-01-03 09:06:41.569 [bus notice] bus started with own address 31/36
2023-01-03 09:06:41.588 [bus notice] signal acquired
2023-01-03 09:06:51.570 [main notice] starting initial full scan
2023-01-03 09:06:51.673 [bus notice] >31
2023-01-03 09:06:51.713 [bus notice] >31
2023-01-03 09:06:51.750 [bus notice] >31
2023-01-03 09:07:12.025 [bus notice] new master 03, master count 2
2023-01-03 09:07:12.535 [bus notice] >31<0306b5090228023e
2023-01-03 09:07:12.573 [bus notice] >31
2023-01-03 09:07:12.616 [bus notice] >31
2023-01-03 09:07:12.664 [bus notice] >31
2023-01-03 09:07:12.703 [bus notice] >31
2023-01-03 09:07:12.747 [bus notice] >31
2023-01-03 09:07:12.790 [bus notice] >31
2023-01-03 09:07:12.836 [bus notice] >31
2023-01-03 09:07:12.875 [bus notice] >31
2023-01-03 09:07:12.919 [bus notice] >31
2023-01-03 09:07:12.962 [bus notice] >31
2023-01-03 09:07:13.009 [bus notice] >31
2023-01-03 09:07:13.050 [bus notice] >31
2023-01-03 09:07:13.093 [bus notice] >31
2023-01-03 09:07:13.631 [bus notice] >31<0306b5090228023e
2023-01-03 09:07:13.674 [bus notice] >31
2023-01-03 09:07:13.723 [bus notice] >31
2023-01-03 09:07:13.764 [bus notice] >31
2023-01-03 09:07:13.805 [bus notice] >31
2023-01-03 09:07:13.849 [bus notice] >31
2023-01-03 09:07:13.894 [bus notice] >31
2023-01-03 09:07:13.937 [bus notice] >31
2023-01-03 09:07:13.977 [bus notice] >31
2023-01-03 09:07:14.025 [bus notice] >31
2023-01-03 09:07:14.065 [bus notice] >31
2023-01-03 09:07:14.107 [bus notice] >31
2023-01-03 09:07:14.151 [bus notice] >31
2023-01-03 09:07:14.200 [bus notice] >31
2023-01-03 09:07:14.735 [bus notice] >31<0306b5090228023e
2023-01-03 09:11:21.195 [main error] scan config 08: ERR: arbitration lost
2023-01-03 09:11:21.728 [bus notice] >31<0306b5090228023e
2023-01-03 09:11:21.766 [bus notice] >31
```

danielkucera, 2023-01-03:
Hi @linoul ,
can you please make a video of the adapter after starting the scan and send to https://upload.danman.eu ?
Let me know when uploaded.

linoul, 2023-01-03:
Thanks for the reply, the video name ebusdscan.MP4 has been uploaded

danielkucera, 2023-01-03:
This doesn't look like an adapter I have made:

[image]

Where did you get it from?

linoul, 2023-01-04:
i made it from your project

danielkucera, 2023-01-04:
Then some of the components has to be damaged - zener diode, transistor, ...

21494, 2023-01-14:
I get similar problem but I have the adapter that you sent me.

Following errors after config in Home Assistant:

```
ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=piephieb6eenoothied2eeNaisae9aeph2Thaing9moo8aih3uwuoGi1aiPhooze --scanconfig --mqttjson --latency=199 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.19:3333 --log=all:notice --log=bus:notice --log=network:notice
2023-01-14 12:23:06.320 [main notice] ebusd 23.1.p20230106 started with auto scan on device 192.168.1.19:3333
2023-01-14 12:23:09.727 [bus notice] bus started with own address 31/36
2023-01-14 12:23:09.728 [bus notice] signal acquired
2023-01-14 12:23:10.568 [bus notice] new master 03, master count 2
2023-01-14 12:23:12.097 [mqtt notice] connection established
2023-01-14 12:23:12.596 [bus notice] new master 10, master count 3
2023-01-14 12:23:12.622 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:23:17.604 [update notice] received unknown MS cmd: 1008b5110101 / 0946480012ffff0000ff
2023-01-14 12:23:20.143 [main error] scan config 08: ERR: arbitration lost
2023-01-14 12:23:22.495 [main error] scan config 15: ERR: arbitration lost
2023-01-14 12:23:22.577 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:23:27.563 [update notice] received unknown MS cmd: 1008b5110101 / 0946483012ffff0000ff
2023-01-14 12:23:29.556 [update notice] received unknown BC cmd: 10feb516080020291214010423
2023-01-14 12:23:32.602 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:23:34.865 [main error] scan config 08: ERR: arbitration lost
2023-01-14 12:23:37.229 [main error] scan config 15: ERR: arbitration lost
2023-01-14 12:23:37.580 [update notice] received unknown MS cmd: 1008b5110101 / 0946483012ffff0000ff
2023-01-14 12:23:39.577 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff7012
2023-01-14 12:23:39.830 [update notice] received unknown MS cmd: 1008b5110102 / 050214964c78
2023-01-14 12:23:42.569 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:23:47.597 [update notice] received unknown MS cmd: 1008b5110101 / 0946487012ffff0000ff
2023-01-14 12:23:49.516 [update notice] received unknown BC cmd: 10feb5160301700e
2023-01-14 12:23:49.539 [main error] scan config 08: ERR: arbitration lost
2023-01-14 12:23:51.904 [main error] scan config 15: ERR: arbitration lost
2023-01-14 12:23:52.559 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:23:57.568 [update notice] received unknown MS cmd: 1008b5110101 / 094648a012ffff0000ff
2023-01-14 12:23:59.529 [update notice] received unknown MS cmd: 1008b5110102 / 050214964c78
2023-01-14 12:24:02.648 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-14 12:24:04.284 [main error] scan config 08: ERR: arbitration lost
2023-01-14 12:24:06.644 [main error] scan config 15: ERR: arbitration lost
```
I tried with different latencies up to 10000 ms, which is the maximum that allows me to configure

Thanks

danielkucera, 2023-01-14:
Is your device supported by ebusd?

21494, 2023-01-14:
I guess. I saw comments from people with the same model working with ebusd for example in this comment (https://github.com/john30/ebusd-configuration/issues/276#issue-1406832744) someone asks about the same model and thermostat that I have. In his case ebusd detects it as vaillant device. 
"arbitration lost" sounds like a hardware problem but I don't know...

danielkucera, 2023-01-15:
Can you please try with the option `--lograwdata` ?

21494, 2023-01-15:
Log Output: 
```
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=piephieb6eenoothied2eeNaisae9aeph2Thaing9moo8aih3uwuoGi1aiPhooze --scanconfig --mqttjson --configpath=/config/ebusd-configuration-master/ebusd-2.1.x/en/ --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.7:3333 --log=all:debug --log=bus:debug --log=network:debug --latency 200000 --lograwdata
2023-01-15 22:39:24.595 [main notice] ebusd 23.1.p20230106 started with auto scan on device 192.168.1.7:3333
2023-01-15 22:39:24.605 [main info] loading configuration files from /config/ebusd-configuration-master/ebusd-2.1.x/en/
2023-01-15 22:39:24.605 [main debug] reading directory /config/ebusd-configuration-master/ebusd-2.1.x/en/
2023-01-15 22:39:24.607 [main debug] file type of /config/ebusd-configuration-master/ebusd-2.1.x/en/vaillant is dir
2023-01-15 22:39:24.607 [main debug] file type of /config/ebusd-configuration-master/ebusd-2.1.x/en/memory.csv is file
2023-01-15 22:39:24.607 [main debug] file type of /config/ebusd-configuration-master/ebusd-2.1.x/en/broadcast.csv is file
2023-01-15 22:39:24.607 [main debug] file type of /config/ebusd-configuration-master/ebusd-2.1.x/en/_templates.csv is file
2023-01-15 22:39:24.608 [main info] reading templates /
2023-01-15 22:39:24.613 [main info] read templates in /
2023-01-15 22:39:24.614 [main info] reading file memory.csv
2023-01-15 22:39:24.620 [main info] successfully read file memory.csv
2023-01-15 22:39:24.620 [main info] reading file broadcast.csv
2023-01-15 22:39:24.627 [main info] successfully read file broadcast.csv
2023-01-15 22:39:24.627 [main info] read config files, got 11 messages
2023-01-15 22:39:24.791 [bus notice] bus started with own address 31/36
2023-01-15 22:39:24.793 [main info] registering data handlers
2023-01-15 22:39:24.793 [mqtt info] mosquitto version 2.0.15 (compiled with 2.0.15)
2023-01-15 22:39:24.813 [bus debug] ERR: SYN received during no signal, switching to ready
2023-01-15 22:39:24.813 [bus notice] signal acquired
2023-01-15 22:39:25.045 [mqtt debug] connection requested
2023-01-15 22:39:25.045 [main info] registered data handlers
2023-01-15 22:39:25.151 [mqtt notice] connection established
2023-01-15 22:39:25.152 [mqtt debug] publish ebusd/global/version "ebusd 23.1.p20230106"
2023-01-15 22:39:25.152 [mqtt debug] publish ebusd/global/running true
2023-01-15 22:39:25.954 [bus notice] new master 10, master count 2
2023-01-15 22:39:25.981 [bus notice] new master 03, master count 3
2023-01-15 22:39:25.981 [update info] received MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-15 22:39:25.982 [update notice] received unknown MS cmd: 1008b510090000005effff01ff00 / 0101
2023-01-15 22:39:25.986 [bus notice] <1008b510090000005effff01ff00410001019a00
2023-01-15 22:39:30.981 [update info] received MS cmd: 1008b5110101 / 0958579007ffff0000ff
2023-01-15 22:39:30.981 [update notice] received unknown MS cmd: 1008b5110101 / 0958579007ffff0000ff
2023-01-15 22:39:30.984 [bus notice] <1008b511010189000958579007ffff0000ffc100
[...]
2023-01-15 22:39:50.089 [bus notice] >31
2023-01-15 22:39:50.089 [bus debug] arbitration lost
2023-01-15 22:39:50.089 [bus debug] ERR: arbitration lost during ready, retry
2023-01-15 22:39:50.089 [bus debug] start request 31
2023-01-15 22:39:50.089 [bus debug] arbitration start with 31
2023-01-15 22:39:50.170 [bus notice] >31
2023-01-15 22:39:50.170 [bus debug] arbitration lost
2023-01-15 22:39:50.170 [bus debug] notify request: ERR: arbitration lost
2023-01-15 22:39:50.170 [main error] scan config 08: ERR: arbitration lost
2023-01-15 22:39:50.170 [mqtt debug] publish ebusd/global/scan "running"
2023-01-15 22:39:50.952 [update info] received MS cmd: 1008b5110101 / 0957579007ffff0000ff
2023-01-15 22:39:50.953 [update notice] received unknown MS cmd: 1008b5110101 / 0957579007ffff0000ff
2023-01-15 22:39:50.958 [bus notice] <1008b511010189000957579007ffff0000ff0d00
```
According with https://cadlab.io/project/25252/master/circuit/d2lyaW5nLmtpY2FkX3NjaA%3D%3D
- I measured on the "TX-DISABLE" net in J2 and it is at ~2.1V even when blue led is ON.  It should not be disabled to transmit? 
- Eventually LED D5 lights up briefly 2 times every 5-10 seconds.
- LED D1 blink constantly.
- Voltage measured in the zener with multimeter is 0V.

Thanks for your help

21494, 2023-01-16:
Ok, I have temporarily solved it by removing resistor R17 so that Tx is always enabled. Do you know why this may be happening?

danielkucera, 2023-01-16:
Which HW version you have? Did you flash some firmware?

21494, 2023-01-16:
I think the hw version is 4. and I flashed the last version I saw in releases: 4.2

danielkucera, 2023-01-16:
which file exactly?
[image]

21494, 2023-01-16:
-v3.0.bin 
I had doubts about which one to use

danielkucera, 2023-01-16:
That's the problem. You need to use firmware.bin - that is for HW v4.0+ , I'll try to make it more obvious

21494, 2023-01-16:
Jajaja ahh ok
I'm going to restore the fix and update to the other firmware. I guess it will work. Thanks Daniel =)