danielkucera/esp-ebus #11: Home Assistant integration

Elijen, 2022-12-29:
What is the recommended way of integrating the ebus adapter into HA? I didn't find a way to configure MQTT on the device so I suppose I have to use another ebusd instance on the HA server to communicate with ebusd on the device?

Elijen, 2022-12-29:
I have installed ebusd add-on and set the adapter as network device and it works ok. Although it seems the (read) values take quite a long time to refresh over MQTT. I wonder if this is the best solution or if there is a better way.

danielkucera, 2022-12-30:
Hi @Elijen , yes, this is probably the best solution. How long is "quite long" ?

Elijen, 2022-12-30:
It can take 30 minutes or longer to get an update of a specific value. Do you know how long does it take ebus to get all the values from the heat pump? I haven't figured out how to refresh a specific value manually via MQTT yet so I am relying on the polling interval which is set to 5s meaning it refreshes all the parameters one by one I think (but I am not sure).

danielkucera, 2022-12-30:
Hi, it seems this has nothing to do with the adapter. 
I haven't been using the mqtt protocol so I cannot help here. 
Try to check the ebusd documentation.

Elijen, 2022-12-30:
@danielkucera Yup, you are right. Thanks anyway. The adapter is really great 👍

bf8392, 2023-04-26:
Can you give a more detailed description on how to integrate with home assistant and secure the device in the network? So nobody else can add values except the connected HA? Maybe Password?