danielkucera/esp-ebus #13: Adapter not working when connected to slave device

mwullink, 2023-01-02:
The adapter functions perfect when it is connected to the Brink Renovent Excellent 300 when this device is running as ebus master.
But the adapter does not power up when i set the Brink Renovent Excellent 300 to be a slave (which is better because otherwise all ebus command will be reverted after a few seconds by the device)

now i'm running the  Brink Renovent Excellent as master and have to send to fanspeed command every 10 seconds to make sure the fanspeed value does not revert to default value.

what might be the issue here with the adapter, is this by design?

otherwise the adapter is awesome!

device: https://www.brinkclimatesystems.nl/documenten/renovent-excellent-300-400-612010.pdf

danielkucera, 2023-01-03:
Hi @mwullink ,
this is an interesting behavior but it makes sense - only master should power the bus.

I can see in the guide there is 24V "Aansluiting" . 
[image]

Is it input or output? 
Can you confirm there is 24V on those terminals and try to connect pin 6 to adapter TP2 test point?

mwullink, 2023-01-03:
Unfortunately the X15 connector is only available on the Plus version of the Brink Revovent.
Could i also use something like this?
https://www.tinytronics.nl/shop/nl/power/voedingen/24v/mean-well-voeding-24v-0.625a-switching-power-supply-rs-15-24

danielkucera, 2023-01-03:
You can try to connect any DC power adapter with output 5V-26V between TP2 (+) and TP3 (-).

mwullink, 2023-01-04:
i tried powering the adapter with my breadboard power supply (5v) the red led's turn on and the blue wifi led was also blinking intermittendly. But the adapter was not getting an ip-address, i could not connect to it or ping it. The adapter also did not start its own wifi accesspoint.

wat am i doing wrong here?

danielkucera, 2023-01-04:
Right, now I realized this won't work probably. 
The bus is not powered this way so there cannot be any communication.
You would need to properly power the bus somehow. 
See page 23 here: https://ebus-wiki.org/lib/exe/fetch.php/ebus/spec_prot_12_v1_3_1_e.pdf

mwullink, 2023-01-05:
my electrical engineering skills are very limited, i have no idea on howto power the bus.
i guess i will just keep running it with home assistant sending the fanspeed command every 10 seconds.
thx for your help.