danielkucera/esp-ebus #14: Questions about the code base

ysard, 2023-01-06:
Hi, I recently discovered the existence of ebus, and its interfaces.
Yours have the great advantage to be totally free and open source (hardware + software); vou saved me a lot of time which allowed me to go directly to the essential : the reverse engineering of the protocol (https://github.com/ysard/ebusd_configuration_chaffoteaux_bridgenet).

In short, I would like to submit some "improvements/optimizations" but some questions remain about the code.

- What is actually the purpose of the code related to the define `TX_DISABLE_PIN` ?
Apparently a previous hardware version referred to it, but this is no longer the case.
I have trouble figuring out how it worked.
Also, you seem to be overriding the option here:
https://github.com/danielkucera/esp8266-arduino-ebus/blob/958e08e2fbf62cc756d7ef32ec4a2ed48b563121/src/main.cpp#L145-L146

- There is a call to `printf` here:
https://github.com/danielkucera/esp8266-arduino-ebus/blob/958e08e2fbf62cc756d7ef32ec4a2ed48b563121/src/main.cpp#L65

Can you tell me more about what this does? I think this is the first time I see it in the Arduino ecosystem without additional code (see https://forum.arduino.cc/t/printf-on-arduino/888528/20).

Thanks ;)

danielkucera, 2023-01-06:
Hi @ysard ,
I'll be happy to accept any contributions.

- ESP8266 spits out data on UART right after boot and there is no way (AFAIK) to stop it. So I addedd TX_DISABLE which will prevent the data from reaching ebus. When it's high (1), you can write anything to Serial and it will not reach the ebus. You need to pull the pin low to open the path.
- printf works by writing to debug output. You don't need to use Serial1.print(...) but you can configure debug output to Serial1 by `Serial1.setDebugOutput(true);` and then just use `printf`

ysard, 2023-01-06:
Understood, TX_DISABLE is related to your interface design.

Ok for printf, I didn't know that ESP libraries allowed that as easily.

Some thoughts here:
https://github.com/danielkucera/esp8266-arduino-ebus/blob/958e08e2fbf62cc756d7ef32ec4a2ed48b563121/src/main.cpp#L64-L67

I will add a `WiFiManager wifiManager(Serial1);` or `wifiManager.setDebugOutput(false);` here,
because by default WiFiManager uses Serial and not Serial1 to send it's debug data (See [here](https://github.com/tzapu/WiFiManager#debug)). Also the printf is redundant with the already debug sent by WiFiManager...

Or maybe I can use a global DEBUG define to switch on/off this kind of behavior ?

ysard, 2023-01-06:
Still about the TX_DISABLE:
https://github.com/danielkucera/esp8266-arduino-ebus/blob/958e08e2fbf62cc756d7ef32ec4a2ed48b563121/src/main.cpp#L79-L80

Using digitalWrite before pinMode seems to be non effective on Arduino.
But even with pinMode, *then* digitalwrite, maybe there will be a glitch that could be resolved with an extra lib (digitalWriteFast for example) or via a register configuration.
See https://community.particle.io/t/can-i-not-call-digitalwrite-before-pinmode/49920/3

danielkucera, 2023-01-06:
> Using digitalWrite before pinMode seems to be non effective on Arduino.

I don't have this experience, it works for me okay. Maybe it is different for ESP8266 arduino than "standard" arduino?

ysard, 2023-01-06:
@danielkucera 
yes I' will not touch it.
I'm reading the implementation for ESP; It's much simpler than on Arduino, but I haven't enough experience here.
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring_digital.cpp

danielkucera, 2023-01-06:
Feel free to submit your PRs, with smallest changes if possible. We can discuss there.

closed 2023-01-06 by ysard