danielkucera/esp-ebus #22: Frequent Arbitration lost errors on writing on the Bus

wrongisthenewright, 2023-01-22:
Hi Daniel,
I'm using your adapter for some month now and in general it works very well. The problem I'm facing is related to frequent errors I get using the adapter to send command on the bus (with ebusdctl read or ebusdctl hex commands) , almost 60% of commands cause some kind of error, the most frequent is aribitration lost.
My ebusd latency is set to 20000.

I'm using the adapter with an Ariston system, It's complex, with boiler, hp, energy manager and remote control unit all wired on the bus. 
Can it be a "bus saturation" problem?

Retrying the command almost always get the correct result.

I've set the potentiometer multiple times, and reading is good so I don't think it's an "electric" problem

wrongisthenewright, 2023-02-03:
With the new adapter (v4.1?) the arbitration lost is much less present, I think now it's caused by my wifi latency as the adapter is far from my AP and with ping time average of 40ms.

danielkucera, 2023-02-03:
I am working on enhanced protocol implementation which should handle the arbitration well even on networks with higher latency. In several weeks it should be available.

HiQual, 2023-02-04:
I hope the enh: protocol will be available for all hardware versions.

danielkucera, 2023-02-04:
Don't worry, it's just a software so yes.

danielkucera, 2023-03-13:
I have just release v5.0 with enh protocol, please test and let me know if it works better now.

HiQual, 2023-03-14:
Thank you for the notification Daniel.  I will try it.
A few questions first;
1. Is it OK to go direct from v3 to v5.
2. Is v5 OK for v2 boards and,
3. Is it still OK to use espota.exe upgrade method from PC to a live running adapter ?

danielkucera, 2023-03-14:
1. yes
2. no, you need HW_v3.x
3. yes

HiQual, 2023-03-14:
[image]
Update using espota.exe fails.
Windows Firewall is disabled.
Python is running.
ebusd is stopped.
Failed also using adapter IP address.
Any ideas ?

danielkucera, 2023-03-14:
Make sure you have the correct file. If it still doesn't work, try this:
https://github.com/danielkucera/esp8266-arduino-ebus#web-interface

danielkucera, 2023-03-14:
which is the last version you installed? you can also disable wifi so the portal appears without resetting

HiQual, 2023-03-14:
Very sorry but I don't know the last version I installed!
By "disable wifi" do you mean on my home router, so that the adapter has nothing to connect to?

HiQual, 2023-03-14:
OK managed to update - many thanks.
Tested without "enh:" in configuration file - OK.
Tested with "enh:" - result below ...

root@maxair:~# ebusctl info
version: ebusd 23.1.23.1
device: esp-ebus.local:3333, enhanced
access: *
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

danielkucera, 2023-03-14:
from readme:
```
3335 - port with enhanced protocol, ebusd config: -d enh:esp-ebus.local:3335
```

HiQual, 2023-03-14:
Sorry about that oversight.
Pleased to report it is up and running.
I am now getting a lot of "Arbitration won" messages which I never received b4.  Are these significant?

2023-03-14 17:07:09.531 [update notice] sent read bai ReturnTemp QQ=31: 26.69;65108;ok
2023-03-14 17:07:09.884 [bus error] send to 08: ERR: read timeout, retry
2023-03-14 17:07:10.319 [bus notice] arbitration won in invalid state skip
2023-03-14 17:07:10.570 [bus error] send to 08: ERR: read timeout, retry
2023-03-14 17:07:10.988 [bus notice] arbitration won in invalid state skip
2023-03-14 17:07:11.253 [update notice] sent read bai ModulationTempDesired QQ=31: 20.0
2023-03-14 17:07:11.515 [update notice] sent read bai Flame QQ=31: off
2023-03-14 17:07:11.838 [update notice] sent read bai Statenumber QQ=31: 30
2023-03-14 17:07:27.096 [bus error] send to 08: ERR: wrong symbol received, retry
2023-03-14 17:07:27.680 [bus notice] arbitration won in invalid state skip
2023-03-14 17:07:27.995 [update notice] sent read bai FlowTemp QQ=31: 27.38;ok
2023-03-14 17:07:28.424 [update notice] sent read bai ReturnTemp QQ=31: 26.69;65108;ok
2023-03-14 17:07:28.728 [update notice] sent read bai ModulationTempDesired QQ=31: 20.0
2023-03-14 17:07:28.988 [update notice] sent read bai Flame QQ=31: off
2023-03-14 17:07:29.181 [bus error] send to 08: ERR: SYN received, retry
2023-03-14 17:07:29.273 [bus notice] arbitration won in invalid state receive command
2023-03-14 17:07:29.553 [update notice] sent read bai Statenumber QQ=31: 30
2023-03-14 17:07:44.785 [bus notice] arbitration won while handling another request
2023-03-14 17:07:45.162 [bus error] send to 08: ERR: read timeout, retry
2023-03-14 17:07:45.590 [bus notice] arbitration won in invalid state skip
2023-03-14 17:07:45.820 [update notice] sent read bai FlowTemp QQ=31: 27.38;ok
2023-03-14 17:07:46.167 [update notice] sent read bai ReturnTemp QQ=31: 26.69;65108;ok
2023-03-14 17:07:46.465 [update notice] sent read bai ModulationTempDesired QQ=31: 20.0
2023-03-14 17:07:46.771 [update notice] sent read bai Flame QQ=31: off
2023-03-14 17:07:47.085 [update notice] sent read bai Statenumber QQ=31: 30
2023-03-14 17:07:52.425 [main error] scan config 05: ERR: read timeout
2023-03-14 17:08:02.795 [bus notice] arbitration won in invalid state skip
2023-03-14 17:08:03.205 [update notice] sent read bai FlowTemp QQ=31: 27.38;ok
2023-03-14 17:08:03.884 [bus notice] arbitration won in invalid state skip
2023-03-14 17:08:04.506 [bus error] send to 08: ERR: read timeout, retry
2023-03-14 17:08:04.944 [bus notice] arbitration won in invalid state skip

danielkucera, 2023-03-14:
what is the full command line?

HiQual, 2023-03-15:
> what is the full command line?

EBUSD_OPTS="--scanconfig=08 -d enh:esp-ebus.local:3335 --latency=90000 --accesslevel=*"

HiQual, 2023-03-15:
I removed the latency yesterday evening and left it running overnight.
EBUSD_OPTS="--scanconfig=08 -d enh:esp-ebus.local:3335 --accesslevel=*"
The log this morning is much improved.
[ebusd-log.zip](https://github.com/danielkucera/esp8266-arduino-ebus/files/10978382/ebusd-log.zip)

danielkucera, 2023-03-15:
Can you please do this?
https://github.com/danielkucera/esp8266-arduino-ebus#nothing-helps-i-need-support

HiQual, 2023-03-17:
Sorry for the delay.  My SBC failed and all had to be rebuilt.
Here is the log.
[ebus dbug log.txt](https://github.com/danielkucera/esp8266-arduino-ebus/files/11001536/ebus.dbug.log.txt)

wrongisthenewright, 2023-03-18:
> I have just release v5.0 with enh protocol, please test and let me know if it works better now.

Hi Daniel,
thank you for your adapter and for you effort in support. I've updated my rev 4.1 adapter to laters FW ant Im currently doing some test. 

I have not had the time to do a thorough test session but the first indication, 
With my prefvious adapter (v2.x, I don't remember which) the arbitration loss was the norm, now even with the old FW the 4.1 board was really better. I'm running now the FW for some day and, in my case,  the enhanced protocol sems to worsen the situation.  I have a script that reads every hour a couple of dozen of parameters not broadcasted on the bus and with enh: enabled I get a lot more errors and, in some cases, the read command fails (so even the retries fails). With FW 5.0 but without the use of enh: I got some arbitration error but the retries got the read done.

Let me know if you need some log gathering.

BTW I Think I'll buy a third adapter as a backup in case the one I use fails/breaks up, have you an estimate of the lead time for the new batch?

danielkucera, 2023-03-18:
Hi,
Please provide log for both cases as described here https://github.com/danielkucera/esp8266-arduino-ebus#nothing-helps-i-need-support

danielkucera, 2023-03-18:
Also please provide output from telnet to port 5555

wrongisthenewright, 2023-03-18:
Hi Daniel,
thisi is the docker run command, the same one for enh and standard (obviously with the -d option different between the 2 runs)
```
docker run -d --name=ebusd \
        --network mynetwork \
        -p 8888:8888 \
        -p 8008:8008 \
        -v ....../Docker/Ebusd/configuration:/etc/ebusd \
        -v ....../Docker/Ebusd/scripts:/scripts \
        -v ....../Ebusd/html:/var/ebusd/html \
        -v /etc/timezone:/etc/timezone:ro \
        -v /etc/localtime:/etc/localtime:ro \
        john30/ebusd:latest \
        --mqtthost=mosquitto \
        --mqttport=1883 \
        --mqttuser=$USER \
        --mqttpass=$PASS \
        --mqttjson \
        --mqttint=/etc/ebusd/mqtt-hassio.cfg \
        --mqtttopic=ebusd/%circuit/%name \
        --enablehex \
        --configpath=/etc/ebusd \
        -d enh:192.168.1.165:3335 \
        --receivetimeout=5000 \
        --latency=2000 \
        --pollinterval=30 \
        --sendretries=10 \
        --acquireretries=5 \
        --acquiretimeout=20 \
        --log=all:debug
```

the output of telnet 5555 port :
```
uptime: 750241 ms
rssi: -82 dBm
```

Both runs were made with the same ebusd CSV, my file don't do polling (because of the arbitration loss errors), I gather usually the messages travelling on the bus and, only occasionally read some values with a crontab script that reads values that aren't commonly transmitted by the devices in my system. To force the error I run the basic script present in ebusd that read all defined lines in the CSV: readall.sh 

Logfile for enhanced protocol:
[ebusd-log-enh-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11008335/ebusd-log-enh-protocol.log)

Logfile for standard protocol:

[ebusd-log-standard-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11008336/ebusd-log-standard-protocol.log)

Let me know if you need anything else on my part.
TY

danielkucera, 2023-03-18:
-82dbm is pretty bad signal. Can you possibly find a better place where to install the adapter?

danielkucera, 2023-03-18:
And the logs are missing very important parameter `--lograwdata=data`

wrongisthenewright, 2023-03-18:
> -82dbm is pretty bad signal. Can you possibly find a better place where to install the adapter?

ATM I cannot, the adapter is near the boiler in my basement, the router is in my apartment 2 floor upstairs. I tryied to connect the adapter to the cables coming from the boiler to the remote command in my flat but I get continuous bus reconfiguraiton (cable not good?) I'll try to change the cable with a shielded one in the future.

wrongisthenewright, 2023-03-18:
> --lograwdata=dat

> And the logs are missing very important parameter `--lograwdata=data`

sorry, these are produced with the requested parameter
Enhanced protocol enabled
[ebusd-log-enh-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11009799/ebusd-log-enh-protocol.log)

Standard protocol enabled
[ebusd-log-standard-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11009800/ebusd-log-standard-protocol.log)

TY

danielkucera, 2023-03-19:
@wrongisthenewright , can you please try this firmware?
[firmware.zip](https://github.com/danielkucera/esp8266-arduino-ebus/files/11010790/firmware.zip)

danielkucera, 2023-03-19:
If that doesn't help, please try also this one:
[firmware2.zip](https://github.com/danielkucera/esp8266-arduino-ebus/files/11010959/firmware2.zip)

HiQual, 2023-03-20:
> 

You could try something like this powerline to wifi adapter.
https://www.amazon.co.uk/TL-WPA4220KIT-Powerline-Broadband-Configuration-UK/dp/B01LXOZ4EN/ref=asc_df_B01LXOZ4EN/?tag=googshopuk-21&linkCode=df0&hvadid=310694401667&hvpos=&hvnetw=g&hvrand=14807792556160462782&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1006667&hvtargid=pla-592318770684&psc=1&th=1&psc=1
There are lots to choose from, They are easy to setup and they work very well.

wrongisthenewright, 2023-03-20:
> For your poor wifi signal you could try something like this powerline to wifi adapter .... https://www.amazon.co.uk/TL-WPA4220KIT-Powerline-Broadband-Configuration-UK/dp/B01LXOZ4EN/ref=asc_df_B01LXOZ4EN/?tag=googshopuk-21&linkCode=df0&hvadid=310694401667&hvpos=&hvnetw=g&hvrand=14807792556160462782&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1006667&hvtargid=pla-592318770684&psc=1&th=1&psc=1 There are lots to choose from, they are easy to setup and they work very well.

> If that doesn't help, please try also this one: [firmware2.zip](https://github.com/danielkucera/esp8266-arduino-ebus/files/11010959/firmware2.zip)

I'll try both fw in the next days. Thanks for your effort, really appreciated.

wrongisthenewright, 2023-03-31:
> If that doesn't help, please try also this one: [firmware2.zip](https://github.com/danielkucera/esp8266-arduino-ebus/files/11010959/firmware2.zip)

Sorry for the late tests but I've been away the last few days. Today I tested directly the last 5.3 release on my v4 adapter, It seems that the enhanced protocol cause, at least in my "extreme" use case, more signal loss errors, these are the logs gathered.

[ebusd-log-enh-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11124212/ebusd-log-enh-protocol.log)
[ebusd-log-standard-protocol.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/11124213/ebusd-log-standard-protocol.log)

wrongisthenewright, 2024-02-11:
Hi, I'm reporting back after some months, sorry but I didn't want to touch the adapter during the peak of winter usage....
I've upgraded both my adapter (hw v2 and 4.1) to v7 FW, my problem with arbitration loss seems constant but atm I'm convinced is wifi related.
The adapter hw 4.1 works pretty well and with it I get a very low error ratio, ATM I'm with the 4.1 in v7 with standard protocol (strangely the ENH one is worst in my case)

From my pov you can close this issue.

Many thanks for you efforts!!!! Great product and support!!

danielkucera, 2024-02-11:
No problem, feel free to close on your own.

closed 2024-02-11 by wrongisthenewright