danielkucera/esp-ebus #24: esp8266-ebus v3.0

Sethach, 2023-02-15:
I have the ebus adapter with version 3.0, which is not properly working at my vaillant system. I assume it is because the bus does not give enough power.
Is it possible to attach the v3.0 to a seperate power supply? I can not find any documentation on the version3.0
thank you

danielkucera, 2023-02-15:
Hi,
on v3, TP2 is the same but:
- there is no TP3 but you can use BUS GND for GND
- there is no JP1 so you need to use 24V power supply or remove D4 for use with lower voltage supply

[image]

Sethach, 2023-02-15:
Thank you!
is it possible to supply the board with 3.3V over the programming interface?
By the way I am on the inform list for the latest version...

danielkucera, 2023-02-15:
It MAY work but I would not recommend it.