danielkucera/esp-ebus #25: Using this eBus with PS5511SZ

El-Omari, 2023-02-16:
Hello everyone, 

i would like to control my Solar storage charge controller PS 5511 SZ. It has an interface eBus. 
 My questions are:
1 could I use this eBus? 
2 Do you have experience with this control? 

Thanks

El-Omari, 2023-04-06:
Hi Daniel,

I got the EBUS to WiFi adapter module an connect it to PS5511 SZ, I installed the recommended ebusd of John.

the module is connected to the router and got an IP. with telnet i could have a connection to the item. My Questions ares, 
1. John use the port 9999 and you recommend the following ports 3333,3334,3335 and 5555. 
2. Which comand shall I use under /etc/default/ebusd? is this correct 
EBUSD_OPTS="--scanconfig -d enh:192.168.178.68:3335 --latency=200000"

thanks for your answer 

i tried to install your

danielkucera, 2023-04-06:
1. yes, john uses a different adapter
2. yes, correct

El-Omari, 2023-04-06:
I'm looking for the correct csv-file for my PS5511 SZ. Could you give some information, where to find theme?

danielkucera, 2023-04-06:
I have no idea, sorry.