danielkucera/esp-ebus #26: read-only interface issue

SzosszeNET, 2023-02-23:
Hi there, hope you are well. This issue has thrown me off for a bit, but would appear that if I start ebusd with the read-only (3334) port I get only errors:

docker run --rm -it -p 8888 john30/ebusd --scanconfig -d esp-ebus.majoros.me:3334 --latency=20 
2023-02-23 23:00:16.639 [main notice] ebusd 23.1.23.1 started with auto scan on device esp-ebus.majoros.me:3334
2023-02-23 23:00:19.016 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error: 
2023-02-23 23:00:19.045 [bus notice] bus started with own address 31/36
2023-02-23 23:00:19.045 [bus notice] device invalid
2023-02-23 23:00:24.072 [bus notice] re-opened esp-ebus.majoros.me:3334
2023-02-23 23:00:24.072 [bus notice] device invalid
2023-02-23 23:00:29.100 [bus notice] re-opened esp-ebus.majoros.me:3334
2023-02-23 23:00:29.100 [bus notice] device invalid
2023-02-23 23:00:34.128 [bus notice] re-opened esp-ebus.majoros.me:3334
2023-02-23 23:00:34.128 [bus notice] device invalid
^C2023-02-23 23:00:34.421 [main notice] SIGINT received
2023-02-23 23:00:39.156 [bus notice] re-opened esp-ebus.majoros.me:3334

However if started with the RW port (3333) it works?

docker run --rm -it -p 8888 john30/ebusd --scanconfig -d esp-ebus.majoros.me:3333 --latency=20 
2023-02-23 22:58:07.676 [main notice] ebusd 23.1.23.1 started with auto scan on device esp-ebus.majoros.me:3333
2023-02-23 22:58:08.483 [bus notice] bus started with own address 31/36
2023-02-23 22:58:08.486 [bus notice] signal acquired
2023-02-23 22:58:10.711 [bus notice] new master 10, master count 2
2023-02-23 22:58:10.766 [bus notice] new master 03, master count 3
2023-02-23 22:58:10.766 [update notice] received unknown MS cmd: 1008b5110101 / 0938390080ffff0000ff
2023-02-23 22:58:11.035 [bus notice] new master 71, master count 4
2023-02-23 22:58:11.035 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff5008ff6c0000ff
2023-02-23 22:58:11.302 [update notice] received unknown MS cmd: 1076b512030f0101 / 076003000080ff00
2023-02-23 22:58:11.516 [update notice] received unknown BC cmd: 10feb505025c00
2023-02-23 22:58:11.780 [update notice] received unknown MS cmd: 1008b51009000037ffffff060000 / 0101
2023-02-23 22:58:12.045 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-02-23 22:58:12.318 [update notice] received unknown MS cmd: 1076b5040100 / 0a00ffffffffffffff5008
2023-02-23 22:58:12.556 [update notice] received unknown MS cmd: 1008b507020937 / 023202
2023-02-23 22:58:12.794 [update notice] received unknown BC cmd: 10feb516080039592223020423
2023-02-23 22:58:15.466 [update notice] received unknown BC cmd: 10feb51603014007
2023-02-23 22:58:15.987 [update notice] received unknown MS cmd: 0376b51206130010b40400 / 0200ff
2023-02-23 22:58:18.639 [bus notice] scan 08: ;Vaillant;HMU00;0607;5103
2023-02-23 22:58:18.639 [update notice] store 08 ident: done
2023-02-23 22:58:18.639 [update notice] sent scan-read scan.08  QQ=31: Vaillant;HMU00;0607;5103
2023-02-23 22:58:18.639 [bus notice] scan 08: ;Vaillant;HMU00;0607;5103
2023-02-23 22:58:19.300 [main notice] read common config file vaillant/broadcast.csv
2023-02-23 22:58:19.465 [main notice] read common config file vaillant/scan.csv
2023-02-23 22:58:19.630 [main notice] read common config file vaillant/general.csv
2023-02-23 22:58:19.896 [main notice] read scan config file vaillant/08.hmu.csv for ID "hmu00", SW0607, HW5103
2023-02-23 22:58:20.223 [main notice] found messages: 65 (0 conditional on 0 conditions, 0 poll, 10 update)
2023-02-23 22:58:20.374 [update notice] sent unknown MS cmd: 3108b5090124 / 09003231323233383030
2023-02-23 22:58:20.582 [update notice] sent scan-read scan.08 id QQ=31: 
2023-02-23 22:58:20.735 [update notice] sent scan-read scan.08 id QQ=31: 
2023-02-23 22:58:21.430 [update notice] sent scan-read scan.08 id QQ=31: 21;22;38;0010047382;1300;105253;N9
2023-02-23 22:58:21.430 [bus notice] scan 08: ;21;22;38;0010047382;1300;105253;N9
2023-02-23 22:58:21.697 [update notice] received read hmu Status01 QQ=10: 28.0;28.5;-;-;-;off
2023-02-23 22:58:21.965 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff5008ff6c0000ff
2023-02-23 22:58:22.233 [update notice] received unknown MS cmd: 1076b512030f0101 / 076003000080ff00
2023-02-23 22:58:22.567 [update notice] received update-write hmu SetMode QQ=10: auto;27.5;-;-;0;1;1;0;0;0
2023-02-23 22:58:22.654 [bus error] own master address 31 is used by another participant
2023-02-23 22:58:23.374 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-02-23 22:58:23.613 [bus notice] scan 15: ;Vaillant;CTLV2;0514;1104
2023-02-23 22:58:23.613 [update notice] store 15 ident: done
2023-02-23 22:58:23.613 [update notice] sent scan-read scan.15  QQ=31: Vaillant;CTLV2;0514;1104
2023-02-23 22:58:23.613 [bus notice] scan 15: ;Vaillant;CTLV2;0514;1104
2023-02-23 22:58:23.766 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323233313030
2023-02-23 22:58:23.923 [update notice] sent scan-read scan.15 id QQ=31: 
2023-02-23 22:58:24.002 [bus notice] max. symbols per second: 109
2023-02-23 22:58:24.158 [update notice] sent scan-read scan.15 id QQ=31: 
2023-02-23 22:58:24.318 [update notice] sent scan-read scan.15 id QQ=31: 21;22;31;0020260921;0953;013860;N1
2023-02-23 22:58:24.318 [bus notice] scan 15: ;21;22;31;0020260921;0953;013860;N1
2023-02-23 22:58:24.483 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv2", SW0514, HW1104
2023-02-23 22:58:24.483 [main error] scan config 15: ERR: element not found
2023-02-23 22:58:26.688 [main error] scan config 36: ERR: read timeout
2023-02-23 22:58:28.882 [bus notice] scan 76: ;Vaillant;VWZIO;0202;0103
2023-02-23 22:58:28.882 [update notice] store 76 ident: done
2023-02-23 22:58:28.882 [update notice] sent scan-read scan.76  QQ=31: Vaillant;VWZIO;0202;0103
2023-02-23 22:58:28.882 [bus notice] scan 76: ;Vaillant;VWZIO;0202;0103
2023-02-23 22:58:29.063 [update notice] sent unknown MS cmd: 3176b5090124 / 09003231323234323030
2023-02-23 22:58:29.245 [update notice] sent scan-read scan.76 id QQ=31: 
2023-02-23 22:58:29.425 [update notice] sent scan-read scan.76 id QQ=31: 
2023-02-23 22:58:29.605 [update notice] sent scan-read scan.76 id QQ=31: 21;22;42;0010031644;0082;010561;N4
2023-02-23 22:58:29.605 [bus notice] scan 76: ;21;22;42;0010031644;0082;010561;N4
2023-02-23 22:58:29.769 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2023-02-23 22:58:29.769 [main error] scan config 76: ERR: element not found
2023-02-23 22:58:30.751 [update notice] received read hmu Status01 QQ=10: 28.0;28.5;-;-;-;off
2023-02-23 22:58:31.019 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff5008ff6c0000ff
2023-02-23 22:58:31.287 [update notice] received unknown MS cmd: 1076b512030f0101 / 076003000080ff00
2023-02-23 22:58:31.551 [update notice] received update-write hmu SetMode QQ=10: auto;27.5;-;-;0;1;1;0;0;0
2023-02-23 22:58:31.816 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-02-23 22:58:39.929 [update notice] received unknown MS cmd: 0376b51206130010b30400 / 0200ff
2023-02-23 22:58:40.719 [update notice] received read hmu Status01 QQ=10: 28.0;28.5;-;-;-;off
2023-02-23 22:58:40.987 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff5008ff6c0000ff
2023-02-23 22:58:41.255 [update notice] received unknown MS cmd: 1076b512030f0101 / 076003000080ff00
2023-02-23 22:58:41.519 [update notice] received update-write hmu SetMode QQ=10: auto;27.5;-;-;0;1;1;0;0;0
2023-02-23 22:58:41.784 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-02-23 22:58:47.898 [update notice] received unknown MS cmd: 0376b51206130010b30400 / 0200ff
2023-02-23 22:58:50.709 [update notice] received read hmu Status01 QQ=10: 27.5;28.5;-;-;-;off
2023-02-23 22:58:50.978 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff5008ff6c0000ff
2023-02-23 22:58:51.245 [update notice] received unknown MS cmd: 1076b512030f0101 / 076003000080ff00
2023-02-23 22:58:51.509 [update notice] received update-write hmu SetMode QQ=10: auto;27.5;-;-;0;1;1;0;0;0
2023-02-23 22:58:51.774 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101

Is this expected and with ebusd :3333 have to be used?

As a followup question the "unknown MS" and "unknown BC", the [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv2", SW0514, HW1104 and [main error] unable to load scan config 76: no file from vaillant with prefix 76 found are potentially all linked and are an ebusd issue?

danielkucera, 2023-02-23:
Hi,
you should be able to use readonly but you need to specify static device configuration and disable scanning - scanning needs to send data and maybe also specify readonly for ebusd.

And yes, those errors point to missing config for your specific HW.

SzosszeNET, 2023-02-24:
Hi Daniel,

Thank you for getting back to me I made some progress, still lots of "issues" but potentially moved to the right direction.
What I have noticed by using a different ebusd csv that adds support to some of my components (potentially with the box standard I would get there - not sure) is that my "unsupported" components the VR921 (makes my installation unmanageable via sensoapp) and the controller (assume the VWZIO) displays a "communication fault" message.

Was about to say that RO seems to solve it, but after running the adapter in RO it looks like it still locks up those two components as well not recognizing it...

Any thoughts?

danielkucera, 2023-02-24:
Can you try this?
https://github.com/danielkucera/esp8266-arduino-ebus#the-adapter-is-loosing-connectivity-breaking-other-ebus-components-communication-d7-is-blinking-wifi-is-disappearing-after-connecting-devices-on-the-bus-show-error-status-or-other-intermittent-failures

SzosszeNET, 2023-02-24:
Hi Daniel, thank you for your kind help. Could not find a 24V power brick unfortunately, so just to confirm if I was to cut between the JP1 pads, testing with a multimeter (continouty test) and doesn't beep it should be fine with a ~5V power brick?

danielkucera, 2023-02-24:
Yes, correct.

SzosszeNET, 2023-02-24:
Thank you, amazing. So far looks good the gateway still works and the controller hasn't dropped any issues either.

closed 2023-02-24 by SzosszeNET