danielkucera/esp-ebus #35: esp32 v5 firmware upload

PiotrHab, 2023-04-03:
Hello Daniel
First of all big thanks for the design, powerfull and affordable product comparing to other solutions.
"Out of the box" it works fine, ebusd on RPi works too, so no problem here.

I wanted to try out latest code, so i downloaded the project to PIO which unfortunately failed to upload to esp32.

pio run -e esp32-c3 -t upload

...compiles ok but then ...

Serial port COM11
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: Invalid head of packet (0x08): Possible serial noise or corruption.      
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
================================================= [FAILED] Took 47.80 seconds =================================================

Environment    Status    Duration
-------------  --------  ------------
esp32-c3       FAILED    00:00:47.800
============================================ 1 failed, 0 succeeded in 00:00:47.800 ============================================ 

OS ? - Win10, VScode, PIO (esp32 platform, prooved to run other esp32 projects with vroom, d1 mini, etc)
Adapter hardware ? - esp32-c3 (latest hardware batch)
Why not OTA? - i want to program my stuff there too, so basically customize it using PIO.
Power scheme? - jumper set to power from USB
Other things? - When adapter is connected on usb-c, win shows unknown USB device

Is there any specific win driver i should install ? 
it doesn't look like it is FT or CH family usb to serial.

danielkucera, 2023-04-03:
Hi @PiotrHab , thanks for detailed description. I haven't had chance to update the documentation yet.
Please short PROG and GND pins before connection the device to USB. It should enumerate differently.

PiotrHab, 2023-04-03:
Works like a charm now, thanks

closed 2023-04-03 by PiotrHab