danielkucera/esp-ebus #48: WDT is disabled for ESP8266

bakadave, 2023-04-19:
Hi, browsing the source code I found this:
```
void wdt_start() {
#ifdef ESP32
  esp_task_wdt_init(6, true);
#elif defined(ESP8266)
  ESP.wdtDisable();
#endif
}
```
what is the point of having a `wdt_start()` function that turns the watchdog off? Am I missing something here?

The device works well regardless, just curious.

danielkucera, 2023-04-19:
Please read about the watchdog here to understand the function:
https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

bakadave, 2023-04-19:
If I understand correctly, by calling `ESP.wdtDisable();` we disable software watchdog and use hardware watchdog only?
Also thank you for your time, one learns a lot from these converstions.

danielkucera, 2023-04-19:
Correct.

closed 2023-04-19 by bakadave