danielkucera/esp-ebus #51: This page isn’t working

adallari1969, 2023-05-14:
Hi
I connected the esp8266-arduino-ebus v5.2 to my ebus
configured the wifi network

the device get the ip, I can ping it 
I can connect to esp-ebus.local port 3334 using telnet - I see unreadable binary data 
but if I try connect to http://esp-ebus.local:5555 i get the error (from browser)
"This page isn’t working192.168.1.132 sent an invalid response.
ERR_INVALID_HTTP_RESPONSE"

If I try to set the esp-bus ip address in the ebusd add on in my Home assistan, I get the error "ERROR: Got unexpected response from the API: Service not enabled"

can you pls help me?

adallari1969, 2023-05-14:
I also tried the reset, but no changes

danielkucera, 2023-05-14:
Hello,
which browser?

adallari1969, 2023-05-14:
Hi
Both chrome and edge

adallari1969, 2023-05-17:
hallo,
any feedback from your side?

danielkucera, 2023-05-20:
Hi @adallari1969 ,
it is related to #39 .
I am still trying to figure out a good way to handle http requests on the status server.
Meanwhile you can use `telnet esp-ebus.local 5555` to check the status server.

adallari1969, 2023-05-21:
Any browser works whit it?

danielkucera, 2023-05-23:
Hm, I was under impression that both Firefox and Chrome works but now I have tested more of them and found out that only Firefox under Linux works.
I'll make the next firmware work with all browsers. For now you can use telnet to check the status:
```
telnet esp-ebus.local 5555
```

danielkucera, 2024-02-04:
resolved in #64

closed 2024-02-04 by danielkucera