danielkucera/esp-ebus #52: Communication fault VWZ AI

xbouthoorn, 2023-07-06:
Hello,

I'm using the v5.2 adapter now for some weeks. At first it worked great, but after a power outage it is giving me issues. The VWZ AI is giving communication faults every x hours. I have to disconnect the adapter from the bus, reset the VWZ and then after some minutes I can reconnect the adapter again to fix it. After some hours it will be down again.

The adapter is powered by a 5v 3A USB adapter, without adapter it didn't work at all.

What can cause this problem?

danielkucera, 2023-07-21:
Sorry for a late response. Can you please capture the adapter LEDs on video when this happens and share the video?

xbouthoorn, 2023-09-19:
After some time the issue didn't reoccur. So closed for now.

closed 2023-09-19 by xbouthoorn