danielkucera/esp-ebus #54: Invalid device

jankorous, 2023-08-25:
hi, I've been struggling with the ebuds setup for two days.

I bought a 5.2 adapter, connected it to the BUS of my electric boiler (Protherm Ray) according to the instructions. After the first start I gave it the iot wifi info it connected to and was assigned an IP.
In HA I added the ebusd daemon and in the settings I added the IP adapter according to the instructions. It gives me an "invalid device" error.

`[bus notice] device invalid`

After two days I don't know where to go anymore and I'm asking for help.
[image]

eBUSd version is 23.2.2

jankorous, 2023-08-25:
And still repeating 

2023-08-25 11:19:57.917 [network info] [00010] client connection opened 127.0.0.1
2023-08-25 11:19:57.918 [network info] [00010] connection closed
2023-08-25 11:22:37.281 [bus notice] device invalid
2023-08-25 11:22:42.421 [bus notice] device status: resetting
2023-08-25 11:22:42.421 [bus notice] re-opened 192.168.40.151:3335
2023-08-25 11:22:42.424 [bus notice] device status: reset

danielkucera, 2023-08-26:
Hi @jankorous , from the readme, you have two options how to refer to the device:
- raw protocol: `esp-ebus.local:3333` or  `<IP>:3333` 
- enh protocol: `enh:esp-ebus.local:3335` or `enh:<IP>:3335`

Please check your config.

jankorous, 2023-08-26:
I've solved this issue with a one clockwise turn on trimmer. (?) I can not explain it...

closed 2023-11-14 by jankorous