danielkucera/esp-ebus #55: Firmware v6.1 HW v2.1

HiQual, 2023-09-21:
Hello Daniel.
I just applied the v6.1 update firmware file v3.x using espota.py
On successful completion of flash it did not auto reconnect to my wi-fi.
It was in reset mode so I could directly access the esp-ebud SSID
I input my home wi-fi credentials - it restarted and connected OK as esp-ebus.local
I could ping it but it disconnected itself after about 3 minutes.
It had reverted to reset/configuration mode.
I have repeated this a few times, but the connection only lasts a few minutes.
Any suggestions please?

danielkucera, 2023-09-21:
Hi,
after those few minutes, what does it do exactly?
It disconnects and the config resets? What does the config page say?

HiQual, 2023-09-22:
Good morning Daniel.
In the few minutes the adapter stays connected as client to my router and it responds to ping requests.
After a few minutes it stops responding to ping requests on both esp-eBus.local and the IP address.
Looking at available networks I see that the adapter has reverted to AP Wi-Fi mode broadcasting esp-eBus.
[image]

danielkucera, 2023-09-22:
You are right, I confirm this is a bug introduced in v6.1, earlier versions should be fine. I will work on this soon.

HiQual, 2023-10-14:
Fixed in 6.2, 6.3
Many thanks.

closed 2023-10-14 by HiQual