danielkucera/esp-ebus #56: New 6.1 device bought from elecrow has protected wifi

Spancii, 2023-10-04:
Hi Daniel,

I bought an adapter from elecrow but this device has wifi password. I have been tried reset the device but unfortunately i can't. Do you have any idea what is the password or why i can't reset the device to default.

Many thanks!

danielkucera, 2023-10-04:
Hi, can you please provide a screenshot of the problem?

Spancii, 2023-10-05:
Hi 
[image]

danielkucera, 2023-10-05:
Do you have QC sticker on your device? It looks like they forgot to load correct firmware.
Do you think you can flash the firmware by yourself?

Spancii, 2023-10-05:
![379669406_175703752242460_1203277469209931175_n](https://github.com/danielkucera/esp8266-arduino-ebus/assets/96431383/e4850813-9323-4495-8bd8-4
[image]
5743480783c)

yes i have. 

I'll try tomorrow

Spancii, 2023-10-05:
[image]

danielkucera, 2023-10-05:
Ok, I'll update the flashing guide meanwhile.

ActiveByte, 2023-10-05:
Same issue, PASS - QC6 sticker on the bottom

danielkucera, 2023-10-06:
Can you please try to connect using password "lectronz" and then open the web interface on http://192.168.4.1/ ?

Spancii, 2023-10-06:
Hi Daniel,
 
which port is the TP1? The flasher just connecting....

danielkucera, 2023-10-06:
I am sorry, I didn't manage to update the guide.
Download Flash download tools from here: https://www.espressif.com/en/support/download/other-tools
Connect PROG and GND pin with a wire before powering the adapter.
Connect the adapter to a PC using USB cable.
Flash the firmware `firmware-fullflash-HW_v5.x.bin` to address 0x0

etmnidj1, 2023-10-06:
Same problem here, QC3 sticker on the back

danielkucera, 2023-10-06:
Yes, it seems they have messed up it in the factory and loaded wrong firmware to all adapters sold after 2023/09 :(

etmnidj1, 2023-10-06:
I dont have any experience with flashing but will give it a go...

etmnidj1, 2023-10-06:
Im in the flashtool now everything is green, pressed start but nothing seems to happen?

Any tips?

[image]

danielkucera, 2023-10-06:
You need to select COM:
And you need to pick a different file: firmware-fullflash-HW_v5.x.bin

etmnidj1, 2023-10-06:
Selected com 4; it says 2-sync fail

Spancii, 2023-10-06:
i have a worse situation

[image]

danielkucera, 2023-10-06:
you need to specify @ 0x0 , see the screenshot from @etmnidj1

danielkucera, 2023-10-06:
> Selected com 4; it says 2-sync fail

Are there any other ports to select? Can you try different ones?

etmnidj1, 2023-10-06:
Some questions:

Connect PROG and GND pin with a wire before powering the adapter. -> I see no pins; just holes. I connected these with a wire
Connect the adapter to a PC using USB cable. -> done (do i leave the wires in PROG and GND or must these removed after powering up?
Flash the firmware firmware-fullflash-HW_v5.x.bin to address 0x0 -> got error message 2-sync fail

[2023-10-06 10:04:07,278][ESP8266Loader_spi[1]][espDownloader.py][line:561][ERROR]: IOError: the serial port should probably be removed
[2023-10-06 10:04:07,278][ESP8266Loader_spi[1]][espDownloader.py][line:745][ERROR]: ESP8266 Chip sync error esp_sync_blocking.

etmnidj1, 2023-10-06:
> > Selected com 4; it says 2-sync fail
> 
> Are there any other ports to select? Can you try different ones?

Tried port 5 but that gave a com fail

danielkucera, 2023-10-06:
Yes, there are only holes, just use a piece of copper wire and keep it there during the whole flashing procedure.
You can also try this tool: https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ but it works in Chrome only (not Firefox).

etmnidj1, 2023-10-06:
Tried the other tool in chrome but there's nothing on the com ports; just bluetooth devices and spp1 on port 5

Spancii, 2023-10-06:
@danielkucera thank you! I can burning with the web tool. My device is already work.

danielkucera, 2023-10-06:
> Tried the other tool in chrome but there's nothing on the com ports; just bluetooth devices and spp1 on port 5

Which operating system you have? Can you send a screenshot of the selection?

etmnidj1, 2023-10-06:
Seems like i managed to flash te correct firmware tried anotherusb cable and computer. Filled in the wifi credentials but cannot connect; als not to the modules wifi?? Will flash again

etmnidj1, 2023-10-06:
Do I have to reset after flashing  or? If yes how is this done??

ActiveByte, 2023-10-06:
I was able to login with the password lectronz wich i found in the config file, but http://192.168.4.1/ was not reachable.

I was able to flash the esp with the esptool and the fullflash image yesterday and got it working now.

etmnidj1, 2023-10-06:
I have access to the web interface; I select my wifi network; fill in password en press save.

What must I do after that? I cannot connect to 192.168.4.1 when I connect my computer to my own wifi network?

Rgrds Niels

danielkucera, 2023-10-06:
@etmnidj1 please follow https://github.com/danielkucera/esp8266-arduino-ebus#quickstart

keithwithnall, 2023-12-28:
@danielkucera I have the same issue with a newly purchased v6.1 board from Elecrow with Pass QC3 label, but with no initial wi-fi transmission at all. Have apparently successfully loaded firmware-fullflash-HW_v5.x.bin from the v6.2 Assets folder using the Adafruit web ESP Tool at offset 0x0 (no errors) and then reset the device. However still no wi-fi transmission from the device - tried for hidden networks too but no esp-eBus SSID at all. Any suggestions as to what else I can try?

danielkucera, 2023-12-29:
Hi @keithwithnall ,
If you are getting restarts after about 2 minutes, you are probably getting caught here:
https://github.com/danielkucera/esp8266-arduino-ebus/blob/master/src/main.cpp#L321
meaning this one: https://github.com/danielkucera/esp8266-arduino-ebus/blob/master/src/main.cpp#L316
has to equal to connected.

Can you verify the adapter is really not available in your network?

keithwithnall, 2023-12-30:
Hi @danielkucera 
Thanks for your quick response. I can confirm that the device is not, and has never, connected to the network. I know this as it has never transmitted an open network to allow me to enter the network credentials, and the network is Ubiquiti Unifi based and shows me every device that has been connected. I have also used a wi-fi analyser to check for any hidden transmissions from the device. Is it possible to get any debugging information from the device that would help you? If so please advise how I can do this.
Thanks Keith

danielkucera, 2024-01-01:
This does not make any sense. Can you please contact me directly so we can arrange sending the device to me for investigation?

keithwithnall, 2024-01-02:
Message sent via your blog contact form - thanks

keithwithnall, 2024-01-12:
Hi Daniel, It appears that I definitely had a module with a hardware issue. Elecrow have sent a replacement and it connects to the network exactly as expected. I just need to sort out the HA end of things now. Thanks again for all of your help which was much appreciated.

zipkid, 2024-01-19:
Hi,

I just received an adapter from Elecrow. Solid led D1 & D7 and no wifi AP.

[image]
[image]

It seems like they are still sending out the badly provisioned units. 😢

I will try to flash the firmware to it one of the following days. 

Kind regards,

Stefan.

danielkucera, 2024-01-19:
Hi @zipkid ,
did you try to connect it to the ebus?
When disconnected, D1 is expected to be on constantly.

keithwithnall, 2024-01-19:
Also note that it will transmit wi-fi if connected to a simple power adaptor with a USB-C cable, but if connected to a computer with USB-C it may not enter wi-fi AP mode even if it is a good unit.

zipkid, 2024-01-19:
Hello,

I cannot connect it to ebus as that only gets installed on Feb 1st :-)
I have tried with a simple power adapter to usb-c and that also gives no AP.
When connecting via computer usb and using minicom nothing happens.

The flashing instructions say "using a wire connect pins PROG and TP3" 
I see the PROG pin (hole) but i don't find one labeled TP3... ?
Where do i need to connect PROG to?

zipkid, 2024-01-19:
I have connected my usbserial adapter, connected PROG to GND, tried, switched TX & RX. 
Same result always
`╭─zipkid@android18 ~/tmp
╰─➤  esptool.py --port /dev/cu.wchusbserial143240 write_flash 0x0 ~/Downloads/firmware-fullflash-HW_v5.x-be7bb3.bin                                                                                        1 ↵
esptool.py v3.2
Serial port /dev/cu.wchusbserial143240
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```
╭─zipkid@android18 ~/tmp
╰─➤  esptool.py --port /dev/cu.wchusbserial143240 write_flash 0x0 ~/Downloads/firmware-fullflash-HW_v5.x-be7bb3.bin                                                                                        2 ↵
esptool.py v3.2
Serial port /dev/cu.wchusbserial143240
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0xFF): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```

keithwithnall, 2024-01-19:
I contacted Daniel OK using the contact form on his blog site https://blog.danman.eu/contact/
Keith

zipkid, 2024-01-19:
I tried the same message again and this time it went through....

sergius18100, 2024-01-25:
Hello,
another brand new 6.1 device from Elecrow, no wifi on start. Reset several times, flashed version 6.2 and then 6.3 beta, reset each time, still no wifi. Connected to the terminal session on putty, here is what it shows:
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403825b6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
E (1034) wifi:sta is connecting, return error
Anything that can be tweaked, or is it bad hardware?

danielkucera, 2024-02-01:
I was hoping to find some bug in the device from @zipkid but it only needed an erase and firmware flash and it started working. I'll prepare a new firmware where we will see some debug messages on the USB serial. @AndrevdG , give me a few days please. Meanwhile, can you please try SPI MODE: DIO?

AndrevdG, 2024-02-01:
@danielkucera, I have tried with DIO but it does not make difference, same error.

Just for fun, with the web tool, I erased the adapter before reflashing it. That makes no difference in behavior.
The putty serial console displays the same behavior. I also tried resetting (did that before as well), this also does not make a difference. To confirm I am doing this correctly:
For a v6.1 board, the reset is a short between ground and to-ebus (all leds are on while doing this):
[image]

Other than that, no worries, I will wait for your response. It would be great if we could figure out why it is not working for me.

AndrevdG, 2024-02-01:
> @danielkucera, I have tried with DIO but it does not make difference, same error.
> 

Actually, the reason that the flash exe tool did not work was my mistake... 
[image]
If I select the correct board, the exe tool also flashes the adapter fine. Still no AP though

sergius18100, 2024-02-01:
I had the problem of not seeing the wifi at all, as mentioned by keithwithnall and AndrevdG. Flashed the device several times with the latest firmware, tried with a charger and with a computer.
In the end, it turns out that the device actually works somehow, but the wifi shows very late some minutes after boot, which creates the impression it does not initiate. Having found out this, got several root problems that lead to these symptoms.
- The device apparently would not connect to a WPA3 network, neither would it connect to a non-broadcasted SSID. I had to create a separate WPA2 network with broadcasted SSID, after which the device connected.
- Once configured, it reboots after exactly three minutes of being on line. This is different from two minutes that AndrevdG has, but seems to be the same behaviour. Once rebooted, it remains silent for a few minutes, but connects to wifi after that, lives for another three minutes and reboots again.
The wifi signal is good, access point just next to the device, so it cannot explain the reboots. What can be done?

danielkucera, 2024-02-01:
@sergius18100 , the firmware in artifact has enabled some logging to USB serial port. Can you try that and capture the logs so we can see what is happening there?

PiotrHab, 2024-02-01:
@danielkucera , I have some other use case which could help with wifi response.
I'm playing with the device very occasionally (when i have time) and i observed that after long time of no ebusd connection, the device becomes very slow at wifi response (status page takes up to 3-4 seconds to show, ebusd reports device connection lost and reconnection). 
I figured out that is because of wifi modem sleep. After WiFi.begin(); , i've added esp_wifi_set_ps(WIFI_PS_NONE);
And that problem seems to be gone.
I'm aware this makes device power hungry but on the other hand it's more wifi responsive.
So maybe that could solve some issues reported here...

AndrevdG, 2024-02-01:
With the updated firmware:
[image]
From the logging it appears to start the AP, but as you can see from the windows Wifi manager, no esp-eBus ssid is visible.

I checked this with two (be it win11) devices and my mobile (android) to be sure, but no AP shows up

AndrevdG, 2024-02-01:
Don't know if it helps, but I saw it attempting to connect to a default AP, so I tried to set my mobile hotspot in Windows to it. It does also not connect to that AP. 
[image]

danielkucera, 2024-02-02:
Please try again:
https://github.com/danielkucera/esp8266-arduino-ebus/actions/runs/7752846735

the default password is: ebusebus

AndrevdG, 2024-02-03:
Well, that definitely did something. With that firmware the AP shows up in the SSID list and I can successfully connect to it. 

It starts with:
[image]

after I update the settings:

[image]

It did not reconnect to my WIFI immediately. However, after disconnecting / reconnecting the adapter:
[image]

and also mDNS:
[image]

I'll attach it to ebus and see if I get some actual data

AndrevdG, 2024-02-03:
Actually, it appears that http://esp-ebus.local/param this (from the quickstart) does not work, so not sure how to adjust the PWM value (it is unset currently). Going to http://esp-ebus.local asks for a login

AndrevdG, 2024-02-03:
Ok, going over your code it appears the correct url is http://esp-ebus.local/config.

I found the login on the IotWebConf usermanual (user:admin, pw: \<whatever the AP password was\>), so I managed to connect. Port 3333 and 3334 appear to be working, presuming the expected output is something like this:
[image]

Telnetting (or using a browser) to port 3335 or 5555 does not work however, this immediately disconnects putty and the browser just does:
[image]

danielkucera, 2024-02-03:
Yes, this I moved from wifimanager to iotwebconfig so some aspects are different.
However, the ports 3335 and 5555 should still work with telnet, I haven't done any changes there.
Can you please check again?

AndrevdG, 2024-02-03:
~~Just tried again, with putty the session is disconnected immediately (I can see putty pop up, but no error message and the window just disappears).~~ However, I have added the adapter to Home Assistant now and the ebusd add-on appears to be able to read 3335, so it appears to be working.

I guess it is an issue with putty (it may be an older version). Using telnet in a commanline window appears to work. 3335 gives garbage (as I suppose is expected) and 5555 gives a status and exits (that may also be expected):

[image]

danielkucera, 2024-02-03:
Yes, that is correct

AndrevdG, 2024-02-03:
As a follow up, the adapter has been running for several hours now connected to ebus and providing data to home assistant. 

Thank you for your quick responses and help to get the adapter functional!

[1 more reply]

closed 2026-11-09 by danielkucera