danielkucera/esp-ebus #57: Signal lost

JaccovdS, 2023-10-06:
I've followed the steps in the README but when I enable `ebusd` I get the following:
```
[bus notice] signal acquired
[bus error] signal lost
[bus notice] signal acquired
[bus error] signal lost
[bus notice] signal acquired
```
It keeps going on like this. I already tried changing the latency but that doesn't make a difference. I guess I need to change a parameter somewhere but I don't know which one?

I've got version 6.0 and power it via USB.

danielkucera, 2023-10-08:
Do this please: https://github.com/danielkucera/esp8266-arduino-ebus#nothing-helps-i-need-support

JaccovdS, 2023-10-10:
I did a bit more trial and error. While the LED was blinking fine, the PWM value was apparently too low. I set a higher PWM and it solved the problem.

closed 2023-10-10 by JaccovdS

danielkucera, 2023-10-10:
Thank you for the info. Can you please tell what value worked for?

JaccovdS, 2023-10-10:
I put the value to 100 in the end.