danielkucera/esp-ebus #59: Vaillant BAI cant control

bayc4d, 2023-12-03:
2023-11-29 21:00:11.996 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0505;3302
2023-11-29 21:00:11.996 [bus notice] scan 08: ;Vaillant;BAI00;0505;3302
2023-11-29 21:00:12.007 [main notice] read common config file vaillant/general.csv
2023-11-29 21:00:12.008 [main notice] read common config file vaillant/broadcast.csv
2023-11-29 21:00:12.009 [main notice] read common config file vaillant/scan.csv
2023-11-29 21:00:12.011 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0505, HW3302

bai.308523.inc files i edit r to w some commands but i cant control temperature or on off option.

anyone help ? 

i buy from elecrow EBUS to WiFi adapter module v6.1. 

can i upload opentherm firmware this board? or solution for control with ebusd mqtt.

danielkucera, 2023-12-04:
Hi @bayc4d ,
this is rather a question to ebusd software but feel free to keep it open also here.
What do you mean by opentherm firmware?

bayc4d, 2023-12-04:
i cant control with ebusd my boiler. so i want to try opentherm with adapter.

https://diyless.com/product/esp8266-thermostat-shield

https://github.com/ihormelnyk/opentherm_library/

according to my research i need vailant vr33 converter for use with ebus and opentherm protocol.