danielkucera/esp-ebus #60: Home assistant loses connection with EBUS adapter

etmnidj1, 2024-01-02:
Hi Daniel,

I managed to install the  EBUS to WiFi adapter module v6.1 into my vaillant geotherm heatpump. 

I use HomeAssistant as a dashboard to monitor the heatpump. After setup everything works fine but I notice that after two to three days the connection is lost (both local IP address and in home assisitant ) with the ebus adapter module. 

Wifi coverage is excellent nearthe pump so that shouldn't be a prblem.

Any suggestions what I can do?

Please let me know how I can provide you with logs etc. 

Regards

Niels

danielkucera, 2024-01-02:
Hi @etmnidj1 ,

does the adapter completely looses WiFi connection? Can you ping it?

Can you try to power it from an external adapter if the problem persists? See https://github.com/danielkucera/esp8266-arduino-ebus#the-adapter-is-loosing-connectivity-breaking-other-ebus-components-communication-d7-is-blinking-wifi-is-disappearing-after-connecting-devices-on-the-bus-show-error-status-or-other-intermittent-failures

etmnidj1, 2024-01-02:
Hi Daniel,

This is what I get back after pingin

C:\Users\RWS>ping 192.168.11.248

Pinging 192.168.11.248 with 32 bytes of data:
Reply from 192.168.11.250: Destination host unreachable.
Reply from 192.168.11.250: Destination host unreachable.
Reply from 192.168.11.250: Destination host unreachable.
Reply from 192.168.11.250: Destination host unreachable.

Ping statistics for 192.168.11.248:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

danielkucera, 2024-01-02:
remove jumper

etmnidj1, 2024-01-02:
Ok how do I do that?

danielkucera, 2024-01-02:
catch it with pliers or tweezers and pull it up

etmnidj1, 2024-01-02:
I meant where is the jumper on the board??