danielkucera/esp-ebus #62: No data showing in HomeAssistent, errors in log

Peter-79, 2024-01-21:
Hi,
I want to connect my vaillant EcoTEC VCW346 (with VRC720) with my home assistant, but nothing seems to work.
I use this adapter: https://lectronz.com/products/ebus-to-wifi-adapter-module (I have version 5.1), and I succesfully upgraded the firmware to version 6.1. I tried several times to re-calibrate the variable resistor, that seems OK.
Every time I get the error "received unknown MS cmd: ...." or something like this (see attached logs)
What can cause this error? 
Notice that the adapter is powered by the eBUS.
Kind regards and many thanks for any help on this issue!

[ebusd.log](https://github.com/danielkucera/esp8266-arduino-ebus/files/14001490/ebusd.log)

brozikcz, 2024-01-24:
Did you try back to switch to older version of FW?

You can also try to use port `3333` without support of enhance protocol.

Peter-79, 2024-01-26:
Hi,
I first tried with an older firmware and I had the issues. Now with this firmware, the issue persists.
The WiFi-signal is 3 bars on the internal webpage of the device. (It is only a few meters between the device and the router).

Peter-79, 2024-01-27:
The return I get from the status-server is:
async mode: true
                software serial mode: true
                                          uptime: 475843713 ms
                                                              last_connect_time: 2230 ms
                                                                                        reconnect_count: 1
                                                                                                           rssi: -61 dBm
free_heap: 158076 B
                   reset_code: 15
                                 loop_duration: 22 us
max_loop_duration: 7651 us
version: v6.3
nbr arbitrations: 83356
nbr restarts1: 119
nbr restarts2: 28
nbr lost1: 2676
nbr lost2: 0
nbr won1: 80365
nbr won2: 168
nbr late: 1
nbr errors: 0
pwm_value: 130

Thank you for looking into this
Kind regards,
Peter

danielkucera, 2024-02-04:
-61 dBm is quite good. Can you please try the latest version v7.0 ?