danielkucera/esp-ebus #63: Power problem - Wolf

TFIPS, 2024-01-30:
When I connect the adapter to my Wolf boiler, the voltage on the eBus collapses. According to Wolf, a participant should use a maximum of 18mA. The adapter exceeds this limit. According to my measurements, these are always only peaks, for example, when the Wi-Fi is connected. Could the problem be resolved by using larger capacitors? If so, which of the two would need to be increased?

TFIPS, 2024-01-30:
Hello @danielkucera,
thank you very much for your prompt response. For now, I've added another 100 microfarad capacitor in parallel, and it's already working. Perhaps this issue could be addressed in the next hardware revision.

danielkucera, 2024-02-28:
Hi @TFIPS , did the capacitor solve the issue long term?

ilgio, 2024-11-05:
I bought the v6.3 board
[image]
. It keeps disconnecting from Wi-Fi. Even without bus connection

danielkucera, 2024-11-06:
Hi @ilgio ,

the adapter restarts automatically after 2 minutes without client connection. Do you have ebusd configured and connected to the adapter?

yarikuaks, 2025-02-02:
I have board version 6.3. When connecting the power supply from e-bus, jumper stands in bus vcc - D8 blinks constantly. The board does not connect to the access point. I tried to add a capacitor to the 3.3V 10V 470µf line, it did not give any result. If I remove the jumper and connect the power supply via type C, the board immediately connects to the access point.  Also, I'd like to take a moment to ask for what  JP 1 is needed?

danielkucera, 2025-02-02:
@yarikuaks , read the jumper description here: https://github.com/danielkucera/esp-arduino-ebus?tab=readme-ov-file#v52