danielkucera/esp-ebus #65: open http://esp-ebus.local/param to adjust PWM value

tomm10000, 2024-02-02:
Hello, Before uploading the software, I could set PWM, now I don't know how to enter this option.
I'm trying http://192.168.1.135/param but it doesn't work

danielkucera, 2024-02-02:
Hello, which HW version do you have?

tomm10000, 2024-02-03:
> Hello, which HW version do you have?

HW 6.1

```
Chip ID: 543204b78c50, ESP32-C3, rev 4
Hostname: ebus-b78c50
PCB: 20.0
Flash: 4194304
Version: v5.2-dev-3318-g8fc8f3f479
Reset reason: 1
eBUS r/w: 1/0
Host r/w: 4/20
Up time: 216
Free heap: 99168
ebusd connected: yes (inactive)
eBUS signal: no signal
```

danielkucera, 2024-02-03:
And which software did you upload?

tomm10000, 2024-02-03:
Through this website, I guess that FW is not correct
https://adapter.ebusd.eu/v5/firmware.en.html

This one update? firmware-fullflash-HW_v5.x-be7bb3.bin
https://github.com/danielkucera/esp8266-arduino-ebus/releases

danielkucera, 2024-02-03:
Yes, the first one is incorrect, completely different hardware.

Yes, use the one from releases on this page

tomm10000, 2024-02-03:
Can I upload via USB-C? + NodeMCU PyFlasher

tomm10000, 2024-02-03:
[image]

tomm10000, 2024-02-03:
[image]

tomm10000, 2024-02-03:
The jumper is also not understandable to me

> power from any adapter 5-24V: remove jumper and connect adapter to VCC and GND pins

[image]
[image]

danielkucera, 2024-02-03:
What is the question exactly?

tomm10000, 2024-02-03:
Which pins are these?
using a wire connect pins PROG and TP3 ( TP3 ?? )
I don't understand it, different markings

> connect your adaptor in a following way (dongle - module):
> 3V3 <-> 3V3
> TX <-> ESP-RX
> RX <-> ESP-TX
> GND <-> GND
>

danielkucera, 2024-02-03:
This is for the old version. For 6.1 just connect PROG and GND and flash over the USB-C port

tomm10000, 2024-02-03:
@danielkucera
Thank you for your help

Adds a link to the page where you can upload the firmware firmware-fullflash-HW_v5.x-be7bb3.bin
Address 0x0

https://adafruit.github.io/Adafruit_WebSerial_ESPTool/

danielkucera, 2024-02-03:
> now I have to upload the software ?

What do you mean by this?

tomm10000, 2024-02-03:
Does this software still require any update, because I only have this simple configuration panel

Now D1 works OK

> device: 192.168.1.135:3333, TCP, enhanced
> signal: acquired
> symbol rate: 14
> max symbol rate: 22
> reconnects: 1
> masters: 1
> messages: 11
> conditional: 0
> poll: 0
> update: 4
> address 31: master #8, ebusd
> address 36: slave #8, ebusd

danielkucera, 2024-02-03:
It's all there is. You will need to setup ebusd to connect to your adapter to decode the data.

tomm10000, 2024-02-03:
> `It's all there is. You will need to setup ebusd to connect to your adapter to decode the data.`

Thank you
There is a connection  :-)

> root@raspberrypi:~# ebusctl info
> version: ebusd 23.3.23.3-29-g2ea20ed8
> device: 192.168.1.135:3335, TCP, enhanced
> signal: acquired
> symbol rate: 25
> max symbol rate: 25
> reconnects: 0
> masters: 3
> messages: 11
> conditional: 0
> poll: 0
> update: 4
> address 03: master #11
> address 08: slave #11
> address 10: master #2
> address 31: master #8, ebusd
> address 36: slave #8, ebusd
> 
> root@raspberrypi:~# ebusctl scan result
> 08;Vaillant;BAI00;0600;5502
> 15;Vaillant;47000;0130;6002;21;12;49;0020108131;0082;011388;N2

closed 2024-02-04 by danielkucera