danielkucera/esp-ebus #67: gas boiler hermann

pabloMG85, 2024-03-03:
hy,hermann gas boiles semns to use the same ebus comunitation protocol,the ebus sheld may work on this manufacture

danielkucera, 2024-03-03:
Is this a question?

pabloMG85, 2024-03-03:
yes pls

danielkucera, 2024-03-03:
I cannot assure it will work but you can test and if you won't like it you can return it for full refund.

pabloMG85, 2024-03-03:
hy,so y have the 6.1 version y instaled the adaptor y have de d1 led flashlight very fast ,home asistant ebusd compliment is conects to the adaptor but it seams that dont discover any sensors

danielkucera, 2024-03-06:
Please provide ebusd logs

pabloMG85, 2024-03-06:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=feepoo3quaeDohw5ahg2aethoes2SahxeeV8ael9fooboopooRahthay3chai1oo --scanconfig --mqttjson --latency=10 --pollinterval=5 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.146:3335 --log=all:notice
2024-03-03 17:02:33.995 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.146:3335
2024-03-03 17:02:35.154 [bus notice] bus started with own address 31/36
2024-03-03 17:02:35.178 [bus notice] signal acquired
2024-03-03 17:02:35.327 [mqtt notice] connection established
2024-03-03 17:04:45.439 [main notice] update check: version 23.3 available
2024-03-04 17:04:55.824 [main notice] update check: version 23.3 available
2024-03-05 17:05:03.481 [main notice] update check: version 23.3 available
2024-03-06 17:05:06.136 [main notice] update check: version 23.3 available

danielkucera, 2024-03-06:
One problem I see right away, you need to use correct device string:
- `IP:3333` or
- `enh:IP:3335`

pabloMG85, 2024-03-06:
[image]

closed 2024-03-06 by danielkucera