danielkucera/esp-ebus #68: checking firmware to update

xjols, 2024-03-05:
Hello,

How I can check the running firmware version in the adapter ? I don't find this information in the info page. Only info about Wifi Manager and arduino...
I think that it will be v6.3 for Build Date ?

Thank you!
[image]

xjols, 2024-03-05:
Sorry I see the actual running version in Telnet port response:

```
async mode: true
software serial mode: true
uptime: 84940136 ms
last_connect_time: 1620 ms
reconnect_count: 1
rssi: -61 dBm
free_heap: 156520 B
reset_code: 3
loop_duration: 22 us
max_loop_duration: 6484 us
**version: v6.3**
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 130
```

xjols, 2024-03-05:
Now I just tried to update the firmware from v6.3 to the latest v7.1 from web page /update? After finished and reestart seems that the firmware is the old. Tried 2 times. Same result checked by telnet, the current version is 6.3.

I see a warning in upload page suggesting to upload from http://192.168.4.1/update, but this webpage is not working. 
[image]
Which is the correct procedure ?

Thanks!

danielkucera, 2024-03-05:
See #66

xjols, 2024-03-06:
Sorry for duplicate.

closed 2024-03-06 by xjols