danielkucera/esp-ebus #69: Compiling firmware

HuffYk, 2024-03-10:
Hi, quick question. When compiling firmware in vscode using basically

pio run -e esp32-c3-ota

should be generated "bin" file same as released here which can be used also in OTA upgrade? I see firmware file is binary different so not sure if it is correct one. Also size is different in few bytes so not sure if it is because of different platform.io version or firmware released here is created different way?

thanks

danielkucera, 2024-03-10:
Hello,

Not all library versions are strictly fixed so there might me minor differences.

Can you share your results and how do you compare them?

HuffYk, 2024-03-10:
Hi, this is 1st time for me to compile anything for esp :), but I just opened your project in vscode, it installed platform.io and in its UI I tried to build project. I tried also just with running "pio run -e esp32-c3-ota" and same results. PlatformIO core is latest one, 

Attached is generated firmware - its size is 16 bytes longer and when doing binary comparison to latest firmware-HW_v5.x-62ebb2.bin e.g. in Beyond Compare (using hex compare) there are many changes 

[firmware.zip](https://github.com/danielkucera/esp-arduino-ebus/files/14551031/firmware.zip)

danielkucera, 2024-03-10:
That is interesting. If you compare used libs, they are the same, see logs from CI:
https://github.com/danielkucera/esp-arduino-ebus/actions/runs/8071727098/job/22051966190#step:8:76

It's possible that the build is not idempotent - maybe there are some date strings compiled in (like compilation date/time etc.)

But your result looks okay and usable.

HuffYk, 2024-03-10:
I see CI/CD is for esp32-c3 (but anyhow bin files are not same even when I try to compile this one) so what exactly is difference between: esp32-c3, esp32-c3-ota, esp32-c3-ota-vpn ?

I suppose difference in binary comparison might be because of order of modules compiled into bin file. Only order of comilation is different when checking everything... all versions are indeed same. I even tried to use Python 3.12 (as before I used 3.10) 

So which exactly should I build to try update myself?  esp32-c3 or esp32-c3-ota ?

thanks

danielkucera, 2024-03-10:
The only difference is how they are flashed to the device when you click Upload in platformio:
- esp32-c3 - USB serial
- esp32-c3-ota - OTA via network
- esp32-c3-ota-vpn - OTA to a remote location I use for testing - will not work for you

HuffYk, 2024-03-10:
Understand, so if I want to flash it manually using web page, I can use same as is in CI: esp32-c3.

thanks

danielkucera, 2024-03-11:
You can use any of those.

HuffYk, 2024-03-11:
it worked ok, thanks!

closed 2024-03-11 by HuffYk