danielkucera/esp-ebus #70: Enhancement of status page

HuffYk, 2024-03-11:
I did a fork with small changes (mainly to status page):
- added check for ESP8266 or ESP32 and show it with pragma message (ends with pragma error otherwise)
- refactored status_string fnc to use String class
- added new info to status page:
  - ESP32 chip model and rev
  - ESP32 chip cores
  - ESP32 chip temperature
  - CPU chip speed
  - Flash chip speed
  - Flash size
  - SDK version
  - Wifi SID
  - Wifi IP address
 
if interested these are two relevant commits (forgot to put it into one PR):
- https://github.com/HuffYk/esp-arduino-ebus/commit/bde52397c2e8d22402e999a1b1226a02b9ba44c7
- https://github.com/HuffYk/esp-arduino-ebus/commit/30cc5c2f2481bf74364ac6d901936a34c1f44d0b

danielkucera, 2024-03-11:
Hello,
could you please create a PR?
Although this is broken now already, I'd like to have a uniform format for all the status variables:
```
lowercase_name_of_variable_with_underscores: value unit
```
Could you also please change that?

VladimirHirner, 2024-03-11:
sure, here it is: https://github.com/danielkucera/esp-arduino-ebus/pull/71