danielkucera/esp-ebus #72: Vaillant aroSTOR VWL BM 270/5 heat pump no usefull data can be collected from the eBUS

stadlerw, 2024-03-17:
i am using the EBUS to WiFi Adapter Module V6.1 with firmware version 7.1. The adapter ist connected to the internal eBus of the heatpump and attached to my internal network via wirless lan. Accessing the http status page is possible 

`async mode: true
software serial mode: true
uptime: 91822071 ms
last_connect_time: 579214 ms
reconnect_count: 3 
rssi: -71 dBm
free_heap: 152248 B
reset_code: 3
loop_duration: 65 us
max_loop_duration: 11554 us
version: v7.1
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 130`

I am using homeassistant and the eBUSd Addon, Current version: 23.2.3.  MQTT is enabled and basic informations can be received. When starting the adapter and trying to identify the Vaillant device no further parameters are collected or forwarded via MQTT.

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=OhJaec1quooPhoo5xoh4ain4thaishaiSee6uchooNaecheaRohngai5jegh3yae --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en/ --accesslevel=* --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=vaillant --device=enh:192.168.125.34:3335 --log=all:debug --scanconfig --lograwdata=data --latency=2000 --log=all:debug
2024-03-17 19:09:41.438 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.125.34:3335, enhanced
2024-03-17 19:09:41.438 [main info] loading configuration files from /config/ebusd-configuration/latest/en/
2024-03-17 19:09:41.438 [main debug] reading directory /config/ebusd-configuration/latest/en/
2024-03-17 19:09:41.438 [main debug] file type of /config/ebusd-configuration/latest/en/memory.csv is file
2024-03-17 19:09:41.438 [main debug] file type of /config/ebusd-configuration/latest/en/_templates.csv is file
2024-03-17 19:09:41.438 [main debug] file type of /config/ebusd-configuration/latest/en/broadcast.csv is file
2024-03-17 19:09:41.438 [main debug] file type of /config/ebusd-configuration/latest/en/vaillant is dir
2024-03-17 19:09:41.438 [main info] reading templates /
2024-03-17 19:09:41.439 [main info] read templates in /
2024-03-17 19:09:41.439 [main info] reading file memory.csv
2024-03-17 19:09:41.439 [main info] successfully read file memory.csv
2024-03-17 19:09:41.439 [main info] reading file broadcast.csv
2024-03-17 19:09:41.440 [main info] successfully read file broadcast.csv
2024-03-17 19:09:41.440 [main info] read config files, got 11 messages
2024-03-17 19:09:41.582 [bus notice] device status: resetting
2024-03-17 19:09:41.582 [bus notice] bus started with own address 31/36
2024-03-17 19:09:41.582 [main info] registering data handlers
2024-03-17 19:09:41.582 [mqtt info] mosquitto version 2.0.18 (compiled with 2.0.15)
2024-03-17 19:09:41.586 [mqtt debug] connection requested
2024-03-17 19:09:41.586 [main info] registered data handlers
2024-03-17 19:09:41.587 [bus notice] device status: reset
2024-03-17 19:09:41.587 [mqtt notice] connection established
2024-03-17 19:09:41.587 [mqtt debug] publish vaillant/global/version "ebusd 23.2.p20230716"
2024-03-17 19:09:41.587 [mqtt debug] publish vaillant/global/running true
2024-03-17 19:09:41.594 [bus notice] signal acquired
2024-03-17 19:09:44.100 [bus notice] <fffffffffffeffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:46.605 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:49.192 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:51.586 [main debug] performing regular tasks
2024-03-17 19:09:51.586 [mqtt debug] publish vaillant/global/scan "finished"
2024-03-17 19:09:51.653 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:54.201 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:57.164 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:09:57.593 [mqtt debug] publish vaillant/global/uptime 16
2024-03-17 19:09:57.595 [mqtt debug] publish homeassistant/binary_sensor/vaillant_global_running/config {
[... 83 lines cut ...]
2024-03-17 19:10:25.283 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:27.778 [bus notice] ...<fffffffffffffffffffffffffeffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:29.643 [mqtt debug] publish vaillant/global/uptime 48
2024-03-17 19:10:30.282 [bus notice] ...<fffffffffffffefffffffffffffffffffffffffffffffffeffffffffffffff...
2024-03-17 19:10:31.588 [main debug] performing regular tasks
2024-03-17 19:10:33.252 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:35.307 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:38.270 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:40.822 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:41.589 [main debug] performing regular tasks
2024-03-17 19:10:43.267 [bus notice] ...<fffffffffffffffffffffffffffeffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:45.661 [mqtt debug] publish vaillant/global/uptime 64
2024-03-17 19:10:45.833 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffe2...
2024-03-17 19:10:48.336 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:51.245 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:51.589 [main debug] performing regular tasks
2024-03-17 19:10:53.784 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:56.302 [bus notice] ...<ffbfffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:10:58.802 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff...
2024-03-17 19:11:01.364 [bus notice] ...<ffffffffffffffffffffffffffffffffffffffff7ffffffffffffffffffffe...
2024-03-17 19:11:01.590 [main debug] performing regular tasks
2024-03-17 19:11:01.677 [mqtt debug] publish vaillant/global/uptime 80
`

any idea what needs to be adjusted?

danielkucera, 2024-03-17:
Hi @stadlerw , on the first sight, it seems to me there is too many 0xFF symbols. 
Can you try to experiment with the PWM setting?

stadlerw, 2024-03-19:
Hi Daniel,

i did some testing with the PWM setting; the most reasonable value is 151.
Atted you will find the logs with the date received with PMW 151.

i got bus notice statements similar like this one

2024-03-19 20:17:03.431 [bus notice]
...<fefc00e080f0e0f8e080c0f880fef800f8c0e0f8fe00fc00e000fe80e080e0...

I am wondering why there are no master or slave device ids shown in the
adapters status page or in the logs.

I think its now the challenge to identify the corresponding configuration
files from Vaillant.

BR,
Wolfgang

danielkucera, 2024-03-20:
These values:
```
...<fefc00e080f0e0f8e080c0f880fef800f8c0e0f8fe00fc00e000fe80e080e0...
```
still don't seem correct.

stadlerw, 2024-03-22:
i checked the wiring and tested with an external power supply. This did not change the previous behaviour and the data quality. 
will look fo ran alternative solution

stadlerw, 2024-03-24:
i think the bus in the Vaillant aroSTOR VWL BM 270/5 heat pump is somehow proprietary and not a standard eBUS. 
The measured voltage does not comply to standard (HIGH (1)=> 15-24 VDC; LOW (0) => 9 – 12 VDC).

closed 2024-03-24 by stadlerw

danielkucera, 2024-03-25:
I have checked the installation manual and it doesn't mention ebus, most likely it uses a different interface.