danielkucera/esp-ebus #73: EBUS to WiFi Adapter Module V6.1 - errors in logs

raftheunis87, 2024-03-25:
Hello,

I've recently purchased your EBUS to WiFi Adapter Module V6.1 module.

I have updated the module to the latest firmare (firmware-HW_v5.x-62ebb2.bin).

I have a Vaillant FlexoTherm VFW 87/4 with a Vaillant VRC 700 thermostat. I have connected the module to the ebus wires behind the VRC 700. 

I was able to connect it to my local network, and I have configured the ebusd add-on in Home Assistant as follows:

`
scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: 192.168.30.44:3335
mode: enh
configpath: /config/ebusd-configuration/ebusd-2.1.x/en
`

The configs on the configpath are from https://github.com/jonesPD/ebusd-configuration

I get the following errors in the logs in the add-on:

`
2024-03-25 19:04:40.836 [update notice] received unknown MS cmd: 1008b516081000ffff03032130 / 0b00000003037930b04fe14a
2024-03-25 19:04:41.966 [update notice] received unknown MS cmd: 1008b516081002ffff03046130 / 0b0200020304613000569647
2024-03-25 19:04:42.840 [update notice] received unknown MS cmd: 1008b516081002ffff03044130 / 0b020102030441308051c247
2024-03-25 19:04:43.142 [update notice] received unknown MS cmd: 1008b516081000ffff03042130 / 0b0000020304793060af6f4a
2024-03-25 19:04:43.346 [update notice] sent poll-read hmu BuildingPumpHours QQ=31: 16373
2024-03-25 19:04:43.630 [update notice] received unknown MS cmd: 1008b516081002ffff02002130 / 0b0202ff020021307874e749
2024-03-25 19:04:43.910 [update notice] received read hmu State01 QQ=10: 19.5;20.0;10.750;-;54.5;off
2024-03-25 19:04:44.161 [update notice] received unknown MS cmd: 1076b5110101 / 09ff28c00aff6d0000ff
2024-03-25 19:04:44.431 [update notice] received unknown MS cmd: 1076b512030f0201 / 076c030000801500
2024-03-25 19:04:44.735 [update notice] received unknown MS cmd: 1008b516081002ffff02004130 / 0b0201ff0200413010849849
2024-03-25 19:04:45.005 [update notice] received update-write hmu SetMode QQ=10: 00 00 ff ff ff 07 00 00
2024-03-25 19:04:45.269 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2024-03-25 19:04:45.575 [update notice] received unknown MS cmd: 1008b516081002ffff02006130 / 0b0200ff02006130f0f33349
2024-03-25 19:04:45.880 [update notice] received unknown MS cmd: 1008b516081002ffff02008130 / 0b12ffff0200813000000000
2024-03-25 19:04:46.188 [update notice] received unknown MS cmd: 1008b516081002ffff0200a130 / 0b12ffff0200a13000000000
2024-03-25 19:04:46.493 [update notice] received unknown MS cmd: 1008b516081002ffff0200c130 / 0b12ffff0200c13000000000
2024-03-25 19:04:46.798 [update notice] received unknown MS cmd: 1008b516081002ffff0200e130 / 0b12ffff0200e13000000000
2024-03-25 19:04:46.956 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2024-03-25 19:04:47.196 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2024-03-25 19:04:47.338 [update notice] received unknown MS cmd: 1008b516081002ffff02000131 / 0b12ffff0200013100000000
2024-03-25 19:04:47.457 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2024-03-25 19:04:47.636 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2024-03-25 19:04:47.637 [main error] scan config 76: ERR: invalid position
2024-03-25 19:04:47.821 [update notice] received unknown MS cmd: 1008b516081002ffff02002131 / 0b12ffff0200213100000000
2024-03-25 19:04:48.682 [mqtt error] decode scan.76 id: ERR: invalid position
2024-03-25 19:04:49.337 [update notice] sent poll-read hmu BuildingPumpStarts QQ=31: 3913
2024-03-25 19:04:50.719 [update notice] received unknown MS cmd: 1008b516081002ffff02004131 / 0b12ffff0200413100000000
2024-03-25 19:04:51.524 [update notice] received unknown MS cmd: 1008b516081002ffff02006131 / 0b12ffff0200613100000000
2024-03-25 19:04:51.829 [update notice] received unknown MS cmd: 1008b516081002ffff02008131 / 0b12ffff0200813100000000
2024-03-25 19:04:52.134 [update notice] received unknown MS cmd: 1008b516081002ffff0200212e / 0b020eff0200212e38e1cf49
2024-03-25 19:04:52.441 [update notice] received unknown MS cmd: 1008b516081002ffff0200412e / 0b020dff0200412eb057aa49
2024-03-25 19:04:52.744 [update notice] received unknown MS cmd: 1008b516081002ffff0200612e / 0b020cff0200612eb8a89349
2024-03-25 19:04:53.054 [update notice] received unknown MS cmd: 1008b516081002ffff0200812e / 0b020bff0200812ee0494549
2024-03-25 19:04:53.357 [update notice] received unknown MS cmd: 1008b516081002ffff0200a12e / 0b020aff0200a12e80037948
2024-03-25 19:04:53.663 [update notice] received unknown MS cmd: 1008b516081002ffff0200c12e / 0b0209ff0200c12e40192848
2024-03-25 19:04:53.928 [update notice] received read hmu State01 QQ=10: 19.5;20.0;10.750;-;54.5;off
2024-03-25 19:04:54.196 [update notice] received unknown MS cmd: 1076b5110101 / 09ff28c00aff6d0000ff
2024-03-25 19:04:54.460 [update notice] received unknown MS cmd: 1076b512030f0201 / 076c030000801500
2024-03-25 19:04:54.763 [update notice] received unknown MS cmd: 1008b516081002ffff0200e12e / 0b0208ff0200e12e0010f147
2024-03-25 19:04:55.030 [update notice] received update-write hmu SetMode QQ=10: 00 00 ff ff ff 07 00 00
2024-03-25 19:04:55.275 [update notice] sent poll-read hmu CompressorHours QQ=31: 5495
2024-03-25 19:04:55.464 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2024-03-25 19:04:55.769 [update notice] received unknown MS cmd: 1008b516081002ffff0200012f / 0b0207ff0200012f407e2f48
2024-03-25 19:04:56.073 [update notice] received unknown MS cmd: 1008b516081002ffff0200212f / 0b0206ff0200212f80f54a48
[... 18 lines cut ...]
2024-03-25 19:05:12.641 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2024-03-25 19:05:12.799 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2024-03-25 19:05:12.960 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2024-03-25 19:05:12.960 [main error] scan config 76: ERR: invalid position
2024-03-25 19:05:13.233 [update notice] sent poll-read hmu CompressorStarts QQ=31: 4847
2024-03-25 19:05:13.681 [update notice] received read hmu State01 QQ=10: 19.5;20.0;10.750;-;54.5;off
2024-03-25 19:05:13.949 [update notice] received unknown MS cmd: 1076b5110101 / 09ff28c00aff6d0000ff
2024-03-25 19:05:13.970 [mqtt error] decode scan.76 id: ERR: invalid position
2024-03-25 19:05:14.219 [update notice] received unknown MS cmd: 1076b512030f0201 / 076c030000801500
2024-03-25 19:05:14.478 [update notice] received update-write hmu SetMode QQ=10: 00 00 ff ff ff 07 00 00
2024-03-25 19:05:19.251 [update notice] sent poll-read hmu CondensationTemp QQ=31: 13.5
2024-03-25 19:05:25.108 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2024-03-25 19:05:25.290 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2024-03-25 19:05:25.484 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2024-03-25 19:05:25.657 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2024-03-25 19:05:25.658 [main error] scan config 76: ERR: invalid position
2024-03-25 19:05:25.902 [update notice] sent poll-read hmu CondensorInletTemp QQ=31: -0.1
2024-03-25 19:05:26.667 [mqtt error] decode scan.76 id: ERR: invalid position
2024-03-25 19:05:31.230 [update notice] sent poll-read hmu CondensorOutletTemp QQ=31: -0.1
`
I also get an error on both the VRC 700 and on my heat pump.

Any idea what I can do to resolve this?

Thanks in advance.

danielkucera, 2024-03-25:
Hello @raftheunis87 ,
I am not sure how this line reflects the configuration:
```
network_device: 192.168.30.44:3335
```
but for port 3335 you should use enhanced protocol so in config it shold be: `enh:192.168.30.44:3335`

Can you confirm this is the case?

raftheunis87, 2024-03-25:
Hey @danielkucera ,

That enh is configured under the 'mode' part in the config.

danielkucera, 2024-03-25:
Okay then. In the ebusd log I don't see many problems but your devices complaining are not okay. 
Can you check if powering the adapter from USB resolves the problems with your controller and pump?