danielkucera/esp-ebus #75: VRSOL 2.0 + V6.1 adapter + HAOS + EBUSD

telemake69, 2024-08-23:
Hello all

I am trying to connect my WRSOL 2.0 from Weishaupt to my domotic system (HAOS running over Raspberry).
And still facing "scan config fc: ERR: arbitration lost"

**Adapter**
EBUS to WiFi Adapter Module V6.1 (from ELECROW) 
Adapter connected to my network via Wifi
[image]
[image]

**Domotic**
HA OS over Raspberry 4B
Raspberry connecte to my network via Ethernet LAN

**Ebusd configuration**
scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: 192.168.1.51:3334

**Ebusd logs**
2024-08-23 08:35:39.839 [main notice] SIGTERM received
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=tahXidoow4Ohgoh8ChaeghaepheibeuX0ahpheiquo5wor7ieDai0lee8quo4Aip --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.51:3334 --log=all:notice
2024-08-23 08:35:50.135 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.51:3334
2024-08-23 08:35:52.105 [bus notice] bus started with own address 31/36
2024-08-23 08:35:52.217 [bus notice] signal acquired
2024-08-23 08:35:52.516 [mqtt notice] connection established
2024-08-23 08:38:02.371 [main notice] update check: version 23.3 available
2024-08-23 08:40:05.975 [bus notice] new master f7, master count 2
2024-08-23 08:40:13.757 [main error] scan config fc: ERR: arbitration lost
2024-08-23 08:40:27.068 [main error] scan config fc: ERR: arbitration lost
2024-08-23 08:40:35.366 [bus notice] device invalid

"Mosquitto logs"
024-08-23 08:35:52: New connection from 172.30.33.3:41096 on port 1883.
2024-08-23 08:35:52: New client connected from 172.30.33.3:41096 as ebusd_23.2_277 (p1, c1, k60, u'addons').
2024-08-23 08:36:01: New connection from 172.30.32.2:39394 on port 1883.
2024-08-23 08:36:01: Client <unknown> closed its connection.
2024-08-23 08:36:28: Saving in-memory database to /data//mosquitto.db.
2024-08-23 08:38:01: New connection from 172.30.32.2:57510 on port 1883.
2024-08-23 08:38:01: Client <unknown> closed its connection.
2024-08-23 08:40:01: New connection from 172.30.32.2:34392 on port 1883.
2024-08-23 08:40:01: Client <unknown> closed its connection.
2024-08-23 08:42:01: New connection from 172.30.32.2:46726 on port 1883.
2024-08-23 08:42:01: Client <unknown> closed its connection.

I have followed Daniel's Trouble shooting recommandations
- The adapter is really close to my router (Wifi signal is at maxium level)
- On the adapter, D7 (Power) is lighting, D1 (RX) is flashing and FROM-EBUS led is lighting (and besomes some times OFF very quiclky)
- I can see unreadable date using telnet
- But still no data on ebusd and still "scan config fc: ERR: arbitration lost" issues

I have try several configuration for latency, from 0 to 10 000 (max supported by Ebusd), unsucessfully.

PS : 2 strange things
- Why FROM-EBUS led is sometimes getting OFF very quickly ?
- Where does the adress "172.30.32.2:37218" in Mosquitto come from (while my local network is more 192.168.1.xxx) ?

Can you please help ?

danielkucera, 2024-08-23:
Hi @telemake69 ,
first of all 3334 is **listen only port**
try to use port 3335 with enhanced proto:  ebusd config: `enh:esp-ebus.local:3335`

telemake69, 2024-08-23:
Hi Daniel,
Listening only was my first approach indeed.

**With enh:192.168.1.51:3335**
2024-08-23 09:11:18.224 [main notice] SIGTERM received
2024-08-23 09:11:18.287 [mqtt debug] publish ebusd/global/uptime 432
2024-08-23 09:11:19.077 [mqtt debug] publish ebusd/global/signal false
2024-08-23 09:11:19.077 [mqtt debug] publish ebusd/global/scan 
2024-08-23 09:11:19.148 [main notice] ebusd stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=tahXidoow4Ohgoh8ChaeghaepheibeuX0ahpheiquo5wor7ieDai0lee8quo4Aip --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.51:3335 --log=all:notice
2024-08-23 09:11:28.453 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.51:3335, enhanced
2024-08-23 09:11:29.631 [bus notice] device status: resetting
2024-08-23 09:11:29.632 [bus notice] bus started with own address 31/36
2024-08-23 09:11:29.647 [bus notice] device status: reset
2024-08-23 09:11:29.647 [bus notice] signal acquired
2024-08-23 09:11:30.045 [mqtt notice] connection established
2024-08-23 09:13:39.914 [main notice] update check: version 23.3 available
2024-08-23 09:15:06.021 [bus notice] new master f7, master count 2
2024-08-23 09:15:10.498 [main error] scan config fc: ERR: read timeout
2024-08-23 09:16:12.845 [bus notice] scan fc: ;TEM;17986;2422;1112
2024-08-23 09:16:12.845 [update notice] store fc ident: done
2024-08-23 09:16:12.845 [update notice] sent scan-read scan.fc  QQ=31: TEM;17986;2422;1112
2024-08-23 09:16:12.845 [bus notice] scan fc: ;TEM;17986;2422;1112
2024-08-23 09:16:13.104 [main error] HTTP failure: receive error: 404 Not Found
2024-08-23 09:16:13.104 [main error] unable to load scan config fc: list files in tem ERR: element not found
2024-08-23 09:16:13.104 [main error] scan config fc: ERR: element not found

It seems I have less logs but still no data in reception...

danielkucera, 2024-08-23:
Hi @telemake69 ,
this makes more sense. It seems like there is not a valid config for your device `TEM;17986;2422;1112`.
You can get some info about the process here: https://github.com/john30/ebusd/wiki/4.7.-Automatic-configuration

telemake69, 2024-08-23:
Thanks i will have a look on it
I thought configuration was more to customize data echange (to be able to decode them) but not to initialize the communication with the device.
Is it not the case ?

danielkucera, 2024-08-24:
You need to have config to be able to interpret register data and how to access them.
You can try to use config for a similar device if there is no exact match.