danielkucera/esp-ebus #80: Error message on thermostat after connecting adapter

Proim, 2024-10-11:
I have the V6.3 and connected it to the ebus of my system (Vaillant VR71). It's succesfully connected with the Home Assistant add-on and I'm receiving data. However, I noticed ever since connecting it to the ebus an error message is present on my thermostat (Vaillant VRC720). In Dutch it says "Signaal sensor S9 FM5 ongeldig', or in English 'Signal sensor S9 FM5 invalid'. There's is nothing connected to the S9 input on the VR71 itself (only S1, S2 and S3 are connected). Has anyone else experienced this, anything I could try?

I don't notice anything different if the adapter is powered externally or not.

danielkucera, 2024-10-14:
Hi @Proim ,

did you also remove the jumper when powering the adapter externally?

Proim, 2024-10-15:
I didn't realize I had to change something on the adapter. Am I correct it is this one? blob:https://imgur.com/8bfd3599-d87d-4ad2-be74-65fc43a91a5d

It seems that restarting might also have done something. After a first restart of the thermostat the error was gone, but came back after 5 or 10 minutes, but after another restart it has been gone for at least a night now. Still, if you can confirm the position of the jumper I can take it off just to be sure.

danielkucera, 2024-10-15:
The image link doesnt work for me.
You can upload pictures directly here by dragging into text box.

Proim, 2024-10-15:
Sorry, still very new to github :) 
This should be ok now:
[image]

danielkucera, 2024-10-15:
Not that thing. Check the marking J4.

Proim, 2024-10-15:
I see thanks. Safest is probably to disconnect everything, take out the jumper and connect again? (I'm mainly asking because I'm happy that finally the adapter connected to wifi and data is being transmitted, so don't want to mess it up).

danielkucera, 2024-10-15:
Yes, that sounds reasonable.

Proim, 2024-10-21:
Sorry for the late response. I've disconnected everything, took out the jumper, connected again and all is working. The error itselfs was gone again so hopefully it stays that way. Thanks for thinking along.

closed 2024-10-21 by Proim