danielkucera/esp-ebus #81: esp-ebus V6.1 with no signal

docava65, 2024-10-15:
Hi Daniel,

I still have the disconnecting issue. Regardles of the configuration changes made in ebusd I get always no signal and disconnections, 
This is the latest ebusd config: EBUSD_OPTS="--scanconfig -d enh:192.168.129.53:3335 -p 3335 --latency=20000"

doca@RPi3:/etc/default $ sudo systemctl status ebusd.service
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
     Loaded: loaded (/lib/systemd/system/ebusd.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-10-15 13:04:22 CEST; 33min ago
    Process: 5823 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=0/SUCCESS)
   Main PID: 5825 (ebusd)
      Tasks: 4 (limit: 779)
        CPU: 1.144s
     CGroup: /system.slice/ebusd.service
             └─5825 /usr/bin/ebusd --scanconfig -d enh:192.168.129.53:3335 -p 3335 --latency=20000

Oct 15 13:04:22 RPi3 systemd[1]: Starting ebusd.service - ebusd, the daemon for communication with eBUS heating systems....
Oct 15 13:04:22 RPi3 systemd[1]: Started ebusd.service - ebusd, the daemon for communication with eBUS heating systems..

doca@RPi3:/var/log $ sudo lsof -i -P -n | grep LISTEN
vncserver  628  root   11u  IPv6   4093      0t0  TCP *:5900 (LISTEN)
vncserver  628  root   12u  IPv4   4094      0t0  TCP *:5900 (LISTEN)
sshd       677  root    3u  IPv4  14842      0t0  TCP *:22 (LISTEN)
sshd       677  root    4u  IPv6  14844      0t0  TCP *:22 (LISTEN)
cupsd     2576  root    6u  IPv6  66542      0t0  TCP [::1]:631 (LISTEN)
cupsd     2576  root    7u  IPv4  66543      0t0  TCP 127.0.0.1:631 (LISTEN)
ebusd     5825  root    5u  IPv4 104852      0t0  TCP *:3335 (LISTEN)

doca@RPi3:/etc/default $ ping 192.168.129.53
PING 192.168.129.53 (192.168.129.53) 56(84) bytes of data.
64 bytes from 192.168.129.53: icmp_seq=1 ttl=255 time=92.8 ms
64 bytes from 192.168.129.53: icmp_seq=2 ttl=255 time=115 ms
64 bytes from 192.168.129.53: icmp_seq=3 ttl=255 time=35.7 ms
64 bytes from 192.168.129.53: icmp_seq=4 ttl=255 time=58.4 ms
--- 192.168.129.53 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 35.749/75.498/115.050/30.557 ms

doca@RPi3:/etc/default $ sudo ebusctl info
version: ebusd 23.3.23.3
update check: OK
device: 192.168.129.53:3335, TCP
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

doca@RPi3:/etc/default $ sudo ebusctl find -V
broadcast datetime = no data stored [ZZ=fe, passive read]
broadcast error = no data stored [ZZ=fe, passive read]
broadcast id = no data stored [ZZ=fe, passive read]
broadcast id = no data stored [ZZ=any, active read]
broadcast signoflife = no data stored [ZZ=fe, passive read]
memory eeprom = no data stored [ZZ=any, active read]
memory ram = no data stored [ZZ=any, active read]

doca@RPi3:/etc/default $ sudo ebusd -f -d 192.168.129.53:3335
2024-10-15 13:34:57.471 [bus notice] device status: transport opened
2024-10-15 13:34:57.471 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: 192.168.129.53:3335, TCP
2024-10-15 13:34:58.170 [bus notice] bus started with own address 31/36
2024-10-15 13:34:58.170 [network error] unable to start TCP server on port 8888: error 98
2024-10-15 13:34:58.170 [bus error] device status: transport closed
2024-10-15 13:34:58.171 [bus notice] device invalid
2024-10-15 13:35:03.308 [bus notice] device status: transport opened
2024-10-15 13:35:03.308 [bus notice] re-opened 192.168.129.53:3335
....

I've also tried with different PWM values and have tried different firmware versions, same status: no signal

async mode: true
software serial mode: true
uptime: 176483 ms
last_connect_time: 275 ms
reconnect_count: 1 
rssi: -58 dBm
free_heap: 158784 B
reset_code: 3
loop_duration: 50 us
max_loop_duration: 3966427 us
version: v7.1
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 180

After a while, the device can't be contacted

doca@RPi3:/var/log $ sudo ebusctl info
error connecting to localhost:8888

I don't understand why it tries to connect to port 8888, I've included the 3335 port in ebusd???
Could you please also tell me what is the D8 led is actually doing, I see it stays still for almost 8 seconds then off and again lit for 8 sec.?

Hopefully the above information give you hints on what is going on. I think, this device is malfunctioning, see the image below.
[image]

danielkucera, 2024-10-15:
Hi @docava65 ,

can you try to power the device from usb adapter? Don't forget to remove the jumper as described in readme.
Also, can you please provide ebusd logs? Please attach them as txt file.

docava65, 2024-10-15:
Hi Daniel,

I've powered up the device from an USB adapter, still the same: no signal.
Please find attached today's ebusd log

Rgds,
________________________
Daniel Ocampo
*CISA, CISM*
***@***.*** ***@***.***>*
*Linkedin <https://be.linkedin.com/in/danielocampo>*

2024-10-15 00:00:05.155 [main notice] SIGHUP received
2024-10-15 00:00:16.025 [bus error] device status: transport closed
2024-10-15 00:00:16.025 [bus notice] device invalid
2024-10-15 00:00:21.163 [bus notice] device status: transport opened
2024-10-15 00:00:21.164 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:00:21.168 [bus notice] device status: reset
2024-10-15 00:03:41.330 [bus error] device status: transport closed
2024-10-15 00:03:41.330 [bus notice] device invalid
2024-10-15 00:03:46.409 [bus notice] device status: transport opened
2024-10-15 00:03:46.409 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:03:46.415 [bus notice] device status: reset
2024-10-15 00:07:05.513 [bus error] device status: transport closed
2024-10-15 00:07:05.513 [bus notice] device invalid
2024-10-15 00:07:10.659 [bus notice] device status: transport opened
2024-10-15 00:07:10.659 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:07:10.664 [bus notice] device status: reset
2024-10-15 00:10:29.702 [bus error] device status: transport closed
2024-10-15 00:10:29.702 [bus notice] device invalid
2024-10-15 00:10:34.844 [bus notice] device status: transport opened
2024-10-15 00:10:34.847 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:10:34.851 [bus notice] device status: reset
2024-10-15 00:13:54.089 [bus error] device status: transport closed
2024-10-15 00:13:54.089 [bus notice] device invalid
2024-10-15 00:13:59.233 [bus notice] device status: transport opened
2024-10-15 00:13:59.233 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:13:59.237 [bus notice] device status: reset
2024-10-15 00:17:16.126 [bus error] device status: transport closed
2024-10-15 00:17:16.126 [bus notice] device invalid
2024-10-15 00:17:21.273 [bus notice] device status: transport opened
2024-10-15 00:17:21.273 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:17:21.277 [bus notice] device status: reset
2024-10-15 00:20:41.332 [bus error] device status: transport closed
2024-10-15 00:20:41.332 [bus notice] device invalid
2024-10-15 00:20:46.405 [bus notice] device status: transport opened
2024-10-15 00:20:46.406 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:20:46.410 [bus notice] device status: reset
2024-10-15 00:24:05.517 [bus error] device status: transport closed
2024-10-15 00:24:05.517 [bus notice] device invalid
2024-10-15 00:24:10.661 [bus notice] device status: transport opened
2024-10-15 00:24:10.661 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:24:10.665 [bus notice] device status: reset
2024-10-15 00:27:30.111 [bus error] device status: transport closed
2024-10-15 00:27:30.111 [bus notice] device invalid
2024-10-15 00:27:35.256 [bus notice] device status: transport opened
2024-10-15 00:27:35.256 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:27:35.260 [bus notice] device status: reset
2024-10-15 00:30:52.353 [bus error] device status: transport closed
2024-10-15 00:30:52.353 [bus notice] device invalid
2024-10-15 00:30:57.403 [bus notice] device status: transport opened
2024-10-15 00:30:57.403 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:30:57.407 [bus notice] device status: reset
2024-10-15 00:34:17.662 [bus error] device status: transport closed
2024-10-15 00:34:17.662 [bus notice] device invalid
2024-10-15 00:34:22.807 [bus notice] device status: transport opened
2024-10-15 00:34:22.807 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 00:34:22.813 [bus notice] device status: reset
2024-10-15 00:37:40.052 [bus error] device status: transport closed
2024-10-15 00:37:40.052 [bus notice] device invalid
[... 945 lines cut ...]
2024-10-15 11:20:28.680 [bus notice] device status: transport opened
2024-10-15 11:20:28.680 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:20:28.685 [bus notice] device status: reset
2024-10-15 11:23:49.154 [bus error] device status: transport closed
2024-10-15 11:23:49.154 [bus notice] device invalid
2024-10-15 11:23:54.298 [bus notice] device status: transport opened
2024-10-15 11:23:54.298 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:23:54.303 [bus notice] device status: reset
2024-10-15 11:27:11.802 [bus error] device status: transport closed
2024-10-15 11:27:11.802 [bus notice] device invalid
2024-10-15 11:27:16.946 [bus notice] device status: transport opened
2024-10-15 11:27:16.946 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:27:16.950 [bus notice] device status: reset
2024-10-15 11:30:36.192 [bus error] device status: transport closed
2024-10-15 11:30:36.192 [bus notice] device invalid
2024-10-15 11:30:41.337 [bus notice] device status: transport opened
2024-10-15 11:30:41.337 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:30:41.341 [bus notice] device status: reset
2024-10-15 11:34:00.992 [bus error] device status: transport closed
2024-10-15 11:34:00.992 [bus notice] device invalid
2024-10-15 11:34:06.136 [bus notice] device status: transport opened
2024-10-15 11:34:06.136 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:34:06.140 [bus notice] device status: reset
2024-10-15 11:37:23.234 [bus error] device status: transport closed
2024-10-15 11:37:23.234 [bus noti

danielkucera, 2024-10-15:
Can you please provide log with all options specified here?
https://github.com/danielkucera/esp-arduino-ebus?tab=readme-ov-file#nothing-helps-i-need-support

docava65, 2024-10-15:
Hi Daniel,

Attached you will find the log. Below you can find the status of ebusd when
started with the required options

***@***.***:/etc/default $ sudo systemctl status ebusd.service
● ebusd.service - ebusd, the daemon for communication with eBUS heating
systems.
     Loaded: loaded (/lib/systemd/system/ebusd.service; enabled; preset:
enabled)
     Active: active (running) since Tue 2024-10-15 21:34:36 CEST; 7s ago
    Process: 1302 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited,
status=0/SUCCESS)
   Main PID: 1304 (ebusd)
      Tasks: 4 (limit: 779)
        CPU: 441ms
     CGroup: /system.slice/ebusd.service
             └─1304 /usr/bin/ebusd --scanconfig -d enh:192.168.129.53:3335
--lograwdata=data --latency=2000 --log=all debug

Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.424 [network debug]
SSL state 4=SSLv3/TLS read server certificate: type 0x1001=loop,connect,
val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.426 [network debug]
SSL state 39=TLSv1.3 read server certificate verify: type
0x1001=loop,connect, val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.426 [network debug]
SSL state 11=SSLv3/TLS read finished: type 0x1001=loop,connect, val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.426 [network debug]
SSL state 16=SSLv3/TLS write change cipher spec: type 0x1001=loop,connect,
val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.427 [network debug]
SSL state 18=SSLv3/TLS write finished: type 0x1001=loop,connect, val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.427 [network debug]
SSL state 1=SSL negotiation finished successfully: type 0x20=done, val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.427 [network debug]
SSL state 1=SSL negotiation finished successfully: type
0x1002=exit,connect, val 1=?
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.427 [main info]
configPath URL is valid
Oct 15 21:34:36 RPi3 ebusd[1302]: 2024-10-15 21:34:36.427 [network debug]
SSL state 1=SSL negotiation finished successfully: type 0x4008=write,alert,
val 256=close notify
Oct 15 21:34:36 RPi3 systemd[1]: Started ebusd.service - ebusd, the daemon
for communication with eBUS heating systems..

***@***.***:/etc/default $ sudo ebusctl info
version: ebusd 23.3.23.3
device: 192.168.129.53:3335, TCP, enhanced
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

________________________
Daniel Ocampo
*CISA, CISM*
***@***.*** ***@***.***>*
*Linkedin <https://be.linkedin.com/in/danielocampo>*

2024-10-15 00:00:05.155 [main notice] SIGHUP received
2024-10-15 00:00:16.025 [bus error] device status: transport closed
2024-10-15 00:00:16.025 [bus notice] device invalid
2024-10-15 00:00:21.163 [bus notice] device status: transport opened
2024-10-15 00:00:21.164 [bus notice] re-opened 192.168.129.53:3335
[... 961 lines cut ...]
2024-10-15 10:56:35.162 [bus error] device status: transport closed
2024-10-15 10:56:35.162 [bus notice] device invalid
2024-10-15 10:56:40.309 [bus notice] device status: transport opened
2024-10-15 10:56:40.310 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 10:56:40.314 [bus notice] device status: reset
2024-10-15 11:00:00.782 [bus error] device status: transport closed
2024-10-15 11:00:00.782 [bus notice] device invalid
2024-10-15 11:00:05.926 [bus notice] device status: transport opened
2024-10-15 11:00:05.926 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:00:05.930 [bus notice] device status: reset
2024-10-15 11:03:23.123 [bus error] device status: transport closed
2024-10-15 11:03:23.123 [bus notice] device invalid
2024-10-15 11:03:28.268 [bus notice] device status: transport opened
2024-10-15 11:03:28.269 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:03:28.273 [bus notice] device status: reset
2024-10-15 11:06:48.947 [bus error] device status: transport closed
2024-10-15 11:06:48.947 [bus notice] device invalid
2024-10-15 11:06:54.091 [bus notice] device status: transport opened
2024-10-15 11:06:54.091 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:06:54.097 [bus notice] device status: reset
2024-10-15 11:10:11.535 [bus error] device status: transport closed
2024-10-15 11:10:11.536 [bus notice] device invalid
2024-10-15 11:10:16.639 [bus notice] device status: transport opened
2024-10-15 11:10:16.639 [bus notice] re-opened 192.168.129.53:3335
2024-10-15 11:10:16.643 [bus notice] device

danielkucera, 2024-10-16:
Hi @docava65 ,

it looks like the log didn't get forwarded to the issue. Can you upload it manually?

docava65, 2024-10-16:
Hi Daniel,I’m travelling abroad for 10 days,I can’t do that now, I’ll. come back later on this, thanks for your swift support.Sent from my mobile