danielkucera/esp-ebus #82: Device not starting

realrewa, 2024-11-02:
Hi!
While powered via usb, 3.3v pin has 3.3v.
While connected to ebus, same pin has 2.7v, esp led no active, device not starting.
D1 led blinking with d7 at same time.
Ebus voltage 21.7v while not connected, 20.7v with connected device.
Any help will be appreciated.

danielkucera, 2024-11-03:
Hi @realrewa ,
most likely, this is your issue: https://github.com/danielkucera/esp-arduino-ebus?tab=readme-ov-file#the-adapter-is-loosing-connectivity-breaking-other-ebus-components-communication-d7-is-blinking-wifi-is-disappearing-after-connecting-devices-on-the-bus-show-error-status-or-other-intermittent-failures

You probably need to power the adapter using USB, as the bus cannot supply enough power.

realrewa, 2024-11-03:
Sorry, missed ebus-vcc jumper from readme. All works fine. Thank you!

closed 2024-11-03 by realrewa