danielkucera/esp-ebus #84: factory reset

ilgio, 2024-11-16:
How can I do a factory reset? To remove from memory the wifi that connects to the furthest AP and not to the closest one

danielkucera, 2024-11-16:
https://github.com/danielkucera/esp-arduino-ebus?tab=readme-ov-file#config-reset

ilgio, 2024-11-16:
yes but what is pin RESET_PIN: TO-EBUS (GPIO20) ?
[image]
[image]

danielkucera, 2024-11-16:
yes

ilgio, 2024-11-16:
where is pin RESET_PIN: TO-EBUS (GPIO20) ?

danielkucera, 2024-11-16:
yes

ilgio, 2024-11-16:
Yes what?

danielkucera, 2024-11-16:
[image]