danielkucera/esp-ebus #87: esp ebus v6.3 firmware 7.1 wifi password length restrictions?

baahver, 2024-11-18:
I just received my esp ebus adapter v6.3. But am having trouble connecting to wifi. 
If I connect it to my guest network, with a password length of 20 then it is possible to connect. But the guest network is not accessible from lan. 
When I try to use my lan wifi with a password of 36 characters, this adapter is not able to make a connection. 
I assume this is because of this very long wifi password? 
Changing this password is not something I could manage easily. 
Is the length of the wifi password the cause of this connection trouble? Is there a solution possible?

danielkucera, 2024-11-18:
Hi @baahver ,
interesting issue, can you please try firmware from here? 
https://github.com/danielkucera/esp-arduino-ebus/suites/31019094814/artifacts/2201863868

baahver, 2024-11-18:
Yes I can. But what firmware should I use from that file? On the back there is printed: esp-ebus v6.3 danman.eu

baahver, 2024-11-18:
I just installed this firmware but still the same issue. 
I create a access point on my linux server, with a smaller password. 24 characters. And this is working.
But I would rather use the default wifi ssid.
Any other suggestions?

danielkucera, 2024-11-18:
I just tested the updated firmware and it works with 40chars password. Can you send me what is shown in adapter status?

baahver, 2024-11-19:
```
async mode: true
software serial mode: true
uptime: 9064 ms
last_connect_time: 430 ms
reconnect_count: 1 
rssi: -63 dBm
free_heap: 164656 B
reset_code: 1
loop_duration: 9 us
max_loop_duration: 4071745 us
version: v7.1
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 130
```

danielkucera, 2024-11-19:
You did not update the firmware correctly, it still shows `version: v7.1`.
After upgrade, this field should be empty

danielkucera, 2024-11-20:
Show your config please.

baahver, 2024-12-06:
Hi Daniel,
I'm starting ebusd in docker with:
```
  ebusd:
    container_name: ebusd
    image: john30/ebusd
    ports:
      - 8888:8888
      - 9797:8080
    environment:
      EBUSD_CONFIGPATH: '/config/ebusd-configuration/latest/en'
      EBUSD_CONFIGLANG: 'en'
      EBUSD_LATENCY: '40'
      EBUSD_SCANCONFIG: ''
      EBUSD_DEVICE: 'enh:192.168.188.104:3335'
      EBUSD_MQTTPORT: '1883'
      EBUSD_MQTTHOST: '172.25.0.13'
      EBUSD_MQTTUSER: 'mosquitto'
      EBUSD_MQTTPASS: 'mqttwachtwoord'
      EBUSD_MQTTCLIENTID: 'ebusd'
      EBUSD_MQTTINT: /etc/ebusd/mqtt-hassio.cfg
      EBUSD_PORT: 8888
      EBUSD_HTTPPORT: 8080
      EBUSD_LOGLEVEL: 'debug'
    networks:
      eigennet:
        ipv4_address: 172.25.0.22
    restart: unless-stopped
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /home/ahbart/docker/ebusd/config:/config
```

But I do not get any ebus data from my Vaillant heating system.
In ebusd log I see: 
```
2024-12-06 15:19:25.737 [main debug] performing regular tasks
2024-12-06 15:19:28.620 [mqtt debug] publish ebusd/global/uptime 2368
2024-12-06 15:19:28.620 [mqtt debug] publish ebusd/global/uptime 2368
2024-12-06 15:19:35.738 [main debug] performing regular tasks
2024-12-06 15:19:44.638 [mqtt debug] publish ebusd/global/uptime 2384
2024-12-06 15:19:44.638 [mqtt debug] publish ebusd/global/uptime 2384
2024-12-06 15:19:45.738 [main debug] performing regular tasks
2024-12-06 15:19:55.739 [main debug] performing regular tasks
2024-12-06 15:20:00.654 [mqtt debug] publish ebusd/global/uptime 2400
2024-12-06 15:20:00.654 [mqtt debug] publish ebusd/global/uptime 2400
2024-12-06 15:20:05.739 [main debug] performing regular tasks
2024-12-06 15:20:15.739 [main debug] performing regular tasks
2024-12-06 15:20:16.671 [mqtt debug] publish ebusd/global/uptime 2416
2024-12-06 15:20:16.671 [mqtt debug] publish ebusd/global/uptime 2416
2024-12-06 15:20:25.740 [main debug] performing regular tasks
2024-12-06 15:20:32.688 [mqtt debug] publish ebusd/global/uptime 2432
2024-12-06 15:20:32.688 [mqtt debug] publish ebusd/global/uptime 2432
2024-12-06 15:20:35.740 [main debug] performing regular tasks
2024-12-06 15:20:45.740 [main debug] performing regular tasks
2024-12-06 15:20:48.704 [mqtt debug] publish ebusd/global/uptime 2448
2024-12-06 15:20:48.704 [mqtt debug] publish ebusd/global/uptime 2448
2024-12-06 15:20:55.740 [main debug] performing regular tasks
2024-12-06 15:20:55.740 [main error] update check connect error
2024-12-06 15:21:04.719 [mqtt debug] publish ebusd/global/uptime 2464
2024-12-06 15:21:04.720 [mqtt debug] publish ebusd/global/uptime 2464
2024-12-06 15:21:05.741 [main debug] performing regular tasks
2024-12-06 15:21:15.741 [main debug] performing regular tasks
2024-12-06 15:21:20.734 [mqtt debug] publish ebusd/global/uptime 2480
2024-12-06 15:21:20.734 [mqtt debug] publish ebusd/global/uptime 2480
```
And the device status page
```
async mode: true
software serial mode: true
uptime: 126936 ms
last_connect_time: 1209 ms
reconnect_count: 1 
rssi: -69 dBm
free_heap: 160644 B
reset_code: 3
loop_duration: 22 us
max_loop_duration: 1221661 us
version: 
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 0
```
It looks like as if there is no ebus data found? 
Am I doing something wrong?

baahver, 2024-12-06:
The lights are red and I can make a network connection to the adapter. http://192.168.188.104/status
Could this be a power issue? Should I connect a usb-c power adapter to it?

danielkucera, 2024-12-06:
Try setting pwm value to 130. I can see 0 in status now.

baahver, 2024-12-06:
Yes thank you. That makes available:
- Ebusd basv
- Ebusd broadcast
- ebusd hmu
With a lot of entities. But all unavailable.
So I assume I'm missing some config now.

bilouxp, 2025-10-02:
Hello, I have the same problem as @baahver 
Can you give me a link to download the BIN file like @baahver's?
Thank you.