danielkucera/esp-ebus #97: PWM tuning?

JayUK, 2024-12-23:
Hi there,

I have the adapter connected to a Vaillant Ecotec 837 combi boiler. After a lot of trial and error I discovered I had to remove the jumper - which is not documented anywhere.

Anyway, the adapter and boiler seem to be communicating, I have the default PWM value of 130.

Does this value need optimising/tuning? How do you know if it's set to the ideal value?

danielkucera, 2024-12-24:
Hi @JayUK ,
It is somehow documented but I admit not very clearly because it is inherited from previous version:
https://github.com/danielkucera/esp-arduino-ebus?tab=readme-ov-file#v52

If you don't see communication errors in ebusd logs the value should be correct, it looks like 130 fits most installations.

yarikuaks, 2025-01-26:
I have a GB-E-20-S boiler and a Wolf DRT 2D control panel with SCOM selected via jumpers. The Wi-Fi module eBus 6.3 has its jumper removed, and power is supplied via USB-C. In this setup, the D1 LED blinks at a high frequency, with a 4-second pause between blink series.

In the log, I see the following:

```  
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=aeph8ogeithu1ief5Rohcohbooxeich6ohphah9eec6IeChoom9nie0Oiqui4ahn --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.0.171:3333 --log=all:notice  
2025-01-26 13:46:03.748 [device notice] transport opened  
2025-01-26 13:46:03.752 [main notice] ebusd 24.1.p20241115 started with broadcast scan on device: 192.168.0.171:3333, TCP  
2025-01-26 13:46:04.372 [bus notice] bus started with own address 31/36  
2025-01-26 13:46:04.733 [mqtt notice] connection established  
2025-01-26 13:47:56.498 [device error] transport closed  
2025-01-26 13:47:56.498 [bus notice] device invalid  
2025-01-26 13:48:01.502 [device notice] transport opened  
2025-01-26 13:48:01.502 [bus notice] re-opened 192.168.0.171:3333  
2025-01-26 13:48:14.550 [main notice] update check: revision 24.1 available  
2025-01-26 13:51:20.585 [device error] transport closed  
2025-01-26 13:51:20.585 [bus notice] device invalid  
```

I tried changing the frequency from 90 to 180, but it made no difference.

danielkucera, 2025-01-28:
Hi @yarikuaks ,

can you send contents of device status (`/status`) before and after the 4 second pause?

yarikuaks, 2025-02-02:
> Hi [@yarikuaks](https://github.com/yarikuaks) ,
> 
> can you send contents of device status (`/status`) before and after the 4 second pause?

async mode: true
software serial mode: true
uptime: 53656 ms
last_connect_time: 274 ms
reconnect_count: 1 
rssi: -45 dBm
free_heap: 161216 B
reset_code: 3
loop_duration: 22 us
max_loop_duration: 3914266 us
version: v7.1
nbr arbitrations: 0
nbr restarts1: 0
nbr restarts2: 0
nbr lost1: 0
nbr lost2: 0
nbr won1: 0
nbr won2: 0
nbr late: 0
nbr errors: 0
pwm_value: 130

danielkucera, 2025-02-02:
I suppose, this is after the pause. 
I don't see any reconnects there but I see max_loop_duration of about 4sec, this is interesting... 
What could be blocking there? I need to do some checks.

yarikuaks, 2025-02-03:
don't waste your time. I made a mistake, my Wolf GB-E-20-S does not support EBus, only SCOM.