danielkucera/esp-ebus #100: Adapter address identity on eBUS

21Development, 2025-01-13:
When scanning my eBUS using

ebusctl info

I see the following listed; I'm trying to determine if this is the adapter or my vr65?

address 31: master #8, ebusd
address 36: slave #8, ebusd

Should the adapter show up when scanning?

danielkucera, 2025-01-14:
Hi @21Development ,
none of these is your vr65, these two represent ebusd master and slave.

I can tell you more if you provide logs as described in troubleshooting readme.

21Development, 2025-01-14:
Hi @danielkucera,

I've had a adapter running for 6+ months and only just got round to learning and setting it up properly. It seems like none of the devices on my eBUS are fully supported out of the box. I've started to create the inc/csv files for my boiler (ecoTEC 418) which is looking okay so far. I'm struggling with a csv for my room stat (Vaillant / Netatmo vSmart) as I needed to start the csv from scratch and the vr65 (controlls the 2-port valves and NTC) doesn't seem to be detected on the eBUS at all.

I've copied a few mins of the logs below in case you can see anything obvious :)

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=deleted --mqttpass=deleted --scanconfig --mqttjson --configpath=/config/ebusd/ebusd-configuration/ebusd-2.1.x/en --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.my.ip:3335 --log=all:notice --lograwdata=data --latency=2000 --log=all:debug
2025-01-14 14:59:36.827 [device notice] transport opened
2025-01-14 14:59:36.828 [main info] registering data handlers
2025-01-14 14:59:36.835 [mqtt info] mosquitto version 2.0.20 (compiled with 2.0.20)
2025-01-14 14:59:36.837 [mqtt debug] connection requested
2025-01-14 14:59:36.837 [main info] registered data handlers
2025-01-14 14:59:36.837 [main notice] ebusd 24.1.p20241115 started with broadcast scan on device: 192.my.ip:3335, TCP, enhanced
2025-01-14 14:59:36.837 [main info] loading configuration files from /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/
2025-01-14 14:59:36.837 [main debug] reading directory /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/
2025-01-14 14:59:36.838 [main debug] file type of /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/broadcast.csv is file
2025-01-14 14:59:36.838 [main debug] file type of /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/vaillant is dir
2025-01-14 14:59:36.838 [main debug] file type of /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/_templates.csv is file
2025-01-14 14:59:36.838 [main debug] file type of /config/ebusd/ebusd-configuration/ebusd-2.1.x/en/memory.csv is file
2025-01-14 14:59:36.838 [main info] reading templates /
2025-01-14 14:59:36.840 [main info] read templates in /
2025-01-14 14:59:36.840 [main info] reading file broadcast.csv
2025-01-14 14:59:36.841 [main info] successfully read file broadcast.csv
2025-01-14 14:59:36.842 [main info] reading file memory.csv
2025-01-14 14:59:36.843 [main info] successfully read file memory.csv
2025-01-14 14:59:36.843 [main info] read config files, got 11 messages
2025-01-14 14:59:36.846 [bus notice] bus started with own address 31/36
2025-01-14 14:59:36.846 [device notice] reset
2025-01-14 14:59:36.860 [bus debug] ERR: SYN received during no signal, switching to ready
2025-01-14 14:59:36.860 [bus notice] signal acquired
2025-01-14 14:59:37.254 [mqtt notice] connection established
2025-01-14 14:59:37.254 [mqtt debug] publish ebusd/global/version "ebusd 24.1.p20241115"
2025-01-14 14:59:37.254 [mqtt debug] publish ebusd/global/running true
2025-01-14 14:59:37.254 [mqtt debug] publish ebusd/global/running true
2025-01-14 14:59:37.919 [bus notice] new master 10, master count 2
2025-01-14 14:59:37.959 [bus notice] new master 03, master count 3
2025-01-14 14:59:37.959 [update info] received MS cmd: 1008b5100900007a6affff04ff00 / 0101
2025-01-14 14:59:37.959 [update notice] received unknown MS cmd: 1008b5100900007a6affff04ff00 / 0101
2025-01-14 14:59:37.959 [bus notice] <1008b5100900007a6affff04ff00060001019a00
2025-01-14 14:59:38.736 [network info] [00001] client connection opened 127.0.0.1
2025-01-14 14:59:38.737 [network info] [00001] connection closed
2025-01-14 14:59:39.061 [bus debug] ERR: ACK error during receive response ACK, switching to skip
2025-01-14 14:59:39.096 [bus notice] <0364b5120202006600000069c09800
2025-01-14 14:59:39.738 [network debug] dead connection removed - 0
2025-01-14 14:59:43.258 [update info] received MS cmd: 1008b5110100 / 084e03ff7a040f0001
2025-01-14 14:59:43.258 [update notice] received unknown MS cmd: 1008b5110100 / 084e03ff7a040f0001
2025-01-14 14:59:43.263 [bus notice] <1008b51101008800084e03ff7a040f00010300
2025-01-14 14:59:43.550 [update info] received MS cmd: 1008b5110101 / 0968440080ff2c0100ff
2025-01-14 14:59:43.550 [update notice] received unknown MS cmd: 1008b5110101 / 0968440080ff2c0100ff
2025-01-14 14:59:43.551 [bus notice] <1008b511010189000968440080ff2c0100ff8400
2025-01-14 14:59:44.455 [bus notice] <0364b512020200660069c09800
2025-01-14 14:59:44.455 [bus debug] ERR: SYN received during receive response, switching to ready
2025-01-14 14:59:46.844 [main debug] performing regular tasks
2025-01-14 14:59:46.844 [main notice] starting initial broadcast scan
2025-01-14 14:59:46.844 [bus info] send message: 31fe07fe00
2025-01-14 14:59:46.846 [bus debug] start request 31
2025-01-14 14:59:46.846 [bus debug] arbitration start with 31
2025-01-14 14:59:46.900 [bus debug] arbitration won
2025-01-14 14:59:46.900 [bus debug] arbitration delay 18 micros
2025-01-14 14:59:46.900 [bus info] arbitration delay 18 - 18 micros
[... cut ...]
2025-01-14 15:00:20.155 [bus debug] send/receive symbol latency 8 ms
2025-01-14 15:00:20.164 [bus debug] send/receive symbol latency 8 ms
2025-01-14 15:00:20.173 [bus debug] send/receive symbol latency 8 ms
2025-01-14 15:00:20.173 [bus debug] switching from send command to send command CRC
2025-01-14 15:00:20.182 [bus debug] send/receive symbol latency 8 ms
2025-01-14 15:00:20.182 [bus debug] switching from send command CRC to receive command ACK
2025-01-14 15:00:20.197 [bus debug] switching from receive command ACK to receive respo

danielkucera, 2025-01-14:
Okay, I see there are multiple devices on the bus scanned correctly.

What is the actual question?

21Development, 2025-01-14:
> Okay, I see there are multiple devices on the bus scanned correctly.
> 
> What is the actual question?

I see there is a CSV for the vr65 (https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/64.v65.csv) but I don't see anything other than 08 (boiler), 15 (stat) and 31/36 (abusd) when running ebusctl info or ebusdctl find i.e. no 64/65 device

Am I missing something to expose the vr65 ? Is there a way to probe deeper on the eBUS to find it and/or make it show up so I can attach the 64.v65.csv and see the data?

danielkucera, 2025-01-14:
Most of messages to `64` are accompanied by an error. Do you power the device from bus or an adapter? Did you try to adjust PWM value?

21Development, 2025-01-14:
Are you referring to messages in the log above? Where do you see them?

the vr65 is a mains powered device - it essentially acts as a wiring center where the 2-port valves connect to.

Not tried adjusting PWM, is that done on the device configuration I/F? (I can’t remember what creds I set, i guess that means a reset?)

21Development, 2025-01-15:
@danielkucera Thanks for the tip about PWM!

I remembered my creds and adjusted PWM from 130 to 150, restarted ebusd and now see:

ebusctl info
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010002460.inc" ([PROD='0010002460']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=SCC00;SW=0600;HW=3903", loaded "vaillant/15.scc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 64: slave, scanned "MF=Vaillant;ID=V6500;SW=0007;HW=8301", loaded "vaillant/64.v65.csv"

Is it okay to leave PWM at 150 or should I try optimise to higher/lower values? - Should there be a 'master' showing for the 64 slave?

danielkucera, 2025-01-15:
I am glad that worked.

> Is it okay to leave PWM at 150 or should I try optimise to higher/lower values?

Sure, whatever value works best for you.

> - Should there be a 'master' showing for the 64 slave?

That is not a requirement of the protocol.