danielkucera/esp-ebus #120: Load config filie from WebUI for EBUS_INTERNAL

ajvdw, 2025-04-26:
Is it possible to add a lightweight WEBUI to load/clear/save commands in batch from a device-config.json file
This way the community can provide config files for specific devices or configurations.

danielkucera, 2025-04-26:
Yes, this is certainly possible thanks to very neat command interface provided by @yuhu- .
It just needs an extra http endpoint providing the interface and handling the data POST.

yuhu-, 2025-05-07:
If loading via the web UI works properly, I would remove the option to insert and remove eBUS commands via MQTT.

ajvdw, 2025-05-08:
For inspiration: Have a look at https://demo.emsesp.org/

It runs on an ESP32-C3, it's the UI that I have in mind for the future of the WEBUI.
The project is on Github: http://github.com/Emsesp/EMS-ESP32

ajvdw, 2025-05-17:
I've added the functionality in PR #127

yuhu-, 2025-05-18:
> It runs on an ESP32-C3, it's the UI that I have in mind for the future of the WEBUI.

To achieve this, more than just some changes to the existing system are required.

closed 2025-05-24 by ajvdw