danielkucera/esp-ebus #122: Suggestion: Remove dependency on MQTT and add an EBUS-Device-status endpoint

ajvdw, 2025-05-02:
I'm trying to add functionality to upload and save configurations (EBUS_INTERNAL).
It's giving me a hard time because of the dependency on MQTT.
Isn't it a good idea to separate the store part from MQTT.
An endpoint to retrieve the status of the EBUS device every 15 seconds or so would enhance the usablity greatly
Have a look at my first attempt for loading and saving the config (still work in progress).

https://github.com/ajvdw/esp-arduino-ebus

danielkucera, 2025-05-02:
Hi @ajvdw , thank you for the interest.
I can only see the implementation of http handlers (https://github.com/danielkucera/esp-arduino-ebus/compare/master...ajvdw:esp-arduino-ebus:master)

What is the dependency you see problematic? Can you point out specific lines?

danielkucera, 2025-05-04:
@ajvdw , I see now.
@yuhu- , there is quite some of mqtt logic implemented in Store class.
Can I/you make it "storage backend" agnostic?
I imagine there would be a separate loop for mqtt which would pull the data from store and publish it to mqtt.

yuhu-, 2025-05-04:
Due to the development process, classes and methods aren't yet clearly separated. Thanks for the tip. I'll take a look.

yuhu-, 2025-05-07:
Can you split your code (config via WebUI, value divider,...) and put into pull requests?

ajvdw, 2025-05-10:
Non MQTT / HA users, can benefit from an endpoint that serves a json array containing all responses to the stored/scheduled commands. It's a matter of having an array that is updated when an mqtt message is published. something like

{
"results":
[
     { "key1": ["30", "kWh", "123" ] },
     { <key>: [<value>, <>, <measurement age in seconds> ] },
]
}

yuhu-, 2025-05-10:
In the current implementation, each result (only the value) is published under `/values` under its own name specified by the user. The advantage of this is that the age of the last update matches the age of the measurement and the size of the message is very small. In my view, the unit should be intrinsically recognizable from the name. 

> measurement unit
> measurement name

This is static information that will always remain the same and it significantly increases the size of the MQTT message.

> measurement age in seconds

The result is only published and not stored internally. In order to implement your suggestion, the result and the timestamp of the last update must be retained for re-publication. Republishing will transmit the entire message and its size will grow linearly with the number of installed commands.

ajvdw, 2025-05-10:
I'm not suggesting to change the mqtt message only an additional webserver endpoint to retrieve all values at once. Typically an application calls this endpoint less often than once a minute for reducing the traffic and load on the MCU.

yuhu-, 2025-05-10:
> only an additional webserver endpoint

Okay. This feature was already integrated in an earlier development stage.

danielkucera, 2025-05-10:
The dependency on mqtt has been removed from store thanks to @yuhu- #124 .
Is there anything else needed withing this issue @ajvdw ?

Please let's keep issues/PRs as small as possible and don't mix different topics.

yuhu-, 2025-05-10:
LOL - The timestamp (millis()) is already stored - my bad.
uint32_t last;      // last time of the successful command

ajvdw, 2025-05-11:
This one can be closed. Do you want me to open another issue for EBUS-Device-status endpoint? I understood that @yuhu- already implemented it in his repo. @yuhu- Can you do a PR for [b4c6403](https://github.com/yuhu-/esp8266-arduino-ebus/commit/b4c640368d85ed93dad1b9403a91cb3d3e42dcf7)

yuhu-, 2025-05-11:
I will implement it again.

ajvdw, 2025-05-17:
Reviewed the code and seems ok.

closed 2025-05-17 by ajvdw