danielkucera/esp-ebus #123: Missing value divider in command structure for EBUS_INTERNAL

ajvdw, 2025-05-07:
A number of entries in the EBUSD configuration have the divider column filled with 10. This division can be done after reading in from MQTT but brings additional complexity for users. It would be nice to support the behavior through configuration.

yuhu-, 2025-05-07:
Can you split your code (config via WebUI, value divider,...) and put into pull requests?

yuhu-, 2025-05-10:
After scaling, the data type is changed to double. I would also suggest rounding the value to three significant decimal places. 
@danielkucera and @ajvdw  I'm not quite sure if this is a good idea to deliver all kinds of number types as double?

danielkucera, 2025-05-10:
I have a different suggestion:
Each command could have the latest raw response saved together with the receive timestamp (millis() ) and both mqtt and http can serve the response in it's preferred/defined shape and format.

yuhu-, 2025-05-10:
> raw response

Conversion, scaling and rounding should be done separately and only upon delivery?

ajvdw, 2025-05-11:
If you allow scaling on all datatypes, the consequence is that can end up with decimals. Doing the conversion upon delivery is good practice.

yuhu-, 2025-05-12:
Take a look at Redesign of data storage #126

ajvdw, 2025-05-17:
Tested and it seems to work.

closed 2025-05-17 by ajvdw