danielkucera/esp-ebus #129: Update error: Not Enough Space

yuhu-, 2025-05-17:
After the last 2 merges the total build size for ESP32 seems to be too big for my device. 
`Flash: [========= ]  91.0% (used 1192388 bytes from 1310720 bytes)`

Upload over espota returns ...
`20:41:05 [ERROR]: No response from device`
`*** [upload] Error 1`

Upload via Web interface returns ...
`Update error: Not Enough Space`

danielkucera, 2025-05-17:
That is weird, it should fit. The CI reports only 90%:
https://github.com/danielkucera/esp-arduino-ebus/actions/runs/15087832984/job/42412657454#step:8:221
and the partition table looks correct too:
```
~/esp/v5.2.1/esp-idf/components/partition_table/gen_esp32part.py .pio/build/esp32-c3/partitions.bin 
Parsing binary partition input...
Verifying table...
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1280K,
app1,app,ota_1,0x150000,1280K,
spiffs,data,spiffs,0x290000,1408K,
coredump,data,coredump,0x3f0000,64K,
```

I will need to investigate...

ajvdw, 2025-05-17:
I used this one for testing

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M

yuhu-, 2025-05-17:
It's of topic but SPIFFS could be a better place for config files.

danielkucera, 2025-05-17:
hmm, but the resulting file is bigger so platformio lies...
```
-rw-rw-r-- 1 danman danman 1312320 máj 17 21:34 .pio/build/esp32-c3/firmware.bin
```

danielkucera, 2025-05-17:
Can you try #130 ?
You will need to flash via serial first. Then the ota should start to work.

yuhu-, 2025-05-17:
> You will need to flash via serial first. Then the ota should start to work.

I only updated via ota or web so far. How do I flash via serial? What do I need for this?

danielkucera, 2025-05-17:
Yes, either that, or if you have been using 
`pio run -e esp32-c3-ota -t upload` then just change to 
`pio run -e esp32-c3 -t upload`

yuhu-, 2025-05-17:
connected via serial or ota ?

danielkucera, 2025-05-17:
serial

yuhu-, 2025-05-17:
Mai 17 23:05:37 vbox kernel: usb 2-1.6: new full-speed USB device number 7 using ehci-pci
Mai 17 23:05:38 vbox kernel: cdc_acm 2-1.6:1.0: ttyACM3: USB ACM device

PR #130 (2025-05-18): maximize firmware partitions to fix ota
Fixes #129
commit d02b3cbe (2025-05-18): maximize firmware partitions to fix ota (#130)

Fixes #129
closed 2025-05-18 by danielkucera