danielkucera/esp-ebus #131: Download commands - save file dialog?

yuhu-, 2025-05-17:
Is it possible to open a save file dialog instead of opening the json document?

yuhu-, 2025-05-18:
..about "save file dialog"

I think it would be better to have just one /commands endpoint.

The commands page should display the currently installed commands by default.

Buttons for 
- download (various formats?)
- upload
- load
- save
- wipe
- values

should be present.

Instead of the save file dialog, the commands could be saved directly.

@danielkucera and @ajvdw What do you think about it?

ajvdw, 2025-05-18:
I'm look at more than 40 commands (3 devices, heat pump, boiler, thermostat). So I don't think I would manipulate them from the UI. The typical use case is IMHO: Someone downloads the command set for his/her device uploads it and never will never access the WebUI again. 

What is needed is a  location where the different configurations can be downloaded, comparable to the ebusd configuration files. To get up to speed quickly, a converter for ebusd configurations files to commands would help.

yuhu-, 2025-05-18:
> I'm look at more than 40 commands (3 devices, heat pump, boiler, thermostat). So I don't think I would manipulate them from the UI. The typical use case is IMHO: Someone downloads the command set for his/her device uploads it and never will never access the WebUI again.

Nevertheless, the manipulation of the commands under an end point could be analogous to the configuration.